SpenceKonde/megaTinyCore #304: Servo library not functional at 20 MHz clock

DaleMitchell, 2021-01-25:
This code produces ~ 511 µs pulses at 16 MHz, but ~6.4 ms pulses at 20 MHz.

```
#include <Servo_megaTinyCore.h>

Servo myservoA;  // create servo object to control a servo

void setup() {
  myservoA.attach(PIN_PA4, 500, 2500);
}

void loop() {
  myservoA.writeMicroseconds(520);
  delay(1000);
}
```

SpenceKonde, 2021-01-25:
Eeeeewwww that's not good.

SpenceKonde, 2021-01-25:
6.4ms.. 6400us.. 128000 clock cycles, and the timer is running from CLK_PER/2 , 64000 TCB clock cycles. Instead of 5000 - somewhere, something is overflowing... and it's ending up gettting set to the maximum timeout period of the timer, or to a small negative number, which ends up as a number near the maximum when cast to an unsigned value...

SpenceKonde, 2021-01-27:
Uhm.... This issue is not reproducing for me, with that exact code. 
[image]

On my ATtiny1606 (don't have the 806, but they're identical except for flash size afaik), I get correct output with 20 MHz system clock.

SpenceKonde, 2021-01-27:
BTW, thanks for reporting this, even though I can't reproduce your reported issue (and this it's a mistake of some sort on your part) - it turned out to be completely hosed on the Dx-series... Now it works there.

SpenceKonde, 2021-01-27:
My suspicion though, is that somehow you were playing with an older version of the library where this line
```
#define REFRESH_INTERVAL       20000UL   // minumim time to refresh servos in microseconds
```
was instead:
```
#define REFRESH_INTERVAL       20000   // minumim time to refresh servos in microseconds
```
This results in something easy to mistake for correct fdunctioning at 16 MHz (it's still wrong, but less obvious), but incredibly obviously trashed at 20MHz

DaleMitchell, 2021-01-31:
Bizarre, the Arduino IDE must have been caching an older version of the servo library. Closing and reopening the IDE fixed the problem.

closed 2021-01-31 by DaleMitchell