SpenceKonde/megaTinyCore #307: Attiny 1616 trouble in paradise

saysphilippe, 2021-02-02:
Recreating the issue in the right repo. 
We are trying to program the 1616 but it's just saying "uploading..." until it times out or I disconnect the USB to the programmer(Nano). I have a breadboard with both the 3216 and 1616 and the 3216 works perfectly fine. What can be wrong. The pinout are identical as well so I'm scratching my head here.

avrdude: Version 6.3-20201216
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

     System wide configuration file is "C:\Users\PhilippeSchjelderup\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.6/avrdude.conf"

     Using Port                    : COM7
     Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
boot-loader FW version: 1
firmware version: 6.00
hardware version: 1
S_MCU:
boot-loader FW version: 1
firmware version: 6.00
hardware version: 1
Serial number: 00:00:00:00:00:00
Device ID: JTAGICE mkII
AVR Part : ATtiny1616
Chip Erase delay : 0 us
PAGEL : P00
BS2 : P00
RESET disposition : dedicated
RETRY pulse : SCK
serial program mode : yes
parallel program mode : yes
Timeout : 0
StabDelay : 0
CmdexeDelay : 0
SyncLoops : 0
ByteDelay : 0
PollIndex : 0
PollValue : 0x00
Memory Detail :

                              Block Poll               Page                       Polled
       Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
       ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
       signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
       prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
       fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
       fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
       lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
       data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
       usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
       flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
       eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

     Programmer Type : JTAGMKII_PDI
     Description     : JTAGv2 to UPDI bridge
     M_MCU hardware version: 1
     M_MCU firmware version: 6.00
     S_MCU hardware version: 1
     S_MCU firmware version: 6.00
     Serial number:          00:00:00:00:00:00
     Vtarget         : 5.0 V
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: jtagmkII_reset(): timeout/error communicating with programmer (status -1)
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_close(): timeout/error communicating with programmer (status -1)

avrdude done. Thank you.

the selected serial port
does not exist or your board is not connected

I have also downloaded the newest version of https://github.com/ElTangas/jtag2updi and tried pressing the reset button on the Nano when upload is just hanging. I have no problem uploading to the 3216 anyhow.

SpenceKonde, 2021-02-02:
I hate to say this but like... are you sure there isn't soldering,
alignment, etc issue, or that UPDI wasn't fused as reset or something on
accident somehow?

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

saysphilippe, 2021-02-02:
That can of course an option, but I soldered the 3216 the same way I did with the 1616. But I will use this set: https://www.aliexpress.com/item/32589469448.html?spm=a2g0o.productlist.0.0.28414243i0AK4a&algo_pvid=ea214278-7ad0-4860-b3b0-b8923030085e&algo_expid=ea214278-7ad0-4860-b3b0-b8923030085e-7&btsid=0bb0624516122572230723360edfc4&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_ on one new one and get back to you.

SpenceKonde, 2021-02-02:
Curse thee! I just bought a set of those little clips too... but yours cost
the same amount as mine did, but are distinctly better because they're so
narrow, while mine are bulkier

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

saysphilippe, 2021-02-02:
I managed to upload thru the clip part, but I have not soldered any different so I think I will reach out to Microchip and ask if they can test it more, because it looks like it really fragile unlike 1614,1604, 202, 402, 806, 3216 as I have tested with success :)

saysphilippe, 2021-02-02:
Since 1616 doesn't tolerate simple soldering I guess I will switch to the 3216 :)

closed 2021-02-02 by saysphilippe