SpenceKonde/megaTinyCore #308: MPLAB SNAP problem ..?

ffred, 2021-02-03:
Hi,
had a problem programming with mEDBG after updating from v2.1.3 to v2.2.6 (had another issue - closed - about that), so I'm now trying to use an MPLAB SNAP (PG164100) but still got problems.

first the SNAP was not known from Avrdude.
```
An error occurred while uploading the sketch
avrdude: Can't find programmer id "snap_updi"
...
```
after looking in my directories, I saw that the avrdude.conf I had in `...\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.6` was an old one without the definition for the SNAP or PICKIT4.
I replaced it with the one found here : `...AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18\etc`

I now go a step further, but got another problem :
```
An error occurred while uploading the sketch
avrdude: usbhid_open(): No device found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145
```

which seems right, since my SNAP got VID 04D8 & PID 9018.
seems to be a recent change from Atmel values to Microchip ones as you can see [here](https://www.eevblog.com/forum/microcontrollers/vid-and-pid-for-mplab-snap/)

now what I don't understand is that in avrdude.conf I have that :
programmer
```
  id    = "snap_updi";
  desc  = "MPLAB(R) SNAP in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x217F, 0x2180, 0x2181;
;
```
so, nothing about 0x2145 (nor 0x9018)  for the SNAP definition... then where does avrdude found the `VID 0x03eb and PID list: 0x2145` reported in the error message ?

will look more into that when I have more times..

ffred, 2021-02-03:
if I edit avrdude.conf to change the SNAP entry like this :
```
programmer
  id    = "snap_updi";
  desc  = "MPLAB(R) SNAP in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbvid = 0x03eb;  
  usbpid = 0x217F, 0x2180, 0x2181, 0x9018;
;
```
still got this error:
```
An error occurred while uploading the sketch
avrdude: usbhid_open(): No device found
avrdude: usbhid_open(): No device found
avrdude: usbhid_open(): No device found
avrdude: usbhid_open(): No device found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x217f, 0x2180, 0x2181, 0x9018
```
so the change for the VID is not taken...

ffred, 2021-02-03:
if I modifiy avrdude.conf to set the "good" vid & pid like that :
```
programmer
  id    = "snap_updi";
  desc  = "MPLAB(R) SNAP in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbvid = 0x04d8;  
  usbpid = 0x9018;
;
```
still got this error :
```
avrdude: usbhid_open(): No device found
avrdude: jtag3_open_common(): Did not find any device matching VID 0x04d8 and PID list: 0x9018
An error occurred while uploading the sketch
```
that's taking too much time for me, the SNAP will probably be the good option for me.........

SpenceKonde, 2021-02-04:
Well.... the next release *will* have the Snap in it's avrdude.conf (also a ton of crap formatting on that one fixed, lol) - though I just added the snap to the list in the other one. 

BUT personally, I couldn't even get the snap's I bought to work with MPLAB so I'm not a real authority on them!

ffred, 2021-02-04:
ok, thanks.
will let you know what works best for me...

SpenceKonde, 2021-02-04:
IMO the Microchip tooling isn't worth the trouble to figure out how to use. Not when a 95 cent serial adapter and a 2 cent resistor will make you a working updi programmer. Not when a $3 nano clone and a few scraps of wire can be turned into a updi programmer. And you can make one of each in the time it takes to just download microchip's bloated software.

Dlloydev, 2021-02-04:
I purchased a SNAP a few months back and yeah,headaches with the installation and setup. However, once you get it working, it really works well. I think what it boils down to is you need to setup a 32-bit MCU first so it'll install or update all the drivers it needs.
Check out [this link on AVR FReaks](https://www.avrfreaks.net/forum/mplab-snap?page=all) (posts 33-37). I didn't follow it exactly (chose a different 32-bit MCU), but it worked! After that, setting up and using an ATtiny1606 was no problem.

ffred, 2021-02-04:
@Dlloydev  ok, thanks.. no more time right now for testing, but will do soon..

@SpenceKonde  already done and use a nano clone with jtag2updi was, ok.. the little mEDBG I was using was really small and also got the RX/TX line for printf debugging. just receive the Curiosity board with the nEDBG. will test that also.

thanks guys..

ffred, 2021-02-08:
just a 5mn testing session.. as @Dlloydev suggested, with Microchip IPE after selecting an ATSAMD21E18 (if I remember well) a new firmware was installed in the MPLAB SNAP. then selecting again an ATTINY 3216 the firmware was changed again and I got replies that seems more encouraging that after my first try, but still not working. don't have time to go further right now...

@SpenceKonde , and with the Curiosity nEDBG, doesn't work since I got an avrdude error saying that it doesn't know the curiosity interface...

will stay with the v2.1.3 for now, with my ATMEGA32U4 mEDBG.
more info later when I could make it work.. had to work on something else right now.

SpenceKonde, 2021-02-08:
Does

SpenceKonde, 2021-02-08:
Does anyone have any ideas how to fix this? I've got nothing and I do not have hardware to test it with either - so I'm working blind here.

SpenceKonde, 2021-02-08:
Oh actually, it looks like I never actually pushed that change. Last week after the last update here I examined the two avrdude.conf files (the one from this core, and the one that came with the new avrdude version. I merged them to create a versions with the enhancements that megaTinyCore always had, plus the new stuff that went into the avrdude arduino18 package. So this may actually fix things. Some of the changes did involve the Microchip programmers.

freemovers, 2021-02-08:
@SpenceKonde, I have a PICkit4 that was not working till this last update. I got one of these to use the 12V high voltage pulse to erase the chip if needed when the UPDI pin is set as a reset pin.

SpenceKonde, 2021-02-08:
Oh sick, like the update 3 hours or so ago? That is awesome if so!

freemovers, 2021-02-08:
Yep, didn't bother to report the issue earlier since I didn't expect others to use the PICkit. But somehow my regular Arduino UPDI programmer didn't work this morning, so gave it another try after I noticed the changes to the avrdude files.

Dlloydev, 2021-02-08:
Awesome! Looks like its time to dig out the SNAP and the Updi-Key circuit I have on a breadboard. I have this combo working as a SNAP-HV programmer with MPLAB so it will be interesting to give this a workout with avrdude and megaTinyCore. My TO-DO list is getting out of hand, but this is now part of it.

SpenceKonde, 2021-02-08:
No shit? Phew! Works for me!

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

ffred, 2021-02-09:
I copied the new avrdude.conf and just tried to put a program for my board on the ATtiny3217 Curiosity Nano and got no error.
can't say more since my program isn't really done for that board, but seems to work.
will try later to connect it to my board to program it.

still doesn't work with the MPLAB SNAP, but it doesn't work either with MPLAB IPE, so the problem is somewhere else for this one and I probably will forget it and go with the curiosity after cutting the the debugger part..

and a last minute try with a borrowed Pickit4 and it's working, on my board, with the same wire adaptor I used with the SNAP !

Dlloydev, 2021-02-09:
Note that the SNAP debugger has a[ 4.7K pulldown resistor (R48) that needs to be removed](http://ww1.microchip.com/downloads/en/DeviceDoc/ETN36_MPLAB%20Snap%20AVR%20Interface%20Modification.pdf) from the UPDI signal.

[image]

ffred, 2021-02-09:
@Dlloydev , thanks for the info. didn't see that one before. will try that..

ffred, 2021-02-10:
ok, with R48 removed I can program my board with the MPLAB SNAP.
so I think it's ok now, we can close this issue..
will add info later when I could got time to cut the Curiosity nDBG and connect it to my board, as I still prefer that to the SNAP. much less problem, no need to change firmware ,cheaper and I have the RX/TX lines working with it. (still not sure if it can work on the MPLAB SNAP).

thanks a lot for your help !!

closed 2021-02-10 by ffred