SpenceKonde/megaTinyCore #309: tinyAVR 2-series

a7hybnj2, 2021-02-04:
They are here! I just ordered a tube of 10 1624's and 1626's 14 and 20 pin respectively.

Let me know if there is anything I can test on my end. 

I got them from Mouser.

SpenceKonde, 2021-02-04:
Woah! Crazy.

Well, I don't have anything written for them - I've been adjusting docs,
but there's no code or anything, lol... they're gonna need new variant
files, new boards.txt entries.... new bootloader configs too I guess,
though I bet the old binaries will work for USART0, just not USART1.

Logic needs to be ported to them...

I don't know when I'll even be able to have working core ready for them...
I was hoping they'd keep getting delayed some more.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

a7hybnj2, 2021-02-04:
Haha awesome. The full series isn't released but the 162x are out.

Most of that stuff sounds way beyond me. I will will play with them once they arrive and if I do anything useful I'll let you know.

If you need a tester for anything let me know.

SpenceKonde, 2021-02-08:
Also... goddamn, thanks for the tip about mouser vs digikey. Digikey is slipping! While Mouser? Mouser is *shipping*...  
Not only does mouser have the new parts that digikey doesn't, their prices on the whole Microchip product line are better! I do hope they change their mind about the VQFN's being non-stock though.

The 1627 is listed as coming in the microscopic VQFN (3x3mm- same as the attiny841's small version and the 1616 (but not the 3216!) . Think we'll get the ultra small package for the 32k parts this time around?  I hope so - you can put them on a breakout board that fits into a DIP socket : without wasting any pin positions (the 4x4mm like the 24-pin tinyAVRs) you lose the two middle pin positions to the chip. -P

freemovers, 2021-02-08:
You probably mean the ATtiny3226 will be the 3x3. The ATtiny1627 is 16k and 4x4.

gcormier, 2021-02-09:
Got my 1624 from Microchip direct but also noticed Mouser has stock too.

I asked Digikey, they were useless, said to ask Microchip. Asked Microchip, they were confused and said to ask the vendor. Disappointing indeed.

Put it on a test board, fired up atmel studio and looks good via atmel ice. Haven't dared to try avrdude, and I guess we're a way out from support for the 2 series.

a7hybnj2, 2021-02-09:
@gcormier I think you could copy the 1614 from avrdude and just change the IDs. I will be getting my chips sometime this week and will see how far I can get. Really I'll be waiting on Spence because the reason I am drawn to the 1626 over 1616 is the 2nd hardware uart...

gcormier, 2021-02-09:
Yup I chose 1624 because of 2 Serial.

Now, using the second serial (Serial1) has proven problematic compiling for 1614.

SpenceKonde, 2021-02-10:
github 'master' branch now hasI sketched in support (board defs made from 1-series with regex's, #ifdefs made less inclined to block compiles instead of spewing warnings, and tone got the new name for the clksel bitfield group codes).

I know all analog related stuff is broken, but please go determine what else is.

SpenceKonde, 2021-02-10:
clock source for millis timekeeping needs to be TCA0, TCB0, or TCB1

gcormier, 2021-02-10:
Just pulled master branch - Not seeing the 2 series

[image]

SpenceKonde, 2021-02-10:
They're under the 0/1-series definitions with the same number of pins

SpenceKonde, 2021-02-11:
Amended the boards.txt to make it more obvious where to look

SpenceKonde, 2021-02-11:
boards.txt also clarifies which timer options work

gcormier, 2021-02-11:
Flashed and some basic stuff works - need to whip out the scope or logic
probes to see if Serial is broken, or, if it's the LIN bus with an issue.

SpenceKonde, 2021-02-12:
Converting to a discussion topic :-)

closed 2021-02-12 by SpenceKonde