SpenceKonde/megaTinyCore #310: PlatformIO documentation

MCUdude, 2021-02-05:
Now that megaTinyCore is supported in PlatformIO, it would be nice if there was a readme file in this repo on *How to get started with megaTinyCore + PlatformIO*.

You can borrow as much as you want from the [MegaCoreX PlatformIO.md](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md) file.

There are a few minor things that differs from the MegaCoreX implementation:

* tinyAVR0/1 bootloaders are not currently supported.
  - `pio run -t bootloader` will not work, use `pio run -t fuses`
 * Dynamic fuse calculation is supported. You can specify things like `board_hardware.eesave = true`, and it will calculate the right fuses for you.
   - Note that `board_hardware.updipin` is used to set the UPDI pin state. It defaults to `updi` if not specified in Platformio.ini, but can also be set to `gpio` or `reset`. 
    - (Since all MegaCoreX targets have a dedicated UPDI pin, they can change their reset pin behaviour with `board_hardware.rstpin`)

If something is unclear, you can always take a look in the [PlatformIO fuses.py](https://github.com/platformio/platform-atmelmegaavr/blob/develop/builder/fuses.py) script.

SpenceKonde, 2021-02-07:
Since I don't use platformIO, this will probably be done using find-replace, and the file from MegaCoreX ;-)

SpenceKonde, 2021-02-07:
Oh dear, there's more to it than that isn't there. Well - I will try to throw something together, will likely borrow very heavily from your file, since like I said, I have never used platformio and don't really know anything about it other than that it's one of those things people say is great and that I will never have time to play with.

SpenceKonde, 2021-03-11:
I am not qualified to write this as I do not have background knowlesge and my schedule is WAY too busy (with core-maintenance related tasks!) to take on learning a new IDE right now.

SpenceKonde, 2021-03-24:
Thaaaanks

closed 2021-03-24 by SpenceKonde