SpenceKonde/megaTinyCore #311: Differential ADC really is useful

technoblogy, 2021-02-10:
Sorry, this isn't really an issue, but I just wanted to comment on your point:

> I can't imagine the differential ADC is a big deal for anyone reading this - I honestly don't think I've ever read a post on the Arduino forums where someone described using a differential ADC mode, and I, ah, read a lot of Arduino/AVR forum posts

Here are some uses:

* Reading the output from an electret microphone; for example:

http://www.technoblogy.com/show?ZU7

* Thermocouple amplifier:

http://www.technoblogy.com/show?2G9S

* High-side current measurement.

Measure the voltage across a 0.1Ω resistor in series with the supply.

Another point; as far as I can see the differential ADC in the ATtiny85 and ATtiny861 is superior to the differential ADC in the DA-series and DB-series:

* In the ATtiny85 and ATtiny861 the differential ADC inputs can go from 0V to VCC. For example, with VCC of 5V you can set a VREF of 1.1V, and even get the 10-bit ADC resolution over the range 3.9V to 5V.

* In the DA-series and DB-series the differential ADC inputs can only go from 0V to VREF, so the differential ADC offers no advantage over taking two normal ADC readings and subtracting them.

It looks like the differential ADC in the tinyAVR 2-series is more like the ATtiny85 and ATtiny861 type, which is great.

closed 2021-02-10 by SpenceKonde

SpenceKonde, 2021-02-10:
Fair point, readme amended, that was too flippant for a readme.

SpenceKonde, 2021-02-10:
Interesting what you say about the limitations of the DA/DB-series ADC in differential mode. Am a little unclear on what you mean by that. 

Speaking of classic ATtiny parts with sexy differential ADCs... have you ever looked at the ADC on the 841? It always seemed so random to me that the 841 would be the part with this super-fancy differential ADC with the multiple gain settings... Do you agree with that assessment?

technoblogy, 2021-02-10:
> Interesting what you say about the limitations of the DA/DB-series ADC in differential mode. Am a little unclear on what you mean by that.

I expected the differential ADC on the DB-series to work like on the ATtiny85, and discussed my puzzling results on AVR Freaks:

https://www.avrfreaks.net/forum/why-am-i-getting-large-error-when-using-differential-adc

The punch line is a bit lost in that discussion, but the key thing is in the datasheets:

ATtiny85:

[image]

AVR128DA28:

[image]

On the ATtiny85 the differential ADC inputs can go up to Vcc provided the difference isn't bigger than VREF.

On the AVR128DB28 the differential ADC inputs can only go up to VREF, same as the single-ended inputs, so you might as well make two single-ended readings and take the difference.

> Speaking of classic ATtiny parts with sexy differential ADCs... have you ever looked at the ADC on the 841?

No, I hadn't noticed that in the 841, but I agree it looks pretty impressive, with up to 100X gain on the differential inputs.

technoblogy, 2021-02-14:
I've just noticed something interesting about the ADC in the new ATtiny 2-series. They have an input voltage range of:

> -100mV to VDD+100mV

That's important because if you want to measure the current consumption of a circuit including the microcontroller, such as for coulomb counting to estimate the remaining battery life, you can put a low-value resistor, such as 0.1Ω, in series with the supply and measure the voltage across it.

However, one side of the resistor will be at a voltage slightly higher than VDD. For example, if the current consumption of your project is 1A it will be at VDD+100mV. This spec. ensures that you can still measure it with the ADC.

SpenceKonde, 2021-02-14:
If it's actually useful on the ends of its range, that's awesome...

technoblogy, 2021-02-14:
Now I just need to get some and try it out!

commit 0a7e1d22 (2021-02-10): Fix #311

Fair point, that was too harsh for a ReadMe.md.