SpenceKonde/megaTinyCore #312: Intermittent Flashing of Xplained ATTiny416 board

minimism, 2021-02-10:
(Using megaTinyCore 2.2.6; Arduino 1.8.13; MacOS 10.15.7)

I only got the Xplained 416 Nano board a few hours ago but I'm having trouble with flashing not always working. It seems that sometimes I can upload code if I first burn the bootloader, but subsequent attempts at flashing fail. Also, successive attempts at burning the bootloader fail.

I am using the _onboard ATMEL mEDBG_ programmer as the _nEDBG_ doesn't work at all (although I suspect that should be the 'right' one?)

Successful bootloader flash:
```
/Users/namke/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/Users/namke/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.2.6/avrdude.conf -v -pattiny416 -cxplainedmini_updi -Pusb -e -Ufuse0:w:0x00:m -Ufuse1:w:0b00000000:m -Ufuse2:w:0x02:m -Ufuse4:w:0x00:m -Ufuse5:w:0xC9:m -Ufuse6:w:0x07:m -Ufuse7:w:0x00:m -Ufuse8:w:0x2:m -Uflash:w:/Users/namke/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.2.6/bootloaders/optiboot_x/optiboot_xplained416.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/namke/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.2.6/avrdude.conf"
         User configuration file is "/Users/namke/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny416
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no       4096   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        128   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.13 (rel. 43)
         Serial number   : ATML2795042700002363
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "tiny"
[... 214 lines cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```

SpenceKonde, 2021-02-10:
The issue with bootloading 416 is - I think - fixed by 2.2.7  which should be available in board manager (and working) now (as of like 10 minutes ago, though there was a broken version listed earlier in the morning). Luckily I botched the binary packaging, so the bad 2.2.7 was never available on board manager, and before I realized that, I checked the CI logs and realized I'd also screwed up the core itself.

Problem uploading implies is consistent with wrong serial port selected or... almost any other problem (first thing I'd check is that if you reset the board, does the LED blink? If not, does an LED + resistor put from PA7 to ground flash when you reset the board? (if no for on-board LED, but yes for PA7 that suggests that the bootloader is getting built as a "normal" 416 bootloader (though I don't know, off-hand, what's different about 416 xplaineed boards - I don't have any!)

minimism, 2021-02-10:
Thanks for the speedy response :)

I just tried updating and get an error with the JSON file:
```
kipping contributed index file /Users/namke/Library/Arduino15/package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character ('{' (code 123)): was expecting comma to separate Array entries
 at [Source: (FileInputStream); line: 3704, column: 10]
 at [Source: (FileInputStream); line: 3703, column: 9] (through reference chain: cc.arduino.contributions.packages.ContributionsIndex["packages"]->java.util.ArrayList[2]->cc.arduino.contributions.packages.ContributedPackage["platforms"])
Skipping contributed index file /Users/namke/Library/Arduino15/package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character ('{' (code 123)): was expecting comma to separate Array entries
 at [Source: (FileInputStream); line: 3704, column: 10]
 at [Source: (FileInputStream); line: 3703, column: 9] (through reference chain: cc.arduino.contributions.packages.ContributionsIndex["packages"]->java.util.ArrayList[2]->cc.arduino.contributions.packages.ContributedPackage["platforms"])
```

minimism, 2021-02-10:
I also noticed that the bootloader upload works when the mEDBG LED is off, but not when it is on. Uploading a sketch still fails on the second/subsequent attempts. Current workaround is: unplug USB; replug USB; burn bootloader; burn sketch. Repeat :) I'm just playing with these at the moment but I am very excited at the prospect of moving a bunch of ATTiny85 code over to the new parts, so your efforts are well appreciated!

SpenceKonde, 2021-02-10:
This may be fixed now, just pressed backspace one time too many while deleting the bogus entry.

minimism, 2021-02-10:
@SpenceKonde it looks like my _Blink_ is toggling pin A7 when using `LED_BUILTIN` on this board. The user LED is attached to pin PB5 on this particular XPLained board.

minimism, 2021-02-10:
> This may be fixed now, just pressed backspace one time too many while deleting the bogus entry.

:sad face:

```
 Error downloading https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.2.7.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.2.7.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.2.7.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
```
I might need to wait for it to propogate through some intermediate network caches…

SpenceKonde, 2021-02-10:
euuurrghhhh more stiupid bullshit from the bunged release... it's there now (always was, but the filename was wrong, because I'd forgotten to correct it the second time around for the different name that got used because github doesn't let you reuse a version number.

SpenceKonde, 2021-02-10:
> @SpenceKonde it looks like my _Blink_ is toggling pin A7 when using `LED_BUILTIN` on this board. The user LED is attached to pin PB5 on this particular XPLained board.

Blink will use PA7 (I'll make a new issue for that, as this *should* be handled way better on official Microchip boards). the question is what the bootloader uses (assuming you even want to use the bootloader with it)

minimism, 2021-02-10:
> Blink will use PA7 (I'll make a new issue for that, as this _should_ be handled way better on official Microchip boards). the question is what the bootloader uses (assuming you even want to use the [...]

Well, with my previous Tiny projects I've not used a bootloader, so I'm not 100% sure what benefits I would gain. I don't honestly know what happens in the Mega32 which is performing the UPDI comms. Happy to help if this is a more widely-encountered issue otherwise I'll just rig up a separate UPDI programmer and use a bare Tiny.

SpenceKonde, 2021-02-10:
I mean, I wouldn't use the bootloader under most circumstances (and I'd sooner hit myself with a hammer than do it on a 4k part :-P) - but the second error you posted was a bootloader upload error, and you're using the Optiboot board definition, so I figured that was your plan. Some people like having one serial port to upload and get serial console on? (you also are sort of forced to use one if you can't get up close to it easily to connect a programmer, eg, the board is at the far end of an RS485 cable, so you'd better be able to program it through that (we do have people doing this with both this and the classic core)). I used to like that, but since I now avoid the arduino serial console like the plague anyway, it's way better for me to just have the IDE pointed at the UPDI programming pins, and serial adapter on the UART pins. (and then I leave the console open the whole time, and I don't have to race the sketch's startup time to open a serial window without missing stuff)

SpenceKonde, 2021-02-10:
Oh - the other good reason to use bootloader is if you need to write flash from application; On DxCore I was able to get this to work both with and without bootloader, but the flash write protect on the 0/1/2-series is a bit pickier it seems, and so the solution for it is more work. Flash write from app is actually in, currrently but I couldn't figure out the interface for it at first glance, and there wasnt an intelligible example either. so I went and did more important things.

minimism, 2021-02-10:
I suppose that I was only using the Optiboot option as it was the only one mentioning the Xplained 416 board … I probably should read some stuff before diving in.
My ATTiny85 debugging is usually via i/o pins and oscilloscope (audio stuff, so there's no time to spit out characters over serial), and I was expecting to do the same with these new parts too. The only bootloader I would be using would be [TinyAudioBoot](https://github.com/ATtinyTeenageRiot/TinyAudioBoot), and that's only for one particular scenario (and with the larger parts for sure).

My thinking was that since the board is mentioned by name, then the programming interface stuff would already be sorted for me; however that may very well have been a false assumption and I should just dig out the serial converter and a resistor…

At the risk of polluting this issue-thread, I'm still getting a file-not-found on the 2.2.7 update tar file (no underscore in the filename?? `Error downloading https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.2.7.tar.bz2`).

SpenceKonde, 2021-02-10:
Okay NOW you are in cache hell, I think. You should see that https://spencekonde.github.io/megaTinyCore/megaTinyCore-2.2.7.tar.bz2 does come up in a browser now (there is not supposed to be an underscore - the underscore shouldn't have been there on anything, ever. 

Sounds like I should do a bit more to clarify the correct board option to choose, heh... 

Reboot

minimism, 2021-02-10:
Right - great, thank you :) After a cup of tea everything works of course. I'm now on 2.2.7.

Putting aside the bootloader; it appears that I can still only upload once per power-cycle to the board. After a successful upload I see:
```

avrdude: verifying ...
avrdude: 1182 bytes of flash verified
avrdude: jtag3_edbg_signoff(): unexpected response 0x03, 0x80

avrdude done.  Thank you.
```
so I wonder whether the jtag signoff is failing somehow and the connection remains, making it unavailable for subsequent programming cycles. I only have a vague idea about how the debug connection works (I use JTAG on a day-to-day basis without really worrying how it works, just that it does!)

SpenceKonde, 2021-02-10:
Huh. Weird. I don't know what to say then x_x Is the f/w on the debug/programming chip updated?

minimism, 2021-02-10:
> Huh. Weird. I don't know what to say then x_x Is the f/w on the debug/programming chip updated?

Good question. I have so far been avoiding installing MPLAB etc. but I may have to to verify that using my old windows laptop. Alternatively I'll just solder the various Tinys that arrived today onto the SOIC-DIP adapters and play with those!

minimism, 2021-02-11:
Well, I just updated the firmare on the board using the freshest download of Microchip Studio, but no change to behaviour :(
```
/Users/namke/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/Users/namke/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.2.7/avrdude.conf -v -pattiny416 -cxplainedmini_updi -Pusb -Ufuse2:w:0x02:m -Ufuse6:w:0x07:m -Ufuse8:w:0x2:m -Uflash:w:/var/folders/tp/79kkq_6x6g71t06rvmw23q400000gp/T/arduino_build_231164/Blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/namke/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.2.7/avrdude.conf"
         User configuration file is "/Users/namke/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

I'll move on to using a bare device and a separate UPDI programmer.

SpenceKonde, 2021-04-01:
closing as my care package from microchip didn't include any of these and even if I had some on hand, it looks like it would need a fix in avrdude, which is not a road I am going to go down,.

closed 2021-04-01 by SpenceKonde