SpenceKonde/megaTinyCore #313: Make LED_BUILTIN use the on-board LED on Microchip development boards

SpenceKonde, 2021-02-10:
From #312

I don't suppose anyone has a list of the Microchip development boards with supported parts on them, and knows which pin they put the LED on? 

Simple to add, though.

SpenceKonde, 2021-03-27:
And nope no list lol....
Boards are:
Curiosity Nano 1607, 1627: LED on PB7, button on PC4 
Curiosity Nano 3217: LED on PA3, button on PB7
Xplained Nano 416: LED on PB5, button on PB4
Xplained Pro 3217, 817:  LED on PB4, button on PB5 (srsly?)

Looks to me like they have a list of pins that aren't important and can be used for the LED/button, but they don't always keep trac of which ones they used...

SpenceKonde, 2021-04-03:
Deferring to 2.4.0, though some things that got sorted out were prereqs for this. This is small enough that it might squeak into 2.3.x

hmeijdam, 2021-05-30:
I have the "Attiny817 Xplained Mini" board and it has the LED on PC0 and the button on PC5

SpenceKonde, 2021-07-19:
New microchip official board definitions are now in the core for 2.4.0 release/

closed 2021-07-19 by SpenceKonde