SpenceKonde/megaTinyCore #314: Eliminate chmod's from platform.txt

SpenceKonde, 2021-02-10:
I think this issue is actually entirely due to their having never had proper file attributes set in the repo... (based on my toolchain experiences). If so we can easily get rid of these :-)

SpenceKonde, 2021-02-26:
Anyone with a setup on linux want to grab this repo, chmod +x megaavr/create_disassembler_listing.sh, commit, and send me a PR?

I figure this is trivial for some of you, and kind of a pain for me because all my dev systems are windows and I don't know how to use git from the command line. 
Thanks

DaleMitchell, 2021-02-28:
First, check if `git config core.fileMode` returns true.

If it returns false, run `git config core.fileMode true`, and then:

`git update-index --chmod=+x megaavr/create_disassembler_listing.sh`

Permissions can be checked with `git ls-files --stage megaavr/create_disassembler_listing.sh`

`100644` is the old value, `100755` I think is the correct value.

I sent a PR.

closed 2021-03-02 by SpenceKonde