SpenceKonde/megaTinyCore #323: Garbled serial output from ATtiny1616 when using Platformio IDE, OK on Arduino IDE

ArnieO, 2021-02-15:
I posted [this issue](https://github.com/platformio/platform-atmelmegaavr/issues/22) also on https://github.com/platformio/platform-atmelmegaavr last week, but no response so far, so I try again here, although I suspect the problem not being megaTinyCore. Any advice would be highly appreciated!

*********************************************************************

Code in PlatformIO IDE:
```
#include <Arduino.h>
void setup() {
  Serial.swap(1);       // Switch to alternate serial port.
  Serial.begin(115200);
  while (!Serial);
}
void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```
platformio.ini:
```
[env:ATtiny1616]
platform = atmelmegaavr
board = ATtiny1616
framework = arduino
upload_port = COM24
monitor_port = COM10
monitor_speed = 115200
```

- When I build and upload this from PlatformIO IDE the serial output on alternate USART pins (PA1, PA2) is garbled (output looks like wrong baud rate):
[image]
- If I use primary USART (delete `Serial.swap(1) `or replace it with `Serial.swap(0)`) the same output continues on alternate USART, nothing on primary USART. I tried `Serial.pins()` instead of `Serial.swap()` -> same result.
- Same code built on Arduino IDE (except first #include line) works as expected, both on primary and alternate USART.

In both cases I program via UPDI / jtag2updi.

Any advise on what is going on here would be highly appreciated, as I prefer working in PlatformIO IDE.

SpenceKonde, 2021-02-15:
I agree, looks like wrong baud rate, that is that chip not running at speed you intended for it to run at. What speed are you intending to run it at?... I don't see anything in there about setting the baud rate. 

I have never used platformIO though, so I'

ArnieO, 2021-02-15:
I can upload and run code without problem - and everything else works exactly like when uploaded from Arduino IDE including OLED display on I2C, blinking LEDs on various pins etc. It is only the serial output that does not work - on both USART outputs. 

So I stripped down the code to the above, to be sure it wasn't something else in my code that was the culprit.

SpenceKonde, 2021-02-15:
you can print to an oled display? Print out the values of USART0.BAUD (a 16-bit register) , and  USART0.CTRLB (which is just 8-bit) and tell me what they are? (including what base you're printing them as)  Actually, I think I just need (USART0.CTRLB & 0x02)
Also let me know the system clock speed,

SpenceKonde, 2021-02-15:
Also what version of megaTinyCore is this based on?

ArnieO, 2021-02-15:
[image]
Binary and hex shown for both.

> Also what version of megaTinyCore is this based on?

That's a bit more tricky. As indicated above, I use the atmelmegaavr platform which as I understand incorporates megaTinyCore according to this thread: https://github.com/platformio/platform-atmelavr/issues/83
But I am not really sure how to find out which version it is.

megaTinyCore is referred to in the documentation here, linking to your repository:
https://docs.platformio.org/en/latest/platforms/atmelmegaavr.html?utm_medium=piohome&utm_source=platformio#packages

ArnieO, 2021-02-15:
**Eureka!**
Clock speed!
This page indicates the default clock speed is 16 MHz.

So I added this to platformio.ini:
`board_build.f_cpu = 20000000L`

And the serial output is now fine!
[image]

Thank so much you for pointing me in the right direction!

SpenceKonde, 2021-02-15:
Woot! And yeah, I was about to tell you that it thinks its running at 16 MJz. The way clock on these works is kinda weird. The OSCCFG fuse selects between nominal 20 MHz vs nominal 16 MHz for internal oscillator..  Either can then be prescaled down (for example to run at lower voltage), They can also both be calibrated within a range of approximately 5/8ths nominal frequency to 1 5/8ths nominal. (even though both of the two oscillator settings can generate both 16 and 20 MHz, there are two separate settings, with their own calibration curves.

SpenceKonde, 2021-02-15:
It's too bad we can't get sll the spiffy features for the Arduino IDE on platform, it seems a fair number of people like that over the Arduino IDE = among others, I'm thinking of automatically setting required fuses on upload, which we do as of 2.2.0. 

There are a bunch of version number defines. Try `Serial.println(MEGATINYCORE);` - failing that, you can print the pieces one at a time with MEGATINYCORE_MAJOR MEGATINYCORE_MINOR, and MEGATINYCORE_PATCH

closed 2021-02-15 by SpenceKonde

ArnieO, 2021-02-17:
@SpenceKonde -> I know you are not using PlatformIO yourself, but in case someone else sees this:

Processor frequency (amongst other things) is set by the so called process of "burning fuses" (I think "setting switches" would be a better term, as they can be reset). But I have not managed to figure out how to burn fuses on for instance an ATtiny1616 from PlatformIO. There is a "Burn fuses" command, which gives this:
[image]

It all looks nice and shiny except for the annoying red text. And nothing happens. 

I can, however, reprogram from Arduino IDE with a changed setting - and then continue programming from PlatformIO assuming the new fuse setting. Which seems to kind of work - except that setting of baud rate with Serial.begin(xxx) does not do anything. I have to set the "real" baudrate with the slightly more cumbersome
[image]
then
[image]

Serial.begin(xxx); is still needed, but must be followed by the setting of USART0.BAUD.