SpenceKonde/megaTinyCore #325: Question on Power Saving example using 1614

ldalep, 2021-02-16:
This is probably obvious but using your example 
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/PowerSave.md 

I changed the following  **  (OBTW getting better than 20ppm on External 32Khz crystal). 

void RTC_init(void)
{
  /* Initialize RTC: */
  while (RTC.STATUS > 0)
  {
    ;                                   /* Wait for all register to be synchronized */
  }
**  RTC.CLKSEL = RTC_CLKSEL_TOSC32K_gc;    /* 32.768kHz Internal Ultra-Low-Power Oscillator (OSCULP32K) */

  RTC.PITINTCTRL = RTC_PI_bm;           /* PIT Interrupt: enabled */

  //RTC.PITCTRLA = RTC_PERIOD_CYC32768_gc /* RTC Clock Cycles 32768, resulting in 32.768kHz/32768 = 1Hz */
**  RTC.PITCTRLA = (0x0E<<3)
  | RTC_PITEN_bm;                       /* Enable PIT counter: enabled */
}

I am getting 2 times faster than the advertised rate

Notice that the enum starts at 4???? for the 1614

typedef enum RTC_PERIOD_enum
{
    RTC_PERIOD_OFF_gc = (0x00<<3),  /* Off */
    RTC_PERIOD_CYC4_gc = (0x01<<3),  /* RTC Clock Cycles 4 */
    RTC_PERIOD_CYC8_gc = (0x02<<3),  /* RTC Clock Cycles 8 */

SpenceKonde, 2021-02-16:
Edited your issue to add code tags. (isn't that much better?)

I would agree with your assessment. Either the io headers or the datasheet is straight up, no excuses, **wrong**. - These are the times when I say "thank you" for the world of malware we find ourselves asdridt with

SpenceKonde, 2021-02-16:
This discrepancy between the io headers and the datasheet is present for **_EVERY SINGLE AVR RELEASED SINCE 2016!!!_**

SpenceKonde, 2021-02-16:
In any event, that ain't *my* example. @freemovers  - did you notice that the timing was busted when you wrote this? Presumably "no"

freemovers, 2021-02-16:
I haven't noticed an issue with timing, and it looks like the enum is matching the datasheet:
[image]
I will do some more research later tonight.

ldalep, 2021-02-16:
My Fault  - I apologize!  That's what you get when you count blinking LED's at  1-mississippi, 2-mississippi........   and seeing it start at something else than 2 cycles.    Again sorry for the RATHOLE. 

DP

SpenceKonde, 2021-02-17:
And *I* was looking at the PRESCALE bitfield of RTC.CTRLA rather than the the PERIOD bitfield of RTC.PITCTRLA.... 

Though in my defence, the organization and register naming on the RTC is somewhat perverse - there's a RTC.CTRLA and RTC.PITCTRLA .... both of which have something measured in a power of two cycles, but starting from a different number!

closed 2021-02-17 by SpenceKonde