SpenceKonde/megaTinyCore #326: 2.2.7 release does not compile code with the `constrain` macro

DaleMitchell, 2021-02-16:
This issue is present with megaTinycore (2.2.7) installed through Arduino IDE's (1.8.13) Board Manager.

Example:
```
void setup() {
  uint8_t testVal = 5;
  uint8_t resultVal = constrain(testVal, 10, 150);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

The above code does not compile, here are the warnings and errors:
```
"C:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino3/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=attiny806 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATtiny806 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.2.7\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=2UL -DMEGATINYCORE_PATCH=7UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 -DNOSERIALEVENT -DMILLIS_USE_TIMERA0 -DUARTBAUD5V "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.2.7\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.2.7\\cores\\megatinycore" "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.2.7\\variants\\txy6" "C:\\Users\\dalem\\AppData\\Local\\Temp\\arduino_build_637196\\sketch\\sketch_feb16a.ino.cpp" -o "C:\\Users\\dalem\\AppData\\Local\\Temp\\arduino_build_637196\\sketch\\sketch_feb16a.ino.cpp.o"
In file included from C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/ArduinoAPI.h:52:0,
                 from C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/Arduino.h:23,
                 from C:\Users\dalem\AppData\Local\Temp\arduino_build_637196\sketch\sketch_feb16a.ino.cpp:1:
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino: In function 'void setup()':
C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/Common.h:45:22: error: 'l' was not declared in this scope
   typeof (low) _l = (l);           \
                      ^
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino:3:23: note: in expansion of macro 'constrain'
   uint8_t resultVal = constrain(testVal, 10, 150);
                       ^~~~~~~~~
C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/Common.h:45:22: note: suggested alternative: '_l'
   typeof (low) _l = (l);           \
                      ^
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino:3:23: note: in expansion of macro 'constrain'
   uint8_t resultVal = constrain(testVal, 10, 150);
                       ^~~~~~~~~
C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/Common.h:46:23: error: 'h' was not declared in this scope
   typeof (high) _h = (h);          \
                       ^
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino:3:23: note: in expansion of macro 'constrain'
   uint8_t resultVal = constrain(testVal, 10, 150);
                       ^~~~~~~~~
C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/Common.h:46:23: note: suggested alternative: '_h'
   typeof (high) _h = (h);          \
                       ^
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino:3:23: note: in expansion of macro 'constrain'
   uint8_t resultVal = constrain(testVal, 10, 150);
                       ^~~~~~~~~
C:\Users\dalem\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.2.7\cores\megatinycore/api/Common.h:47:35: error: expected ';' before '}' token
   _x < _l ? _l : _x > _h ? _h :_x })
                                   ^
C:\Users\dalem\AppData\Local\Temp\arduino_modified_sketch_452498\sketch_feb16a.ino:3:23: note: in expansion of macro 'constrain'
   uint8_t resultVal = constrain(testVal, 10, 150);
                       ^~~~~~~~~
exit status 1
Error compiling for board ATtiny3216/1616/1606/816/806/416/406.
```

Downgrading to 2.2.6 using Boards Manager resolves the issue. Upgrading back to 2.2.7 causes the same issue (even with closing and reopening the IDE, so I'm definitely not experiencing a caching issue like last time with the servo library).

SpenceKonde, 2021-02-18:
uuuuuggghhh.... I copied that from somewhere else, but forgot that it's a macro so it would have zero syntax checking until I tried to use it. 

We need a sketch for testing those - all the little functions/helpers/macros that are in Arduino.h  - something that calls every one of it with each plausible permutation of datatypes and makes sure that it at least compiles.

SpenceKonde, 2021-02-18:
(also, fixed yesterday and thought I had commented here)

closed 2021-02-18 by SpenceKonde

DaleMitchell, 2021-02-18:
Excellent idea on the unit testing! In addition to testing that it compiles, is it possible to check outputs? Perhaps with the help of this project? https://github.com/mmurdoch/arduinounit/

SpenceKonde, 2021-02-18:
If you want to make that happen I would be happy to use it, but you're lucky to even have compile tests :-P See the issue with improving the CI - right now our CI testing is testing only a tiny fraction of what it could be, because most of the tools submenus are ignored (am shocked ANYTHING compiles with those menus not specified, lol)

SpenceKonde, 2021-02-18:
I work in software test, the last thing I want to do after a long day writing tests for software is to write tests for software!