SpenceKonde/megaTinyCore #330: Flashing with Optiboot and UPDI

ArnieO, 2021-02-19:
After having flashed the bootloader I am no longer able to upload code via jtag2updi. 
Settings:
[image]

This happens:
[image]

Shouldn't the programmer be jtag2updi?

freemovers, 2021-02-19:
See [https://github.com/SpenceKonde/megaTinyCore/issues/320](https://github.com/SpenceKonde/megaTinyCore/issues/320)
jtag2updi is only used to upload the bootloader to the MCY. After having a bootloader on the MCU you no longer need a programmer.

ArnieO, 2021-02-19:
I want to flash the ATtiny via UPDI in such a way that it reboots the code when I power cycle it. 
I am on a learning curve on ATtiny so I might have misunderstood something, but my current understanding is that it will then need the bootloader.
So how do I achieve this?

SpenceKonde, 2021-02-19:
What do you mean by "reboots the code"? 

Under the normal usage of the word "reboot" that would mean that execution of the uploaded code would start from the beginning on power up. You don't need a bootloader for that (though "burn bootloader" is also the only way to set some fuses - doing burn bootloader with a board definition that doesn't have a bootloader will just set the fuses. I can't change what that menu option is called though). 

The bootloader allows you to upload a new sketch over a serial port connected to TX and RX (typically immediately after powercycling the board or doing a software reset from within the application (for ersatz-reset), though one could disable UPDI and use hardware reset too.

ArnieO, 2021-02-19:
Huh. 
So why won’t my ATtiny start executing the code I have flashed when I power cycle? I was surprised to discover this, and have been thinking it needed the bootloader. So I would be happy to hear that it is not normal!
Wrong Arduino IDE settings?

closed 2021-02-20 by ArnieO

ArnieO, 2021-02-20:
OK... I found the culprit: Brown-out detection (BOD) level. 
It was not activated, so ATtiny was not reset properly when powered down.

I have experimented a bit, and see that I must avoid running it with BOD deactivated; most times it does not restart at all, by increasing Startup time from the default 8ms I can make it start some times (not reliably). 

I conclude I must used it with either internal BOD active, or use an external voltage supervisor circuit attached to UPID/RESET' pin (with the known disadvantages).

**My design is on 3.3V and I find that setting BOD active / BOD level 2.6V level gives reliable restart.**

ArnieO, 2021-02-20:
@SpenceKonde 
I find the text in the Arduino IDE misleading on these two parameters, so you might want to consider adjusting it:
[image]

I can adjust both parameter and flash chip over jtag2updi without burning bootloader.

SpenceKonde, 2021-02-20:
But when you do a normal upload, the BOD settings arent changed. BOD fuse
isn't written when you upload,  since it is not a "safe" fuse - setting it
higher than vcc bricksthe chip if you cant run it at the highervoltage to
reprogram because its connected to stuff that would be damaged

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

ArnieO, 2021-02-20:
Well, what can I say... 
I have definitively not burned bootloader today, and reflashing the code after changing the BOD setting clearly changed the behavior as explained.

SpenceKonde, 2021-02-20:
I have just tested this, and confirmed that changing BOD settings does *not* change what is written to the chip for upload - the hex file is binary identical and the fuse settings passed to the uploader are unchanged for both avrdude and prog.py uploads.... so there is no possible way that the menu option could make a difference. Unless you choose burn bootloader, which does write "unsafe" fuses (like BOD) which, because of potential to brick the board, we never set unsafe fuses (syscfg1 for reset pin,,bodcfg for the BOD) nor do we set "unused' fuses (ones which are not configured by the core) except on bootload (on the grounds that if someone changed those, they'd want them to stay that way when uploading, whereas someone who does burn bootloader wants the chip configuration to be set to a fully known state)

ArnieO, 2021-02-20:
I fully understand the risk of bricking the chip if such fuses were changed during a normal upload. It is a good solution to do that separately. So this is a bit strange.

I will test this again tomorrow and report back:
- Disable BOD in menu, code upload over jtag2updi, check if chip reboots after power cycling
- Re-enable BOM in menu, re-upload code, re-check response to power cycling

ldalep, 2021-02-20:
For What it is worth - I have problems resetting my 1614 if I leave the RX Pin which is hooked to the USB-Serial TX.  Not sure that it doesn't supply some from the pull up via the LED.  In any case I have to make sure any external supply from another pin is gone or I have to use the return GND from the chip to reset it.   Normally I wouldn't need the external connections so I don't worry about it .... therefore I let the jtag2updi do the reset.

ArnieO, 2021-02-20:
> For What it is worth - I have problems resetting my 1614 if I leave the RX Pin which is hooked to the USB-Serial TX. Not sure that it doesn't supply some from the pull up via the LED. In any case I [...]

Good point, I have the FDTI hooked up (GND, Rx and Tx). I will disconnect it tomorrow and check if that has an impact. 
The reason I am so focused on this is I am bredboarding a PCB design that is based on programming the ATtiny over UPDI only,  so verifying that the chip restarts on power cycling is important.

SpenceKonde, 2021-02-20:
Oh for sure - leaving the TX line of the adapter connected will back-power the chip through the protection diode! If the adapter has a 1-2k resistor (as many do) in series with TX, that'll limit current to a couple of ma - which is likely enough to keep Vcc from falling below the POR rearm voltage, but keep it low enough that the part won't function. If the adapter doesn't have that resistor, the current will be a function of the load on the board with the part (from the attiny and any other parts) and the strength of the output drivers on the TX line. It is certainly possible to exceed the maximum current injection (current through the protection diode) in that way, even with the far more generous allowance that these parts give us compared to classic AVRs (we're allowed to inject +/- 15mA as long as Vcc < 4.9V - which it obviously is when the device isn't powered - though they say only +/-1 mA at Vcc >4.9v - I suspect at 4.9-5v, it's a bit more forgiving than that but that's what the spec says. The AVR Dx-series parts allow a whopping 20mA through the protection diodes with no Vcc dependence! What a change, heh...)

Never leave a TX line connected to an unpowered device (as noted above, if it's protected with that resistor, it will only prevent proper reset, but if it's not, that's how you burn out pins) Leaving powered things connected to I/O pins is one of the most common ways that people on the arduino forums blow pins (blown pins on AVRs - classic ones, at least - almost always manifest as a pin being "stuck" high, low, or somewhere in the middle (because it's stuck both high and low; in this case the chip will also get hot because you have upwards of 50mA flowing through the stuck-on high side driver and the stuck-on low side driver. 

Akways take care that you're not applying external voltages to the I/O pins of your microcontroller board when it is not powered!

ldalep, 2021-02-20:
Much better put  :)

ArnieO, 2021-02-21:
Hi guys, you were (of course) right! 
After having disconnected the FTDI adapter I no longer see the issues I had with restart after power cycling. 

😦 Tto be honest this was a bit embarrassing, as I have enough background to know about the risk of putting a signal on an input pin of a device that is not powered; I fully understand it. 

**But _Thank You_ for reminding me!** 😄

ziggy90127, 2023-12-28:
Well I'm glad I read this page.  Originally I used a cheap USB serial adapter, I removed the LED on the RX line and added a wire to bring DTR out. It has a 1k5 resistor on TX.  It worked great, but 5v only.  I bought a Seeed "USB To Uart 5V&3V3" adapter because it has separate circuitry to drive the LED's, brings DTR out to the connector, and has a 3.3v 5V selector switch.  It worked fine with a schottky and a resistor for UPDI, and bootloader on TX/RX worked too. But then I played with powering the circuit with it's own power supply, and the TX on my Attiny3227 became stuck high.  So I tried another Attiny3227, same problem.  After reading this, I come to realize my swank Seeed adapter has no resistor on TX.  I guess I blew out TX on 2 Attiny's.  Time to order some more, and a 1k5 resistor. Live and learn.