SpenceKonde/megaTinyCore #331: A "Stitch in Time" 32Khz Ext Crystal RTC

ldalep, 2021-02-21:
My goal was to make a good RTC out of a 1614 32Khz Ext Crystal millis().  I am having trouble reading the  var=millis() and getting the correct value all the time.   The following code was ran over night and sampled every 60sec with coolTerm and written to a file with a ms date stamp coming from my mac.   The library is 2.2.7.    

```
// Date and time functions using just software, based on millis() & timer

//#include "RTClib.h"

unsigned long  t = 0;
#define _TCYCL (1000UL*1UL*60UL*1UL)
unsigned long _t1=0;

//RTC_Millis rtc;

void setup () {
     Serial.swap();
     Serial.begin(115200); 
     Serial.println ("Init");
} 

void loop () {

     while ((_t1 - t)  < _TCYCL){
      _t1=millis();  
     } 
     Serial.print(_t1,HEX); Serial.print ("\t\t");  
     Serial.print(t,HEX); Serial.print ("\t\t");
     Serial.print ((_t1 - t),HEX);Serial.print("\t\t");
     Serial.print (millis()); Serial.println("\t\t");
     t=_t1;
}
```

**The run was started at** 
 
[image]

finished at 
[image]

**The absolute accuracy was good   .5sec over ~9Hrs** 

The problem is at random (~30min) I do not get the correct value for millis().  (_t1) is the first column after the date stamp and the highlighted 1A1F800 is smaller than the previous (t) 1A2D2C0  causing the overflow given that it is negative.   I understand that I have to take care of the 1msec resolution on the millis( ) but don't understand why I get the wrong number?  I get the same random numbers using the internal RTC but get no errors when I use the default timers for millis/micro. I have tried 2 other chips (same lot).  Is there a sync issue when reading?

[image]
 more examples

[image]

SpenceKonde, 2021-02-21:
> but don't understand why I get the wrong number?

What I find really really weird is that over time, is works out.... but somehow, individual samples are off sometimes? Can you please tell me how the hell that could happen?
It sounds to me like there's some set of conditions under which millis can briefly experience "backwards time travel" - this is where if the call is made at just the right (wrong) time, you can end up with the result being back in time about one timer overflow - the overflow happens as gathering the data, and so it sees that timer that has just rolled over, but the interrupt has not yet fired to increment the overflow counter.  if backwards time travel could occur with millis (which your test will be fairly good at demonstrating if it's possible, because you're just hammering millis() - and the moment you go back in time, the negative result leads to the timer instantly meeting it's condition and printing out the data.. in the past it was a problem sometimes iwth micros - as you'd expect, since only micros use the current timer value, millis just use the overflow count - EXCEPT when you're using the RTC for millis timekeeping, when.... it... looks like we have about 1 count per millis (1024 Hz)... in millis we're doing the /1024*1000 math with clever bitshifting, (using the wrong one of the two options m=m- (m >>6) - (m >>7). m= m-(m>>5) + (m>>7) - the latter is more accurate. Next time I actually work with millis timekeeping (for that sleeplib) I've mentioned), I must remember to check on how good a job the compiler did at making that efficient. 

> no errors when I use the default timers for millis/micro.
Well, that's because about 9-10 months ago I went through all the millis timing options very meticulously and fixed them! If you'd done that before I went on my timer rampage, youd have seen bad behavior; I think TCD0 as millis was jumping back,and all of them, when at 20 MHz, weren't staying in sync. Unlike ATTinyCore, I actually do millis timing right - you set top to 254, so 255 counts, not 256. This has two benefits,: PWM from the timers, produces evenly distributed duty cycles when passed 1~254 (since 0 and 255 are off and full on no pwm, respectively. If you count to 255, for 256 counts, 1~255 would all generate pwm, but 255 is hardcoded to turn off PWM and write the pin high. So the duty cycle of analogWrite(pin,254) in a TOP=255 one is not 254/255, but 254/256. Otherwise known as 127/128. (and if you've got an active LOW led and an otherwise identical active HIGH led, you'd find that if you tried to pwm them at low duty cycles, you would get smoother PWM at low duty cycles with the active LOW. With proper TOP=254, it is symmetric. 

ATTinyCore (and every classic core I know of) is 256-based. megaTinyCore and DxCore are 255-based. 

ALSO it ends up fixing the math such that exact calculations are much easier to pull off and error from the math itself isn't an systematic issue, somewhere I have the spreadsheet where I found that there are just a few magic values like that.

SpenceKonde, 2021-02-21:
But I think I see a bug in the millis routine for that clock option - pity you just missed the 2.2.8 release

ldalep, 2021-02-21:
I know I just posted it when you put out 2.2.8 - I guess I could check for the value to be always monotonic increasing..... if not read again :)

SpenceKonde, 2021-02-21:
Yeah, Something that I think will probably fix it is checked into the 2.2.x branch now, since 2.3.0 dev needs a big merger anyway.

ldalep, 2021-02-22:
2 hours plus and not one burp..... Looks like you got it.  I will let it run overnight.  As an aside can you give me some idea on how to update patches.   I had to push the 2.2.x in place of the 2.2.8 in my /Users/ldalep/Library/Arduino15/packages/Arduino/hardware.  For some reason putting the directory in the sketch folder under hardware/ it does not find it.   Anyway most of the time I have the sketches in project directories not in their sketch folder.

SpenceKonde, 2021-02-22:
Literally, I started out typing that as an explanation of how yeah it's probably something like that but I couldn't guess at with the intent of trying to give you a nudge in the right direction so you could try to hunt down the bug..

It started with my quoting your "but I don't understand why I get the wrong number" and having typed "Well that makes two of us!"

And I initially had a feeling that it could end up being a and interrupt-safety issue. and as I dug around in there to grab facts to put into my response, it became clear what was actally going on was that we weren't catching the case where the rollover happens during the interrupt. I think it manifested when the the interrupt flag became set right as the user called millis, after millis checked for it but before they read RTC.CNT - hence, when they did, the count they read had rolled over, butr when they read that hadn't haopp0ened back when they checked for the OVF flag.... That's my theory at least..... A great deal iof wefier went into dealing with that kind of thing in micros, where any vulnerability there is much easier to trigger. because while here, the timer is counting one tick every ... 20,000 system clocks? In thatsituation, the timer counts one tick every 20 system clockso whereas in your situation, the it mighyt only trigger every 65536th timer tick, each of which lasted 1024 system clocks. Assuming the compiler didn't do something utterly moronic there and compile that <<16 into something other than a movw. (I am incredibly curuious about how the compiler renders that....There is an obvious intelligetnt way to do it, but I am concerned (just knowing the kind of stuff the compiler sometimes generates) that it might not take full advantage of it, and millis/micros are places where you'd really like the execution time to be reasonably snappy....

ldalep, 2021-02-22:
15 hours -solid ..... 11.2ppm I'm a happy camper.   Great write up and I followed what you did wire.c

closed 2021-02-22 by ldalep