SpenceKonde/megaTinyCore #335: Lost PWM on PA3 using Tiny 412 from 2.1.5 to 2.2.8

ldalep, 2021-02-23:
After downloading 2.2.8 into an old program that worked long time ago the PWM on PA3 quit.  Hardware still works because blink turns the pin on and off.   Do I have to go back and retro 2.1.5 to prove this or is there something obvious missing.  PWM works fine on PA2, PA1

SpenceKonde, 2021-02-24:
Yup. Actually, everyone lost PWM on half the TCA pins. Pure typo, 0x40 instead of 0x04. 5ae2da14d141df1b400bcbb52a35f0d4bf6f6186

closed 2021-02-24 by SpenceKonde

ldalep, 2021-02-24:
Darn an I just started to look at bit_mask when I decided to take a nap - maybe next time

SpenceKonde, 2021-02-24:
bit_mask < 0x40 for all pins with TCA0 PWM at that point in the code flow so that's a clue ;-) 

it's supposed to check if it's > 0x04, to separate the HCMP channels WO3, WO4, WO5, typically on bits 3, 4, and 5 of a port, since whether a channel is outputting PWM depends on the xCMPnEN bits in TCA0.SPLIT.CTRLB. (where x is H for 3~5, L for 0~2), and n is 0~2 (channel number modulo 3)

ldalep, 2021-02-25:
Are you sure about turnOffPWM - I can't turn off  PA3 - x01  is as low as I can go  - other PWM turn off.

SpenceKonde, 2021-02-25:
Are you using the version from the 2.2.x branch? I don't think i propagated
the fix to master since 2.2.x is the one I;m doing the release from -
master is only fit for 2-series development currently

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-02-25:
yeah I can't automatically merge changes to 2.2.x forward to master anymore
because I accidentally merged master into 2.2.x and had to revert it, so
now any pull gets the commits that turn the proto 2.3.0 back into 2.2.x.
and obviously we don;t want those. will have to manually fix that issue or
something :-(

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-02-25:
but not doing that now.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

ldalep, 2021-02-25:
Ok ... what I did was use 2.2.x - I 'm a bit confused how would I have been able to turn on PA3 at all if I hadn't been using the one you fixed

SpenceKonde, 2021-02-26:
Well, I just found yet another bug here o_o I am baffled as to how you could have made PWM come out of PA3 with the version I had checked in period. 

There were also 2 commits that brought these changes - first the analogWrite() fix, then the turnOffPWM fix, a few minutes after I mentioned "turnOffPWM" for the first time in this conversation.

SpenceKonde, 2021-02-26:
Oh, of course, PA3's PWM wasn't broken by the most recent thing I fixed, that was PA7's PWM

SpenceKonde, 2021-02-26:
Anyway, it's confirmed working now with 
```
const uint8_t pins[]={PIN_PA1,PIN_PA2,PIN_PA3,PIN_PA7};
void setup() {
  // put your setup code here, to run once:
  pinMode(PIN_PA1,OUTPUT);
  pinMode(PIN_PA2,OUTPUT);
  pinMode(PIN_PA3,OUTPUT);
  pinMode(PIN_PA7,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  static byte i = 0;
  static byte p = 0;
  i += 16;
  analogWrite(pins[p],i);
  delay(500);
  if (i == 0) { p++; }
  if (p > 3) {p = 0;}
}
```
Releasing 2.2.9 - thanks for reporting this one.

ldalep, 2021-02-27:
THANK YOU THANK YOU - I am back to where I started .....  Is there a donation place that I can help out.  I am a poor old Microwave test engineer retired so my contribution to bug fixing is not of the same level that you do.  I am now going to see if the external crystal working on the 1614 can be put it to sleep and wake it up with knowing how much elapsed time has happened  I will be home free on this project- Thanks again Dr.... serious need to donate something!