SpenceKonde/megaTinyCore #336: Suggestion about UPDI pin option

technoblogy, 2021-02-24:
Despite your warnings on the menu, I still think it's a bad idea even providing an option in the UI that will brick users' devices. Also, the menu option ends up quite unsightly:

**UPDI Pin, READ DOCS or pick UPDI!! (burn bootloader req'd): "UPDI (pick this unless you have an HV UPDI programmer"**

Couldn't this be handled by a **#define** that the user has to put in their program, or even an edit they have to make to one of the core files, which would still provide the option for expert users, but make it unlikely that beginners would accidentally choose it.

SpenceKonde, 2021-02-26:
I agree that the wording of that menu sucks, I hate it. 

No, it cannot be solved by a #define that users have to put in their program to confirm they've read the docs; it is only applied when "burn bootloader" is done, which does not attempt to compile the sketch. Furthermore, as far as I can tell, there is no way for the core to detect that the user has put a #define in their sketch. I'd give a lot for the ability to do that - it would give me a solution for a ton of UX problems that are insoluble without a way for the user to feed information to the preprocessor of the core and/or libraries during compile process. I've spent probably over 8 hours just explaining to people who want me to fix/improve those things that yes! I totally agree with you, the way it is sucks and I would love to do what you describe, but I can't, because it needs to be a preprocessor define, but the core and libraries don't see defines in the sketch!

Finding and modifying files installed by board manager is not an acceptable user experience; those files were clearly intended to never be modified in that location.

SpenceKonde, 2021-02-26:
There has got to be a better way to phrase it - but I know how to talk to embedded processors, not embodied humans. Their syntax is convoluted and there are so many versions of the interpreter in the wild....

technoblogy, 2021-02-26:
OK, I understand the problem. Here's my suggestion; change the menu options so there aren't any warnings on the safe option:

````text
menu.resetpin=UPDI Pin (burn bootloader req'd)

atxy4.menu.resetpin.UPDI=UPDI
atxy4.menu.resetpin.reset=Reset (DANGER - Bricks chip w/out HV UPDI programmer!)
atxy4.menu.resetpin.gpio=IO (DANGER - Bricks chip w/out HV UPDI programmer! Use PCHV)
````

SpenceKonde, 2021-03-03:
Sounds reasonable.

MCUdude, 2021-03-14:
Why does this even need to be an option in the IDE in the first place? Arduino IDE is definitely not meant to be used for batch programming, and a tinker/maker shouldn't really swap out the UPDI interface for a GPIO or reset line; a chip with more IO should be used instead.

I've decided to not let the user "tamper" with the reset pin in MiniCore. If they absolutely have to, they can use PlatformIO, which is a valid option for this core as well. @SpenceKonde you should embrace PlatformIO, It's so much better than Arduino IDE for larger projects, and it's _very_ convenient to be able to store all hardware configuration settings in platformio.ini. You can just open a project and click upload. No messing around in the tools menu! All "pro" settings like the reset pin, compile flags and more obscure fuse settings are only available through PlatformIO with my Arduino cores.

```ini
; Set UPDI pin to GPIO
[env:set_fuses]
; Upload protocol for used to set fuses
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
board_hardware.updipin = gpio ; Dangerous
```

technoblogy, 2021-03-14:
@MCUdude Personally I find the PlatformIO user interface really off-putting. I'm not familiar with Visual Studio Code, and I get visual overload from all the menus, buttons, icons, and panes. But perhaps I should persevere. Can you recommend a good Getting Started article (preferably not a video) for the Arduino on PlatformIO, preferably on Mac?

MCUdude, 2021-03-14:
@technoblogy PlatformIO is really just a command line tool that takes care of the toolchain part and all your project dependencies. You don't _have_ to use any IDE at all.
VSCode + PlatformIO thing is really just a plugin to VSCode that lets you use PlatformIO without having to interact with the command line.

[This guide](https://dronebotworkshop.com/platformio/) seems OK. I found it best to learn how to use PlatformIO by using VSCode, since most guides use VSCode.

If you want to use PlatformIO with/as an IDE, have a look at [this page](https://platformio.org/install/integration).
However, the CLI is _very_ easy and straight forward to use. What's important is that the platformio.ini file is set up correctly, and I provide a [platformio.ini template](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md#platformioini-template).

* To create a new project, create a folder, cd into it and run `pio project init`. You'll now have to create main.cpp inside the src folder, and make sure that `#include <Arduino.h>` is added.
* You can compile the sketch (main.cpp) by running `pio run` or `platformio run`.
* You can upload to the default target (defined in platformio.ini) by running `pio run --target upload`. If you want to upload for a specific target, you can use the environment flag: `pio run --target upload --environment Upload_UPDI` (note that -t can be used instead of --target, and -e can be used instead of --environment)

I've attached a simple PIO test project that's really just a blink sketch for the ATmega4809. if you want to play around a little. you can also have a look at the [MegaCoreX PlatformIO README](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md) if you want to know more about what can be done in the platformio.ini file to better fit your hardware/project.
[pio_test.zip](https://github.com/SpenceKonde/megaTinyCore/files/6137329/pio_test.zip)

I'll have to mention that I pretty much always use Arduino IDE to test and tweak simple sketches. However, for large and professional projects I always use PlatformIO. I used to use Atmel Studio before, but the fact that it can only be used on Windows, has "poor" Arduino support, is a closed source, and doesn't support other programmers than the official Atmel ones made me switch. 

Let me know if you have any questions!

closed 2021-03-29 by SpenceKonde