SpenceKonde/megaTinyCore #339: Yet another Sleep Question

ldalep, 2021-02-28:
I have a 1614 running with external crystal to generate 32Khz RTC.  All of this works and I then wanted to put the chip to sleep and wake it periodically some X time later.  I set up a test sketch as follows:   The loop timer for the millis() times out every 10 sec while the sleep timer times out every second.    Every thing works fine if I delay() after or even better if I  use the while (RTC.STATUS > 0);   after the sleep command.   Blinking LED at the correct sleep rate and outputs text every 10sec.   If I comment the while (RTC.STATUS > 0); out the  while ((millis() - t)  < _TCYCL)  loop exits  every 64000 msec not 10000 like it is suppose to.   The millis() delay is correct when exiting, just not while looping.  Can someone shed light to this Cowboy.....?  OBTW the chips current during standby and the LED not on is a respectable 20ua.

```
void loop () {

  while ((millis() - t)  < _TCYCL) {

    sleep_cpu();                                   //Sleep the device and wait for an interrupt to continue
    while (RTC.STATUS > 0);                /* Wait for all register to be synchronized */
    digitalWrite(PIN_PA7, CHANGE);
  }
```
```
// Date and time functions using just software, based on millis() & timer

#include "RTClib.h"

unsigned long  t = 0;
#define _TCYCL (1000UL*1UL*10UL*1UL)
//#define _TCYCL (100UL*1UL*1UL*1UL)
unsigned long _t1 = 0;
#include <avr/sleep.h>
#define DISABLE_PORTA(pin) (&PORTA.PIN0CTRL)[pin] = PORT_ISC_INPUT_DISABLE_gc;  //Disable on PAx pin
#define DISABLE_PORTB(pin) (&PORTB.PIN0CTRL)[pin] = PORT_ISC_INPUT_DISABLE_gc; //Disable on PBx pin

RTC_Millis rtc;

void RTC_init(void)
{
  /* Initialize RTC: */
  //RTC.CLKSEL = RTC_CLKSEL_TOSC32K_gc;   /* 32.768kHz Internal Ultra-Low-Power Oscillator (OSCULP32K) */
  //RTC.CLKSEL = RTC_CLKSEL_INT1K_gc;
  
  while (RTC.STATUS > 0);                 /* Wait for all register to be synchronized */

  RTC.PITINTCTRL = RTC_PI_bm;           /* PIT Interrupt: enabled */
  RTC.PITCTRLA = RTC_PERIOD_CYC32768_gc /* RTC Clock Cycles 16384, resulting in 32.768kHz/16384 = 2Hz */
                 //RTC.PITCTRLA = (0x0E<<3)
                 | RTC_PITEN_bm;                       /* Enable PIT counter: enabled */
}

ISR(RTC_PIT_vect)
{
  RTC.PITINTFLAGS = RTC_PI_bm;          /* Clear interrupt flag by writing '1' (required) */
}

void setup () {
  Serial.swap();
  Serial.begin(115200);
  Serial.println ("Init");

  // following line sets the RTC to the date & time this sketch was compiled
  rtc.begin(DateTime(F(__DATE__), F(__TIME__)));
  // This line sets the RTC with an explicit date & time, for example to set
  // January 21, 2014 at 3am you would call:
  // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
  
  for (uint8_t pin=0; pin < 8; pin++)
{
  DISABLE_PORTA(pin);
  DISABLE_PORTB(pin);
  
}

  RTC_init();                           /* Initialize the RTC timer */
  pinMode(PIN_PA7, OUTPUT);             /* Configure pin#7 as an output */
  //TCB1.CTRLA |= TCB_RUNSTDBY_bm;
  set_sleep_mode(SLEEP_MODE_STANDBY);  /* Set sleep mode to POWER DOWN mode */
  sleep_enable();

}

void loop () {

  while ((millis() - t)  < _TCYCL) {

    sleep_cpu();                                   //Sleep the device and wait for an interrupt to continue
    while (RTC.STATUS > 0);                /* Wait for all register to be synchronized */
    digitalWrite(PIN_PA7, CHANGE);
  }
  
  t = millis();
  Serial.print("OUT "); Serial.print (millis()); Serial.println("\t\t"); Serial.flush();

}
```

amadeuspzs, 2021-03-03:
There is a working example of sleep/wakeup in https://github.com/SpenceKonde/megaTinyCore/discussions/322, with current measurement down to 1.2uA

SpenceKonde, 2021-03-03:
My guess is that it's still syncing the settings you changes in init_RTC(). - and that you put the chip to sleep before it can sync the settings. (or I guess, "all the settings" ) so you're winding up with period not getting set. The process of syncing to the RTC is 2 RTC clock cycles,,,, which is what? like 1k system clocks? Something like that...

SpenceKonde, 2021-03-03:
Oh wait, you're doing this with a millis() from RTC setting in the core?

Oh! yeah it's not syncing anything, and you're being woken up by the millis interrupt instead :-P I think in the docs I have a sleep function to call that will put it back to sleep when it's the millis interrupt that woke it.

ldalep, 2021-03-04:
Seems to work with the sync - doesn't work if it is missing

SpenceKonde, 2021-03-05:
Right - I think that all makes sense. And if it weren't for the millis interrupt on the RTC itself that the core is waking up for,i if you didn't wait for the sync to finishbefore going to sleep, it would never wake up, because you would have gone to sleep without having turned on the PTC interrupt that would rake you. They could stand to make it clearer that you need to wait

ldalep, 2021-03-05:
So Doc... is that an acceptable way to do it....  It seems that the pacer for for the interrupt works fine wither I have the sync there are not - what happens is I get the negative time thing going with the reported Millis() always short that is the reason it never leaves the loop.  With the sync in place it gives me the correct millis() time.

SpenceKonde, 2021-03-11:
Yeah, there will hopefully be a better way to do this in future with SleepLib, but that hasn't been written.

closed 2021-03-11 by SpenceKonde