SpenceKonde/megaTinyCore #344: Compilation error due to -mshort-calls

DrMarkEy, 2021-03-03:
Sometimes when I try to compile my program, I get an error like:

```
C:\Users\<USER>\AppData\Local\Temp\ccjGNTgP.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_MySketch.ino.cpp.o.3071':
<artificial>:(.text.startup+0x82): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
collect2.exe: error: ld returned 1 exit status
exit status 1

```
After some searching I found, this seems to be caused by the -mshort-calls compiler option, as explained in:
https://stackoverflow.com/questions/8188849/avr-linker-error-relocation-truncated-to-fit

Unfortunately I have no idea how to fix this in the arduino IDE. Is this something that can be fixed in megatinycore or is it an arduino issue?

SpenceKonde, 2021-03-03:
This cannot be fixed by Arduino or megaTinyCore. It would be nice if the avrgcc deveopers could improve their error messaged under these conditions, though, as this is an awful way  to report the error to the user

On small-flash parts (I know it happens with 4k parts and 2k parts, unsure if 8k parts can do it, my prediction is that they  can't), it can give you this error, instead of a more useful description of the compile problem. **The *actual* problem is that your sketch is too big** to fit in the  parts' flash - but it isn't  able to get far enough in the process to give you that error, because it's dying in the middleof the process. Often if you scroll up you'll see an error that something exceeded the size of text. 

You can prove this also by switching to a larger flash chips the wthe same pincount, and and compiling; youwill notice that the compiled size is larger than the amount of  flash in the part you were compiling for

closed 2021-03-03 by SpenceKonde

DrMarkEy, 2021-03-03:
Thanks for the fast reply. Unfortunately I think there must be something more to it: Often when I compile my code for the attiny814 it runs without an error and tells me that about 80% flash memory is in use. Then I only edit one line of code (something like add or remove a Serial.print) and it won't compile any more, giving me this error. If I keep on making edits at other places it will work again at random. Meanwhile the size of the compiled code does only change by a few bytes at most...

Sojourneer, 2023-11-25:
Same here.  I'm at 72% of flash, and still I get this popping up seemingly randomly.