SpenceKonde/megaTinyCore #356: millis()/micros() disabled option doesn't seem to disable delay().

technoblogy, 2021-03-11:
Testing it with an ATtiny414, and LED on PA4 , and the following test program:
````text
void setup() {
  PORTA.DIRSET = PIN4_bm;            // Make LED on PA4 an output
}

void loop() {
  PORTA.IN = PIN4_bm;                // Toggle LED on
  delay(1000);                       // wait for a second
  PORTA.IN = PIN4_bm;                // Toggle LED off
  delay(1000);                       // wait for a second
}
````
the LED still blinks at 0.5Hz.

SpenceKonde, 2021-03-11:
Yes, glad to hear that! Please create an issue if it ever fails to to that!

This is correct and expected behavior. If millis timekeeping is disabled,
an implementation of delay which calls delayMicroseconds(1000) (which does
not depend on the millis timer) in a loop the specified number of times is
used instead.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

technoblogy, 2021-03-11:
I was just about to close my comment as I just realised that! Thanks, David

closed 2021-03-11 by technoblogy