SpenceKonde/megaTinyCore #366: looking for py3 in

a7hybnj2, 2021-03-21:
Arduino: 1.8.13 (Mac OS X), Board: "20-pin tinyAVR 0/1/2-series (ATtinyxxy6), No (saves space), ATtiny3216, 20 MHz Internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (pick this unless you have an HV UPDI programmer!), 8ms, Closer to 5v"

/Users/a7hybnj2/Documents/Arduino/hardware/megaTinyCore/megaavr/tools/python3/python3 -u /Users/a7hybnj2/Documents/Arduino/hardware/megaTinyCore/megaavr/tools/prog.py -t uart -u /dev/cu.usbserial-330 -d attiny3216 -v --fuses 0:0x00 1:0x00 2:0x02 4:0x00 5:0b11000101 6:0x04 7:0x00 8:0x00 -aerase 
java.io.IOException: Cannot run program "/Users/a7hybnj2/Documents/Arduino/hardware/megaTinyCore/megaavr/tools/python3/python3": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:375)
	at processing.app.Editor.lambda$handleBurnBootloader$61(Editor.java:2379)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
Error while burning bootloader.

a7hybnj2, 2021-03-21:
I was trying to upload using some libraries that I haven't used before on this hardware:
```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
```
and they weren't working so I decided to update my core. That's when I switched to the new programmer option and had this failure.

SpenceKonde, 2021-03-21:
When installed manually, additional steps are required to install the
python package required for the pyupdi-style upload tool. See the
ManualPython.md file for a description of this process.  When installed via
boards manager, this is all handled by boards manager.

Due to partly completed changes relating to support for the new 2-series
parts in the github latest version, I recommend using the latest board
manager release unless you are working with the new tinyAVR 2-series parts.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-03-21:
Here's the document if you haven't found it. https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/tools/ManualPython.md

Note that even without installing the python package the core uses. you can still use all programmer options except the pyupdi serial adapter one.

The issue you describe is entirely expected when doing a manual install, and is docmented, since you don't have board manager to go grab dependencies for you. Closing issue because this does not represent a defect in the core.

As I saiid above, (for the first time ever, lol) I cant recommend using the "github latest" version of the core right now because it is in the middle of a major feature update. Nothing has been changed which would improve library compatibility since the 2.2.9 release.

regarding the issue with those libraries, please feel free to create an issue with more information on the problems you are encountering with those; I've been trying to rally the troops to get more information on outstanding library compatibility issues so we can get those squared away.

closed 2021-03-21 by SpenceKonde

a7hybnj2, 2021-03-22:
Thanks! for the super quick reply. I did end up using the board manager(1st time) and everything worked. I do want to try some series 2 stuff and I have a tube of them sitting here ready but I can switch back and forth as needed. All the libs I linked above can be added to the "confirmed" category.