SpenceKonde/megaTinyCore #381: Power Down Sleep breaks Serial.print

markus-seidl, 2021-04-07:
I'm using the 412 and the code mentioned in the documentation to put together a small sleep + serial example:

```
#include <avr/sleep.h>

// SoftwareSerial mySerial(1, 0); // RX, TX

void RTC_init(void)
{
  /* Initialize RTC: */
  while (RTC.STATUS > 0)
  {
    ;                                   /* Wait for all register to be synchronized */
  }
  RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;    /* 32.768kHz Internal Ultra-Low-Power Oscillator (OSCULP32K) */

  RTC.PITINTCTRL = RTC_PI_bm;           /* PIT Interrupt: enabled */
}

ISR(RTC_PIT_vect)
{
  RTC.PITINTFLAGS = RTC_PI_bm;          /* Clear interrupt flag by writing '1' (required) */
}

void sleep(byte seconds) {
  RTC.PITCTRLA = RTC_PERIOD_CYC32768_gc /* RTC Clock Cycles 32768, resulting in 32.768kHz/32768 = 1Hz */
                 | RTC_PITEN_bm;        /* Enable PIT counter: enabled */
  
  for(byte i = 0; i < seconds; i++) {
    sleep_cpu();
  }
}

void setup() {
  RTC_init();
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);  // Set sleep mode to POWER DOWN mode
  sleep_enable();                       // Enable sleep mode, but not going to sleep yet
  
  // Open serial communications and wait for port to open:
  
}

void loop() { // run over and over
  Serial.println("Goodnight moon!");

  sleep(1);
}
```

However, this just produces garbled messages when connecting to the serial port. This could be intentional, as Power Down shuts down peripherals. But in my "production" sketch, the Wire library seems to work perfectly after the power down - the Serial doesn't.

Changing the code to 

```
void loop() { // run over and over
  Serial.begin(9600);
  Serial.println("Goodnight moon!");
  Serial.end();

  sleep(1);
}
```

works flawlessly.

I would expect that this "works", but if this is "intended" (= the world behaves differently as I imagine), maybe a sentence could be added here: https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/PowerSave.md#sleep-and-serial-ports stating that the Serial has to be explicitly reinitialised after a power down.

PS: Thanks for the awesome library, documentation. Except from that "issue" it worked flawlessly!

SpenceKonde, 2021-04-07:
No reinitialization should be required. 

does 
```
#include <avr/sleep.h>

// SoftwareSerial mySerial(1, 0); // RX, TX

void RTC_init(void)
{
  /* Initialize RTC: */
  while (RTC.STATUS > 0)
  {
    ;                                   /* Wait for all register to be synchronized */
  }
  RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;    /* 32.768kHz Internal Ultra-Low-Power Oscillator (OSCULP32K) */

  RTC.PITINTCTRL = RTC_PI_bm;           /* PIT Interrupt: enabled */
}

ISR(RTC_PIT_vect)
{
  RTC.PITINTFLAGS = RTC_PI_bm;          /* Clear interrupt flag by writing '1' (required) */
}

void sleep(byte seconds) {
  RTC.PITCTRLA = RTC_PERIOD_CYC32768_gc /* RTC Clock Cycles 32768, resulting in 32.768kHz/32768 = 1Hz */
                 | RTC_PITEN_bm;        /* Enable PIT counter: enabled */
  
  for(byte i = 0; i < seconds; i++) {
    sleep_cpu();
  }
}

void setup() {
  RTC_init();
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);  // Set sleep mode to POWER DOWN mode
  sleep_enable();                       // Enable sleep mode, but not going to sleep yet
  
  // Open serial communications and wait for port to open:
  
}

void loop() { // run over and over
  Serial.println("Goodnight moon!");
  Serial.flush(); // Wait until the serial data has finished transmitting 
  sleep(1);
}
```

work correctly?

SpenceKonde, 2021-04-07:
Slightly confused about your handling of the PIT - shouldn't you at some point turn it off? Or is the intent - oncei t has slept and woken at least once - for it to continually run and fire a do-nothing interrupt once per second?

markus-seidl, 2021-04-08:
In my first code snippet, I forgot to include the `Serial.begin(9600);` line in the setup routine (because of experimentation).

`Serial.flush();` works.

It seems that the flush is needed and somehow I missed it (PEBKAC Problem). What confused me was, that there is data printed to the Serial Monitor, but it's just garbled:

> 14:53:44.482 ->  ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+⸮ ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+Qȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+Rȡ⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮R⸮⸮⸮+R ⸮⸮닖+⸮⸮⸮j⸮⸮⸮⸮j ⸮⸮닖+⸮⸮R⸮⸮⸮

Thanks!

SpenceKonde, 2021-04-08:
Yeah, that is bizarre. I thought serial was supposed to finish sending the current character before going to sleep. But looking in the documentation, it doesn't actually say that quite as unambiguously as I thought it did...It talks about the context of receiving, but not sending

closed 2021-04-09 by SpenceKonde