SpenceKonde/megaTinyCore #382: Maybe a stupid question, about TCA timer in split mode  (attiny 814)

yuranikspb, 2021-04-07:
I watched example
```
  pinMode(PIN_PB0, OUTPUT); //PB0 - TCA0 WO0, pin7 on 14-pin parts
  pinMode(PIN_PA5, OUTPUT); //PA5 - TCA0 WO5, pin1 on 14-pin parts
  TCA0.SPLIT.CTRLB=TCA_SPLIT_LCMP0EN_bm|TCA_SPLIT_HCMP2EN_bm; //PWM on WO5, WO0
  TCA0.SPLIT.LPER=0xFF; // Count all the way down from 255 on WO0/WO1/WO2
  TCA0.SPLIT.HPER=200;  // Count down from only 200 on WO3/WO4/WO5
  TCA0.SPLIT.LCMP0=0x7F; //50% duty cycle
  TCA0.SPLIT.HCMP2=150;  //75% duty cycle
  TCA0.SPLIT.CTRLA=TCA_SPLIT_CLKSEL_DIV1_gc|TCA_SPLIT_ENABLE_bm; //enable the timer with prescaler of 1
```
and i watched data sheet ? but don't understand:
TCA0.SPLIT.HCMP2   channel to PA5  pin or (wo5)
TCA0.SPLIT.LCMP0   channel to PB0  pin or (wo0)

TCA0.SPLIT.HCMP1      channel to what pin ?
TCA0.SPLIT.HCMP0    channel to what pin ?
TCA0.SPLIT.LCMP2   channel to what pin ?
TCA0.SPLIT.LCMP1   channel to what pin ?

?

SpenceKonde, 2021-04-07:
Lcmp0-2 go to wo0-2, on pins pb0-2 Hcmp0-2 go to wo3-5, on pins
pa3-5.(mapping o 8-pin parts is different)
There is also an alternate pin for each of the pins that can be selected
with one of the portmux registers (6 bits, one per pin, so they can be
chosen individually  unlike larger parts where all the TCA channels move as
a group) though the core only configures it in the case of the 8 pin parts,
iirc.
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

yuranikspb, 2021-04-07:
"Lcmp0-2 go to wo0-2, on pins pb0-2 Hcmp0-2 go to wo3-5, on pins
pa3-5"
Thx.
But where this informationin datasheet.

yuranikspb, 2021-04-07:
I test my last example.Iit doesnt work! why ?

SpenceKonde, 2021-04-07:
That example looks correct to me, except that you don't enable the output on WO3:
 TCA0.SPLIT.CTRLB=TCA_SPLIT_LCMP0EN_bm|TCA_SPLIT_HCMP2EN_bm; //PWM on WO5, WO0

you want
 TCA0.SPLIT.CTRLB=TCA_SPLIT_HCMP0EN_bm;

Quick look at the 214/414/814 datasheet (DS40001912C, I think is the latest version), I see it on pg 185, Figure 20-13. Timer/Counter Block Diagram Split Mode. It's probably stated elsewhere too Atmel datasheets were never great about putting the information on what signal goes to which pin in the part of the datasheet that you are looking when you need it but I think they've gotten worse on that front since the 2016 apocalypse. On the other hand, everything else about the parts and docs has improved, so I'm not complaining. Obviously the most useful 1-2 pages of the datasheets now are always going to be "I/O Multiplexing and Considerations" table from a datasheet covering the largest-pincount part in a family - but it sounds like your question was "how do we know which register in split mode corresponds to which WO pin?" And I think that's just figure 13 from the TCA chapter. Some of the newer parts say it in Register Descrioption, Split Mode for the CTRLB register. 814 datasheet is missing that sentence thoug.

yuranikspb, 2021-04-08:
To change duty cycle on fly 
a must change TCA0.SPLIT.HCMP0 only ?
or i need stop timer and start.

SpenceKonde, 2021-04-08:
No need to stop and start the timer, just write the register. with TCA I think the only things you need to stop timer for are "hard reset" command, and... maybe turning split mode on or off? I think everything else you can leave it enabled for. 

TCD0 is the one you need to stop and restart to change almost anything on, TCA0 isn't weird like that.. Duty Cycle is about the only thing on TCD0 that you can change without stopping the timer.

yuranikspb, 2021-04-08:
Thx!
Please, dont close this issue some days , may bei will keep ask any question in this topic about TCA

closed 2021-04-08 by SpenceKonde