SpenceKonde/megaTinyCore #383: Document I2C buffer size

technoblogy, 2021-04-07:
Knowing the I2C buffer size is quite important for doing things like driving I2C OLED displays, and recently after wondering why I couldn't get a display to work I discovered that it's as small as 16 bytes on some parts, so I think it would be useful to have it documented in this section:

https://github.com/SpenceKonde/megaTinyCore#i2c-twi-support

SpenceKonde, 2021-04-07:
Will include in docs for 2.3.0 - which is coming *VERY* soon. 

It does look like the buffer sizes may be a bit small (on the third hand, for master TX, why does it even NEED a buffer? the I2C calls are blocking, so we should be able to just feed it arbitrary numbers of bytes.... but I that's deeper quicksand than I want to step into currently) - On the smallest RAM parts, yeah 16b seems totally reasonable. But we start with 16b even on parts with 512b of ram, which is probably large enough that 32b would be better. Do you know of any particular breakpoints with I2C buffer size? Like "Below xx bytes, yy doesn't work" where yy is a very popular library?

technoblogy, 2021-04-08:
> on the third hand, for master TX, why does it even NEED a buffer? the I2C calls are blocking, so we should be able to just feed it arbitrary numbers of bytes

Yes, I've written a bufferless I2C master library, but it works in a slightly different way to the Arduino one so it couldn't be used as a replacement:

[Minimal Tiny I2C Routines](http://www.technoblogy.com/show?24R7)

> Do you know of any particular breakpoints with I2C buffer size?

If we could rewrite history I would advocate making all the buffers 32 bytes, because that's easy to remember, and it's what the Arduino has; it's even in their documentation:

https://www.arduino.cc/en/reference/wire

> The Wire library implementation uses a 32 byte buffer, therefore any communication should be within this limit. Exceeding bytes in a single transmission will just be dropped.

However I think the only thing you can do now is leave them as they are, but document them.

SpenceKonde, 2021-04-08:
Oh interesting...

IMO as long as I am raising limits, I think that's acceptable. But I'm realizing there are some serious complications - first off, 2k parts cannot meaningfully fit Wire anyway (at least, they can't do anything , and besides that, they cannot fit a pair of 32-byte buffers. 32-bytes is stretching the bounds of sanity on the 4k parts too 25% of RAM for the I2C buffer? It's not hard to see why the smaller numbers were chosen. It looks to me like everything 512b and up has had 32 byte buffers since forever, and it was just the 128b and 256b RAM parts (2k and 4k flash 0/1-series) that fell under 32b. 32b on a 256 is kind of painful (total 64 bytes of buffer) but workable - it's not like one will be able to fit into the flash the sort of processing that could make use of large blocks of RAM anyway. Before I saw this I'd been planning to bump 256b ram parts to 24b, everything up until 2k to 32b, 2k/3k to 48 , and give 4k Dx-series parts a 66b buffer, with 130 for the 8k+ (the extra bytes were from remembering the common memory chips which can do writes of 64 or 128b at a time, but need 2 additional address bytes before the data) 

The official megaAVR core for the 6k ram 480x gets 128b buffers, though so having seen that, I wonder if I should set it to 128 (or 130) for all the AVR Dx parts with at least 4k of ram (so all except AVR16DD

Maybe it should be 32b for everything with less than 4k ram, 130b for everything with more, so there are only 2 buffer sizes (plus the  "desperation" implementation for 2k parts, which we don't care all that much about working well because one can't fit much of anything there!

technoblogy, 2021-04-09:
> IMO as long as I am raising limits, I think that's acceptable.

Not necessarily, as it may cause some users' programs to unexpectedly run out of memory.

> Maybe it should be 32b for everything with less than 4k ram, 130b for everything with more, so there are only 2 buffer sizes

Why 130 rather than 128 bytes? I always expect them to be a power of 2.

I'd be happy with 32 bytes for everything; easier to remember. But I still think that raising limits may upset some users.

SpenceKonde, 2021-04-11:
We shall see. Only one ram size of megaTinyCore part got its limit raised. And DxCore parts have so much ram that I am not worried about it over there.

SpenceKonde, 2021-04-11:
130 like I said to allow for full page writes to the common I2C eeprom chips that want 2 bytes of address followed by up to 128b of data for a page.

closed 2021-04-11 by SpenceKonde

technoblogy, 2021-04-11:
> 130 like I said to allow for full page writes to the common I2C eeprom chips that want 2 bytes of address followed by up to 128b of data for a page.

OK, got you,
David