SpenceKonde/megaTinyCore #385: Inconsistency between documentation of default timer (ATtiny_x24.md) and boards.txt

MiBiMiFlo, 2021-04-12:
Just found a small inconsistency between the documentation of new x24 parts and the boards.txt.

The documentation mentions the TCB1 as default timer for millis, but the boards.txt seems to define TCA0.

SpenceKonde, 2021-04-12:
Thanks for reporting. I'm so glad to see that people are actually reading my docs!

That's because the documentation has been committed to github but the change to boards.txt has not been. That will be done within the next few hours as I finish up the 2.3.0 release.

MiBiMiFlo, 2021-04-12:
Thanks for the response.
I think lot's of people read the documentation, possibly not everyone would open an issue if something looks wrong.

And bye the way, a big thank you for all the work you have put in this project!

SpenceKonde, 2021-04-12:
This is now corrected in the core, along with the other loose ends for 2.3.0

closed 2021-04-12 by SpenceKonde