SpenceKonde/megaTinyCore #386: Arduino IDE keyword highlighting

MCUdude, 2021-04-12:
Would you be interested in a list of keywords that would make the IDE color-highlight hardware names/registers:

[image]

I've created a _very_ crude awk script that runs through an IO file and adds the preferred keyword. You can see how the MegaCoreX keywords.txt file looks like [here](https://github.com/MCUdude/MegaCoreX/blob/master/megaavr/libraries/MegaCoreX/keywords.txt). If you want I can submit a list you can just copy into the megaTinyCore keywords.txt file.

The only downside is that you can't have a separate keywords.txt file for each target, so if we use the ATtiny3217's iotn3217.h file, we'll get "false" keyword highlighting for pins that don't exist on other targets.

SpenceKonde, 2021-04-12:
Oh shit, yes please!!! I was just thinking the other day about how useful something like that would be; 
The false keyword garbage happens no matter what - it;'s already a thing many places in my keyword lists.

Merging the output from 3227 and 3217 would get it all for megaTinyCore; DxCore would need one representative from each of the major families for *true* complete coverage, though you get 99% of the way there with a DA or DB (DD has a  few mux options for periphterals that the others don;t, D B has the opamops, and DA has the

I imagine at *some point* the list could long enough that it would start to impact performance (especiallty if ao ggomg was em

Also - how many different colors can you get from keywords? It's only 3 for stock Arduinio IDE right>?

MCUdude, 2021-04-12:
> Also - how many different colors can you get from keywords? It's only 3 for stock Arduinio IDE right>?

There's only three 🙁 

> Merging the output from 3227 and 3217 would get it all for megaTinyCore;

I've attached a file you can add to your keywords.txt. It contains all the 3217 and 3227 registers, but only once. There are a bunch of new lines in there because the script adds a new line for every line that starts with /*

[tiny_merged.txt](https://github.com/SpenceKonde/megaTinyCore/files/6298953/tiny_merged.txt)

MCUdude, 2021-04-12:
Here's the keywords.txt file for DxCore as well. I downloaded the latest AT pack and used ioavr128da64.h, ioavr128db64.h and ioavr64db32.h to create the file. This time I also ran it through `cat -s` to remove all the duplicated newlines in there.

[keywords_dxcore.txt](https://github.com/SpenceKonde/megaTinyCore/files/6299824/keywords_dxcore.txt)

Here's also a cleaned-up version for megaTinyCore where duplicated newlines are removed:

[keywords_megatinycore.txt](https://github.com/SpenceKonde/megaTinyCore/files/6299834/keywords_megatinycore.txt)

both files contain the content their "original" keywords.txt files contained. This means that you can simply replace the existing ones with these.

SpenceKonde, 2021-04-13:
Wow, thanks. It sort of blows my mind that having 4000 keywords doesn't produce obvious interface lag, heh... 

Anyway, I checked in a culled version for megaTinyCore (dropped names of bits within a multibit bitfields (unlike classic AVR, it is rare to use those because these don;t have dumbass register layouts with, say, CS in nonconsecutive bits scattered within 2 registers. ) the stuff from power.h (classic AVR only - a while back I duig through that library and remember coming to the conclusion that all the code that did anything was #ifdef'ed out on modern AVR, and all the EVSYS/CCL stuff, since we have libraries to do that for us, and some standard lib stuff I had never heard of before. Probably going to revisit this before next release. I think it's silly to define bit numbers within bitfields that hold numeric value. 

And I added SPLIT and SINGLE so if you use .'s as separators, the TCA registers will come out right.

SpenceKonde, 2021-04-19:
Thaaaaanks - this is pretty awesome. Makes a real difference when writing code that works with these. Am still worried about whether it will cause a slowdown on some systems, heh...

closed 2021-04-19 by SpenceKonde