SpenceKonde/megaTinyCore #388: Wrong azduiino4a toolchain path on windows

avlasic, 2021-04-12:
On Windows latest 2.3.0 release fails to compile with error:
`exec: "C:\\Users\\XXX\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4a/bin/avr-g++": file does not exist`

This is caused because latest toolchain has extra `avr` folder in it's path:
`C:\Users\XXX\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4a\avr\bin`
While old one was:
`C:\Users\XXX\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3\bin`

Probably  toolchain archive shouldn't have that extra folder in it.

SpenceKonde, 2021-04-12:
What the hell? This has been used successfully by DxCore 1.3.4... .unless that version is actually hosed and literally nobody has bothered to report this?!!!

SpenceKonde, 2021-04-12:
Goddamnit. yeah, either nobody has tried DxCore 1.3.3 OR 1.3.4 on windows, or they figured I knew about it and didn't report\

grandaspanna, 2021-04-13:
Just updated to 2.3.0 and hit this. What's the right way to patch this on my installation?

SpenceKonde, 2021-04-13:
Wait 30 minutes and install 2.3.1

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SpenceKonde, 2021-04-13:
2.3.1 is now available and appears to work when installed from board manager.

closed 2021-04-13 by SpenceKonde