SpenceKonde/megaTinyCore #394: 2.3.1 doesn't like ATtiny214

mechatroniks-git, 2021-04-17:
v2.9 compiled the a modified version of this [code](https://raw.githubusercontent.com/tetrabiodistributed/PAPRA-PCB/c38be3ee90f81fe9fc7fcec78938d70674fdf67d/firmware/papracode/papracode.ino) just fine.  The modifications didn't change the timer, just disabled serial debug output and some LED so code can fit in 214. (I'm testing a variety of series 0/1 ICs for completeness)

v2.3.1 throws this error
`
C:\Users\xx\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.3.1\cores\megatinycore\Tone.cpp:84:2: error: 'TCB1' was not declared in this scope
 &TCB1;
  ^~~~
C:\Users\xx\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.3.1\cores\megatinycore\Tone.cpp:84:2: note: suggested alternative: 'TCB0'
 &TCB1;
  ^~~~
  TCB0
exit status 1
Error compiling for board 14-pin tinyAVR 0/1/2-series (ATtinyxxy4).
`
[image]

mechatroniks-git, 2021-04-17:
same with the 204,404,804,1604,414,814,. 1614 & 2-series compile OK.

SpenceKonde, 2021-04-17:
Thanks, this is why we DESPERATELY NEED #306!!!!!

commit 2ccba933 (2021-04-18): Correct issue with TCB0 as millis breaking compile due to tone.cpp

Fixes #394. Even when tone wasn't used, it would generate compile errors that stopped the sketch from buiilding (this was not from an intentional attempt to stop compile - not sure why it didn't always do this,! But we just checked if millis was TCB0, if it was, assumed we have a TCB1, and then get undefined reference errors when we tried to refer to TCB1.
closed 2021-04-18 by SpenceKonde