SpenceKonde/megaTinyCore #411: 1614 with softwareserialwithhalfduplex issue

tinuk74, 2021-05-05:
Hi,

there seems to be a problem using the SoftwareSerialWithHalfDuplex library and the 1614 chip.  I have used the ATTinyCore and ATTiny814 with this library and it works great, but I need more memory so I moved to the 1614.  I get the following error when using the megaTinyCore and ATTiny1614;

Arduino: 1.8.13 (Windows 10), Board: "14-pin tinyAVR 0/1-series w/Optiboot (ATtinyxxy4), ATtiny1614, 8 MHz internal, Enabled (default timer), 1.8V (5 MHz or less), Disabled/Disabled, Closer to 3v, TX:5 (PB2), RX:4 (PB3), UPDI, Optiboot for 8 sec. after power-on & reset, 8ms"

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp: In member function 'void SoftwareSerialWithHalfDuplex::begin(long int)':

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:338:7: error: 'digitalPinToPCICR' was not declared in this scope

   if (digitalPinToPCICR(_receivePin)) {
       ^~~~~~~~~~~~~~~~~

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:338:7: note: suggested alternative: 'digitalPinToPort'

   if (digitalPinToPCICR(_receivePin)) {
       ^~~~~~~~~~~~~~~~~

       digitalPinToPort

In file included from c:\users\martin\appdata\local\arduino15\packages\dxcore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\io.h:99:0,

                 from c:\users\martin\appdata\local\arduino15\packages\dxcore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\interrupt.h:38,

                 from C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:55:

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:379:44: error: 'digitalPinToPCICRbit' was not declared in this scope

     *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                            ^
C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:379:44: note: suggested alternative: 'digitalPinToPort'

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:382:22: error: 'digitalPinToPCMSK' was not declared in this scope

     _pcint_maskreg = digitalPinToPCMSK(_receivePin);
                      ^~~~~~~~~~~~~~~~~

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:382:22: note: suggested alternative: 'digitalPinToPort'

     _pcint_maskreg = digitalPinToPCMSK(_receivePin);
                      ^~~~~~~~~~~~~~~~~

                      digitalPinToPort

In file included from c:\users\martin\appdata\local\arduino15\packages\dxcore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\io.h:99:0,

                 from c:\users\martin\appdata\local\arduino15\packages\dxcore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\interrupt.h:38,

                 from C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:55:

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:383:28: error: 'digitalPinToPCMSKbit' was not declared in this scope

     _pcint_maskvalue = _BV(digitalPinToPCMSKbit(_receivePin));
                            ^

C:\Users\Martin\Documents\Arduino\libraries\SoftwareSerialWithHalfDuplex-master\SoftwareSerialWithHalfDuplex.cpp:383:28: note: suggested alternative: 'digitalPinToPort'

exit status 1

Error compiling for board 14-pin tinyAVR 0/1-series w/Optiboot (ATtinyxxy4).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Modac, 2021-05-05:
The library will have to be ported to the "modern" AVRs.
Do you need SoftwareSerial or could you also use the Hardware Serial/USART peripheral, which has a Half-Duplex mode?

tinuk74, 2021-05-05:
I need the SoftwareSerialWithHalfDuplex library because I am using multiple instances on various pins.  Also the logic is inverted.  Is the port something I can do, or does the author need to do it?

tinuk74, 2021-05-05:
Additional info: SoftwareSerial alone does work, just the SoftwareSerialWithHalfDuplex fails.  I need the half duplex functions so can't use the standard SoftwareSerial.

Modac, 2021-05-05:
SoftwareSerial is already ported and included in this core, if I'm not mistaken. See [here](https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/libraries/SoftwareSerial). That's probably why SoftwareSerial does work.

If I remember correctly the SoftwareSerialWithHalfDuplex made very few changes to the original SoftwareSerial library. So instead of porting the whole library you could just "port the changes".

SpenceKonde, 2021-05-05:
The 841 and 814 are both 14-pin AVRs, with Vcc and Gnd on the same pins. They are very very different parts,. The 841 was the last (and best) of the classic tinyAVR product line - I think at that point, they knew the end times were at hand. Best classic tinyAVR - dual hardware serial ports, and peripherals that are in many respects better than most of the megaAVR line. 

The 814 is a modern AVR, the 8k version of the 1614. The 814 - that is actually a chip that I would recommend against a hobbyist using, because it's just pennies cheaper than the 1614, and the 1614 is one of the "good" 1-series, or 1+-series: The 814 is not - it's only got 1 ADC, 1 TCB and 1 analog compariator, and 512b ram. while the 1614 has 2k of ram, 3 AC's , 2 ADCs,. I love the 1614, you want to be using that, not 814's.. Or the new 1624. also excellent (and the 2-series isn't strictly better than 1-serties, unlike the 0 vs the 1. The 1-series has everything the 0-series has plus more. The 2-series lost some things and gained others. 

But yeahas others have said, libraries that are implemented at a low level need to be ported to the modern parts. Not all of them have, I've been trying to get lists together, but after initial excitement, not many people have been posting abuout libraries they are using: https://github.com/SpenceKonde/megaTinyCore/discussions/351 .SoftwareSerial is an official librarry that Arduino folk ported, though as I understand the porting was pretty minimal - it is written at a surprisingly high level for being a software serial libray... to my absolute shock when I read the code a few months ago... it's not inline assembly which is what you would normally expect for a software serial implementation. I have never thought that the SoftwareSerial library was fit for purpose... how many ports do you need? I take it 2 wasn't enough, since you were adding software ports onto a the 841 with 2 hardware serial ports .And more than3,  Are all these ports sending or receiving? I 2 hardware and 1 software serial would do it, I'd suggest grabbing some 1624's with the two serial ports

Floessie, 2021-05-06:
@SpenceKonde As an aside: On the 841 page you have code snippets for [disabling RX or TX](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x41.md#uart-serial-support). Would be nice to have such info on the 1614 page, too.