SpenceKonde/megaTinyCore #414: Cannot get PWM on both PC0 and PC1 of ATtiny3216 at the same time

nbremer, 2021-05-09:
Hi,

This might just be a case of me misunderstanding the docs, but I've attached and RGB LED to my ATtiny3216, using two digital legs, `PC0` and `PC1` for the green and blue colors, respectively.

However, I'm unable to get the _second_ color that I apply an `analogWrite` to, to behave properly to PWM (I mean the blue and green colors that are attached to the digital pins).

For example, if I run the code below, green and blue turn on fully for `color1`, they turn off for `color2`, but then blue turns on fully again for both `color3` and `color4` while the green color does respond properly:

```c
// RGB LED PINS
#define PIN_LED_RGB_R 9  //PB0
#define PIN_LED_RGB_G 10 //PC0
#define PIN_LED_RGB_B 11 //PC1

void displayColor(byte color[]) {
  analogWrite(PIN_LED_RGB_R, map(color[0], 0, 255, 255, 0));
  analogWrite(PIN_LED_RGB_G, map(color[1], 0, 255, 255, 0));
  analogWrite(PIN_LED_RGB_B, map(color[2], 0, 255, 255, 0));
}//void displayColor

void setup() {
  Serial.begin(9600);

  //RGB LED
  pinMode(PIN_LED_RGB_R, OUTPUT);
  pinMode(PIN_LED_RGB_B, OUTPUT);
  pinMode(PIN_LED_RGB_G, OUTPUT);
}//void setup

void loop() {
  int timing = millis();

  //For reference, when it's totally on
  byte color1[3] = {0, 255, 255};
  displayColor(color1);
  Serial.println("0,255,255");
  delay(2000);

  //For reference, when it should be totally off
  byte color2[3] = {0, 0, 0};
  displayColor(color2);
  Serial.println("0,0,0");
  delay(2000);

  //Blue should be almost off, but isn't, it's completely on
  byte color3[3] = {0, 0, 1};
  displayColor(color3);
  Serial.println("0,0,1");
  delay(2000);

  //Add a smidge of green, blue is still completely on
  byte color4[3] = {0, 1, 1};
  displayColor(color4);
  Serial.println("0,1,1");
  delay(2000);
}//void loop
``` 

If I switch the order of the `displayColor` function and do the `analogWrite` on the blue color _before_ the green color, the results are reversed; the green LED/color doesn't respond to PWM anymore and the blue color does behave properly.

Reading the documentation I understood that both the `PC0` and `PC1` should be able to do PWM (while also using `millis()`, which my full program needs for "blinks" without delay), or did I get that wrong?

Thank you for any help! :)

SpenceKonde, 2021-05-09:
Yeah it sounds like the analogWrite logic for taking 0 as arguments without
taking them off the timers is not quite working. Will investigate

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-05-10:
Didn't have a chance to investigate today - though I finished a new version of the UPDI -via-serial adapter tool, finally - It'll be in the next version. I did some performance optimizations to improve upload speed, so it's like.... 8-20 times faster now? It totally crushes jtag2updi now/ 

This issue is s top priority, I consider it to be a release blocker

Probably also impacts DxCore....

SpenceKonde, 2021-05-11:
Ugh, no it's worse than that... In general it is... all fucked up, writing values to wrong register and in some cases it looks like it writes wrong values too, which makes no sense at all...

SpenceKonde, 2021-05-11:
oh my.... the problem is that writes in rapid succession to those two registers, the second write is ignored., Somehiow it's falling afoul of the syncronization protection on the TCD registers, but why now? It never did before...

SpenceKonde, 2021-05-11:
Fix will be in 2.3.2 and is in github,

closed 2021-05-11 by SpenceKonde

nbremer, 2021-05-11:
Thank you for looking into this so incredibly quickly, and also solving it within a day :D 
Looking forward to the release (and _definitely_ also that amazing increase in the UPDI upload speed!!)