SpenceKonde/megaTinyCore #415: Confused about SRAM and Flash usage between ATtiny13a and ATtiny212

foxabilo, 2021-05-09:
The following code 
```void setup() {
  Serial.begin(9600);
}
void loop() {
  Serial.print('A');
}```
when built for the ATtiny13a uses 0 bytes of SRAM and 88 bytes of Flash
```    RAM:   [          ]   0.0% (used 0 bytes from 64 bytes)
    Flash: [=         ]   8.6% (used 88 bytes from 1024 bytes)```
when building for the ATtiny212 which has a hardware UART so one would assume the code would be somewhat smaller uses 69 bytes of SRAM and 1754 bytes of Flash. 
```    RAM:   [=====     ]  53.9% (used 69 bytes from 128 bytes)
    Flash: [========= ]  85.6% (used 1754 bytes from 2048 bytes)``

Simple things like digitalWrite are taking 90 to 100+bytes each, EEPROM.put is again 90+ bytes a 20 line program to read EEPROM and ADC values is over 2.5k in size when on the ATtiny13a it's 706 bytes.

Am I missing some compiler directive or build option?

maxgerhardt, 2021-05-09:
Addendum: PlatformIO currently uses the 2.1.5 version of this core, the latest version 2.3.1 doesn't compile at all

>avr/bin/ld.exe: region `text’ overflowed by 124 bytes

SpenceKonde, 2021-05-10:
I cannot reproduce your finding of more flash usage on the current version of megaTinyCore -  I see 1660b of flash with current development version and 69 bytes sram.. I think 2.1.5 was from before I rewrote large portions of the serial code in part to shave a few hundred bytes from the compiled size. so I don't know why it's coming out larger for you! I'm no expert on platform IO so I can't comment on that - it was @MCUdude who made this work with platform IO. 

And the reason that Serial on this core takes so much more flash ( is that this is a full feature serial implementation.  that is fully compatible with the one on official Arduino boards. A few versions ago I put a considerable amount of effort into reducing flash usage of the Serial functionality itself, but it is still a huge flash hog, because the virtual functions get compiled in always, with no attempt made to determine if the function is ever actually called. Even with LTO. It's stupid AF. The serial you were iusing on the tiny13 was likely a very tightly optimized one designed only for space and which would have supported only the most basic functions. Even after I worked over some parts of Serial, I am fully aware there remains optimization and cleanup that could be done there. The virtual function shit is incredibly annoying and I have no idea how to "fix" this so the methods don't need to be declared virtual. 

The reason it is using ram when the t13's serial doesn't is that on the tiny13 that is a blocking half-duplex software serial implementation. Hence it needs neither a transmit buffer (execution of other code just stops until it's fully sent instead of sending in the background). And if you don;t set it up to receive, I don't think it pulls in anything related to that. as I said above, I am incredibly frustrated by this, I don;t understand why those member functions have to be virtual and what, if anything, I've tried to look it up to understand it but none of the discussions seem to get to the level I need of "Okasy, so that's what a virtual method means. I have a class that has virtual methods because I maintain a library someone in the past wrote. It runs on microcontrollers with as little as 2k of program space, so it is imperative that optimizations like not including unused functions happen. What do I need to do to so that these don't need to be virtual?"  And I still despite hours of researching this intensely frustrating subject, do not have a crisp understanding of what 'virtual' qualifier is needed for - like I see examples of the bad behavior it prevents, but I can't understand why that behavior would happen without it....  I like C, and I like inline assembly I am not so good with C++

 I do wonder if it is possible for me to, on the 0/1-series parts with only one USART, at least. use #ifdefs to substitute in an alternate more efficient implementation that didn;t need to support multiple instances of the USART. The DRE interrupt in particular isagonizing to look at:
```asm
<snip>\megatinycore/UART0.cpp:48
#else
#error "Don't know what the Data Received interrupt vector is called for Serial"
#endif

#if defined(HWSERIAL0_DRE_VECTOR)
ISR(HWSERIAL0_DRE_VECTOR) {
 554: 1f 92         push  r1
 556: 0f 92         push  r0
 558: 0f b6         in  r0, 0x3f  ; 63
 55a: 0f 92         push  r0
 55c: 11 24         eor r1, r1
 55e: 2f 93         push  r18
 560: 3f 93         push  r19
 562: 4f 93         push  r20
 564: 5f 93         push  r21
 566: 6f 93         push  r22
 568: 7f 93         push  r23
 56a: 8f 93         push  r24
 56c: 9f 93         push  r25
 56e: af 93         push  r26
 570: bf 93         push  r27
 572: ef 93         push  r30
 574: ff 93         push  r31
<snip>\megatinycore/UART0.cpp:49
  Serial._tx_data_empty_irq();
 576: 8f e9         ldi r24, 0x9F ; 159
 578: 98 e3         ldi r25, 0x38 ; 56
 57a: 0e 94 b2 00   call  0x164 ; 0x164 <UartClass::_tx_data_empty_irq()>
<snip>\megatinycore/UART0.cpp:50
}
 57e: ff 91         pop r31
 580: ef 91         pop r30
 582: bf 91         pop r27
 584: af 91         pop r26
 586: 9f 91         pop r25
 588: 8f 91         pop r24
 58a: 7f 91         pop r23
 58c: 6f 91         pop r22
 58e: 5f 91         pop r21
 590: 4f 91         pop r20
 592: 3f 91         pop r19
 594: 2f 91         pop r18
 596: 0f 90         pop r0
 598: 0f be         out 0x3f, r0  ; 63
 59a: 0f 90         pop r0
 59c: 1f 90         pop r1
 59e: 18 95         reti

000005a0 <__vector_17>:
__vector_17():
<snip>\megatinycore/UART0.cpp:40
// first place.

#if defined(HAVE_HWSERIAL0)

#if defined(HWSERIAL0_RXC_VECTOR)
ISR(HWSERIAL0_RXC_VECTOR) {
 5a0: 1f 92         push  r1
 5a2: 0f 92         push  r0
 5a4: 0f b6         in  r0, 0x3f  ; 63
 5a6: 0f 92         push  r0
[... 112 lines cut ...]
 1a8: 98 13         cpse  r25, r24
 1aa: e8 cf         rjmp  .-48      ; 0x17c <UartClass::_tx_data_empty_irq()+0x18>
<snip>\megatinycore/UART.cpp:120
    // Buffer empty, so disable "data register empty" interrupt
    (*_hwserial_module).CTRLA &= (~USART_DREIE_bm);
 1ac: 04 84         ldd r0, Z+12  ; 0x0c
 1ae: f5 85         ldd r31, Z+13 ; 0x0d
 1b0: e0 2d         mov r30, r0
 1b2: 85 81         ldd r24, Z+5  ; 0x05
 1b4: 8f 7d         andi  r24, 0xDF ; 223
 1b6: 85 83         std Z+5, r24  ; 0x05
 1b8: e1 cf         rjmp  .-62      ; 0x17c <UartClass::_tx_data_empty_irq()+0x18>
```
[... cut ...]
My closing thought - do keep in mind you are using the supported part with the very least flash. Every time I get a chance to, I tell people that the user experience of working with a 2k part in Arduino when the core hasn't been very aggressively tweaked for that specific part - is pretty lousy.. Most megaTinyCore users are using 16k and 32k parts, and wouldn't stand for my removing all normal serial functionality. While I try to keep a lid on flash usage, and optimize more than Arduino does, as a matter of policy, I do not bend over backwards (or forwards) to accomodate the bottom of the barrel parts.

maxgerhardt, 2021-05-10:
>I cannot reproduce your finding of more flash usage on the current version of megaTinyCore - I see 1660b of flash with current development version and 69 bytes sram.. 

Are you sure? The 124 bytes overflow is the result I get when I use the Arduino IDE with the latest version, installed as per README.

[image]

[image]

SpenceKonde, 2021-05-10:
Oh! With optiboot, yeah that would do it lol. We do not recommend using optiboot on any parts with less than 8k of flash, and I should probably mark it as not recommended for 8k parts too...  . At 2k you're giving up a quarter of your precious flash on chip that already is too small to use comfortably with arduino... just in order to program with a serial adapter on the serial pins,, instead of programming with.... a serial adapter and a schottky diode (more reliable than the 4.7k resistor method) on the UPDI pins?. Don't use optiboot on a 2k part. 

The only reason we support the bootloader on all 2k parts is because it doesn;t require any extra binaries to be built, and when Bill Westfield did his initial port of Optiboot_x, that's what *he* decided to support. (he didn't notice, either, that the bootloader binaries were identical except for the 8-pin ones with default serial pin; he noticed they were the same for all sizes, but it wasn't until a few months ago that I realized they were identical for the different pincounts other than 8 too., With alt serial pins, all pincounts have the same binary; 3217 uses the same bootloader hex file as the 212!

foxabilo, 2021-05-10:
Agreed.

closed 2021-05-10 by SpenceKonde