SpenceKonde/megaTinyCore #421: Compile issue: exec: "C:\\Users\\...\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist"

TamojitSaha, 2021-05-12:
Hello mates,
I installed the latest **megaTinyCore** and while compiling for Attiny1614 I got the following issue:
```
exec: "C:\\Users\\...\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist
Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.
```
I even installed the **DxCore** but the issue persists.
Any help regarding solving the issue is appreciated.
Thanks in advance!

SpenceKonde, 2021-05-12:
Go back to previous version until 8 figure out why this is happening

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-05-12:
Reinstalling 2.3.2 should fix it. Template I made the board manager entry from was pointed to the defective version of the toolchain instead of the good one.

closed 2021-05-12 by SpenceKonde