SpenceKonde/megaTinyCore #434: OneWire not working.

ankur608, 2021-05-17:
Tested Dallas and Paul's OneWire library on ATtiny3217. Unable to fetch DS18b20 temperature sensor address.

DanSkareda, 2021-05-17:
Try this library: https://github.com/pstolarz/OneWireNg

ankur608, 2021-05-17:
Hey thanks for your reply. Meanwhile, I had tested the codes for ATtiny3217
and 804. Ai'nt compiling for megaAVR.

#include <new> in DSTherm.h:16:10: fatal error.

Though compiles for old AVR and RISC.

-Regards.

SpenceKonde, 2021-05-17:
Please link to code and libraries in question. Paul's version of the library won't work, but my fork of his library should (I submitted a PR to him which fixes it for all current and future modern AVRs 9 months ago, he has shown no interest in merging or even responding to say he won't merge it.

ankur608, 2021-05-17:
Tried OneWireNg library (ATtiny3217) https://github.com/pstolarz/OneWireNg

#include <new> in DSTherm.h:16:10: fatal error

SpenceKonde, 2021-05-17:
Where is the code you are trying to compile? Where did DSTherm.h come from? Where did you get that? I can't help without all the information I need to reproduce the failure.

ankur608, 2021-05-17:
[https://github.com/pstolarz/OneWireNg/tree/master/examples/DallasTemperature](url)

/*
 * Copyright (c) 2019-2021 Piotr Stolarz
 * OneWireNg: Ardiono 1-wire service library
 *
 * Distributed under the 2-clause BSD License (the License)
 * see accompanying file LICENSE for details.
 *
 * This software is distributed WITHOUT ANY WARRANTY; without even the
 * implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the License for more information.
 */
/**
 * Dallas family thermometers access example.
 */
#include "OneWireNg_CurrentPlatform.h"
#include "drivers/DSTherm.h"
#define OW_PIN          10
/*
 * Set to true for parasitically powered sensors.
 */
#define PARASITE_POWER  false
/*
 * Uncomment for power provided by a switching
 * transistor and controlled by this pin.
 */
//#define PWR_CTRL_PIN    9
static OneWireNg *ow = NULL;
static DSTherm *dsth = NULL;
/* returns false if not supported */
static bool printId(const OneWireNg::Id& id)
{
    const char *name = DSTherm::getFamilyName(id);

    Serial.print(id[0], HEX);
    for (size_t i=1; i < sizeof(OneWireNg::Id); i++) {
        Serial.print(':');
        Serial.print(id[i], HEX);
    }
    if (name) {
        Serial.print(" -> ");
        Serial.print(name);
    }
    Serial.println();

    return (name != NULL);
}

static void printScratchpad(DSTherm::Scratchpad *scrpd)
{
    const uint8_t *scrpd_raw = scrpd->getRaw();

    Serial.print("  Scratchpad:");
    for (size_t i = 0; i < DSTherm::Scratchpad::LENGTH; i++) {
        Serial.print(!i ? ' ' : ':');
        Serial.print(scrpd_raw[i], HEX);
    }

    Serial.print("; Th:");
    Serial.print(scrpd->getTh());

    Serial.print("; Tl:");
    Serial.print(scrpd->getTl());

    Serial.print("; Resolution:");
    Serial.print(9 + (int)(scrpd->getResolution() - DSTherm::RES_9_BIT));

    long temp = scrpd->getTemp();
    Serial.print("; Temp:");
    if (temp < 0) {
        temp = -temp;
        Serial.print('-');
    }
    Serial.print(temp / 1000);
    Serial.print('.');
    Serial.print(temp % 1000);
    Serial.print(" C");

    Serial.println();
}

void setup()
{
#ifdef PWR_CTRL_PIN
    ow = new OneWireNg_CurrentPlatform(OW_PIN, PWR_CTRL_PIN, false);
#else
    ow = new OneWireNg_CurrentPlatform(OW_PIN, false);
#endif
    dsth = new DSTherm(*ow);

    delay(500);
    Serial.begin(115200);

#if (CONFIG_MAX_SRCH_FILTERS > 0)
    dsth->filterSupportedSlaves();
#endif

    /*
     * Uncomment to set common configuration for all sensors connected to
     * the bus:
     * - Th = 0, Tl = 0 (high/low alarm triggers),
     * - Resolution: 12-bits.
     */
    //dsth->writeScratchpadAll(0, 0, DSTherm::RES_12_BIT);

    /*
     * The configuration above is stored in volatile sensors scratchpad
     * memory and will be lost after power unplug. Uncomment this line to
     * store the configuration permanently in sensors EEPROM.
     *
     * NOTE: It will affect all sensors connected to the bus!
     */
    //dsth->copyScratchpadAll(PARASITE_POWER);
}

void loop()
{
    OneWireNg::Id id;
    OneWireNg::ErrorCode ec;
    MAKE_SCRATCHPAD(scrpd);

    /* convert temperature on all sensors connected... */
    dsth->convertTempAll(DSTherm::SCAN_BUS, PARASITE_POWER);

    /* ...and read them one-by-one */
    ow->searchReset();
    do
    {
        ec = ow->search(id);
        if (!(ec == OneWireNg::EC_MORE || ec == OneWireNg::EC_DONE))
            break;

        if (!printId(id))
            continue;

        if (dsth->readScratchpad(id, scrpd) != OneWireNg::EC_SUCCESS) {
            Serial.println("  Invalid CRC!");
            continue;
        }

        printScratchpad(scrpd);
    } while (ec == OneWireNg::EC_MORE);

    Serial.println("----------");
    delay(1000);
}

ankur608, 2021-05-17:
Yeah, the Paul's DS18x20_Temperature code works, after applying your patch `https://github.com/PaulStoffregen/OneWire/pull/94/files#diff-cb6329c1edfc0ea335a7ecac38d936cf9872743cb9fd676de71606d74d773517`

Thanks for your kind support.

closed 2021-05-17 by ankur608

SpenceKonde, 2021-05-17:
Yup, no problem. Glad it's working.