SpenceKonde/megaTinyCore #437: In platform.txt tools.avrdude.program.pattern doesn't program all fuses

rl1131, 2021-05-25:
Currently not all fuses are programmed as expected using avrdude/jtag2updi.  I was trying to get the reset/updi pin to function as a reset pin, but when this option is selected in the IDE it would not program the fuses.  Upon closer inspection of the detailed log file only fuses 2,6,8 are being programmed.  By adding "-Ufuse5:w:{bootloader.SYSCFG0}:m" to the tools.avrdude.program.pattern line I was able to get this functionality working.

This may have been left out for a reason, but at least for my purposes, this is needed.

With that said, THANK YOU for all of the work you have put into the ATTiny support!

SpenceKonde, 2021-05-25:
As described in the documentation "upload"/"program" only sets absolutely safe fuses. Only "burn bootloader" sets "unsafe" fuses; obviously the fuse that can brick your chip unless you have an exotic HV programmer counts as "unsafe" 

In fact. the only context in which the core provides support for updi fused as reset is the case where the Optiboot bootloader is being installed. Because otherwise the chip would be unprogramable without exotic hardware, and suddenly the number one tech support issue I would get would be from people who unwittingly bricked their chip, don;t have an HV programmer, and either blame me, or walk away from the product line (and my core) in disgust. 

Actially, holy crap man... that is NOT supposed to bne an option for the non-bootloader configurations. I will adapt that option immediately so that it is only valid for the alt reset pin on the 2-series parts. 

The intended workflow if you need hardware reset (ersatz reset won't do irt? Even if you elevate it's priority?), aren't using a 2-series with alt reset pin) would be to upload using the IDE and set the fuse that rendered it unprogrammable manually (same as resetdisable on classic ATTinyCore. Now, when HV UPDI programmers become available, I may change my position, I will trip up too many newbies leaving a "shoot self in foot" option like that accessible. I am horrified that that made it into the release, It is not supposed to be there.

rl1131, 2021-05-26:
No worries... I can understand the support problems that would come with allowing the part to be effectively bricked.  It was because the option was available in the menu that I thought it should work.  For my purposes I'll just "hack" the platform.txt file. to get the results I need.

Cheers!

SpenceKonde, 2021-05-30:
Yup - if you search for "expert" in the boards.txt,. it's got the lines for it commented out.

closed 2021-05-30 by SpenceKonde