SpenceKonde/megaTinyCore #441: Feature Request:  serial port one-wire half-duplex mode support

EclipseBoom, 2021-05-29:
Please add this as a Serial.begin option like Serial.begin(115200, SERIAL_HALF_DUPLEX);

Document whether transmitted data is loaded into the receive buffer and must be dumped by the application or if that is handled by the driver.

I recommend leaving the port in receive mode until a transmission is requested, then change to transmit mode and return to receive mode when done.

Thanks for all your hard work!

SpenceKonde, 2021-05-30:
Sounds useful. 

Could also have option to toggle the XDIR pin if the half duplex link is RS485.

Modac, 2021-05-31:
I did get half-duplex serial working on the Attiny1614 with the following code. Maybe it's helpful.

```c++
void initOneWireUART()
{
    uint8_t oldSREG = SREG;
    cli();

    // Reset pin modes to default state (INPUT)
    // TODO: Test for swapped pins and set right ones
    pinMode(PIN_HWSERIAL0_RX, INPUT);
    pinMode(PIN_HWSERIAL0_TX, INPUT);

    // Disable RX and TX for config change
    // TODO: Maybe put before pin mode
    USART0.CTRLB &= ~(USART_RXEN_bm | USART_TXEN_bm);

    // Enable internal pullup
    // TODO: change to pinMode call and consider swapp pins, probably move to other pin mode calls
    PORTB.PIN2CTRL = PORT_PULLUPEN_bm;

    // RX Already set as INPUT_PULLUP, just set RX as OUTPUT again?
    /*
    if (PORTMUX.CTRLB & PORTMUX_USART0_bm == PORTMUX_USART0_DEFAULT_gc)
        pinMode(PIN_HWSERIAL0_RX, OUTPUT);
    else
        pinMode(PIN_HWSERIAL0_RX_PINSWAP_1, OUTPUT);
*/
    
    // Set loop back mode on RXD and TXD
    USART0.CTRLA |= USART_LBME_bm;
    // Enable open drain and RX an TX again
    USART0.CTRLB |= USART_ODME_bm | USART_RXEN_bm | USART_TXEN_bm;

    // Restore SREG content
    SREG = oldSREG;
}

void disableReceiver(void)
{
    USART0.CTRLB &= ~USART_RXEN_bm;
}

void enableReceiver(void)
{
    Serial.flush();
    USART0.CTRLB |= USART_RXEN_bm;
}
```

`initOneWireUART()` is called after `Serial.begin()` and the Reciever is disabled and enabled for sending with the corresponding functions.

I think a few lines in the init function can be omitted if the half-duplex mode is set up inside of `begin()`.

The following two resources were pretty helpful:
[USART in One-Wire Mode](http://ww1.microchip.com/downloads/en/Appnotes/AN2658-USART-In-One-Wire-Mode-00002658C.pdf)
[Attiny1614 One-Wire Errata Datasheet Errors](https://www.avrfreaks.net/forum/attiny1614-one-wire-errata-datasheet-errors)

EclipseBoom, 2021-06-10:
This works for me on ATTiny1626.  Thanks!

SpenceKonde, 2021-11-21:
If you have access to DA/DB-hardware, want to give it a try with what I just checked in on DxCore?

SpenceKonde, 2021-12-01:
Ugh, I've got the code written for megaTinyCore, but it's not ready yet. 
It gained 170 bytes compiled size for hello world - that's just not okay. 

No big juicy obvious source of the bloat either - seems like I get nickel and dimed to death...

SpenceKonde, 2021-12-02:
Size is better, but I did something wrong somewhere, because the parts are hanging as soon as I enable serial.

SpenceKonde, 2021-12-06:
Done and in

closed 2021-12-06 by SpenceKonde