SpenceKonde/megaTinyCore #443: my old sketch not wotk with new core 2.3.2

yuranikspb, 2021-06-01:
'byte pinConfig' redeclared as different kind of symbol
 byte pinConfig=3;
\megaTinyCore\hardware\megaavr\2.3.2\cores\megatinycore/api/Common.h:109:9: note: previous declaration 'void pinConfig(pin_size_t, uint16_t)'
 void    pinConfig(       pin_size_t pinNumber, uint16_t mode);
         ^~~~~~~~~

I change my variable name and it compile good.

 pinConfig note private ?

SpenceKonde, 2021-06-01:
I apologize for the failure to document the new function - the declaration was mistakenly left in after I gave up on my first attempt at implementing so I could do a release. 

I already decided to change the name of that function to pinConfigure() when I actually implemented it (done in DxCore but not ported here yet) because I had i realized that that pinConfig() was used by some other widely used thing. The function (well, will be) a wrapper around *ALL* of the configuration options for a specific pin. 

```
pinConfigure(5,(PIN_DIR_INPUT | PIN_OUT_LOW | PIN_PULLUP | PIN_INVERT_ON)); 
// pin is input, output value is pullup is on, and values are inverted
pinConfigure(5,PIN_DIR_OUTPUT);
// pin set output, everything else unchanged 
```
it isn't fast or efficient, but it provides a way any of the hardware's pin configuration functionality, on an Arduino pin number, without having to look up bit and port (either by hand or macros like digitalPinToPort, digitalPinToBitMask and everything in that family), and the second argument (if generated as I show, by bitwise ORing of the core-provided constants) is a 2 uint16_t - 2 bytes, so once you pay the price in flash for pinConfigure once, it's more flash efficient than using multiple calls to other functions and/or macros for stuff that function calls don't exist for

One of the things on my list is a master list of nonstandard functions and guidance on what kind of constants I'll use for internal and/or use-facing constant macros to prevent collisions.
```
PIN_* 
Names beginning with PIN_ are used to define a constant or constant-like value relating to I/O pins, these may be pin numbers, or constants relating to pins. Ones which are expected to be useful are documented, but new constants will be added without warning.
```

I also am pulling a bunch of stuff out of the api/Common.h that shouldn't be there and moving them all to Arduino.h, which is where stuff that isn't part of the official API should be. I

yuranikspb, 2021-06-02:
OK, thx

closed 2021-06-02 by yuranikspb