SpenceKonde/megaTinyCore #452: EEPROM.h bug  chip3216

yuranikspb, 2021-06-14:
```
void setup() {

//delay(2);
flag_first = EEPROM.read(40);

  if (flag_first != 17)
  {
    EEPROM.write(40, 17); //not work if no delay  2 ms
  }  
```

delay(1) not help
delay(2) work good
if  on start do EEPROM.write(random adress, random byte) work good

yuranikspb, 2021-06-14:
i m test with attiny 412 this sketch  if not use delay, EEPROM.write(40, 17); will never work
```
#include <EEPROM.h>
byte flag_first=0;

void setup() {
//delay(2)
flag_first = EEPROM.read(40);

pinMode(4,OUTPUT);

if (flag_first != 17)
  {
    EEPROM.write(40, 17);
    digitalWrite(4,1);
   
  }
else{

digitalWrite(4,0);
}

}
```

SpenceKonde, 2021-06-15:
I uploaded this to latest released of megaTinyCore, to a tiny412, The first time I uploaded, the light went on. As I would expect, Power cycled, led did not go on. Uploaded again, light did not go on. Changed number  (was 17 like you had, changed to 33). Ligt went on first time as it wrote the new value, and not after that. 

Note that, obviously when I set the fuses on the board I set them to retain EEPROM (you must do burn bootloader to apply that setting as is documented in readme - by default , a chip erase (which is performed every time code is uploaded through UPDI) will erase the contents of the EEPROM - unless EESAVE is is set to retain (not the default setting on a new part). 

```
#include <EEPROM.h>
byte flag_first=0;
void setup() {
//delay(2)
flag_first = EEPROM.read(40);
pinMode(1,OUTPUT);
if (flag_first != 33)
  {
    EEPROM.write(40, 33);
    digitalWrite(1,1);  
  }
else{
digitalWrite(1,0);
}
}
void loop() {
}
```
When I do this,

SpenceKonde, 2021-06-15:
Can you provide a method to reproduce this consistently, and describe how you prove that the EEPROM is not being written? Also, ensure that you've set the EESAVE fuse correctly

SpenceKonde, 2021-06-15:
Also what BOD level, and what kind of sort of powert supply at what voltage?

SpenceKonde, 2021-06-15:
Also when it fails, - what are the symptoms?

In any event, with this not happening for me, it is very difficult for meto speculate on what's going on much lkess fix.

yuranikspb, 2021-06-15:
Bod level 4.2 
Core latest
Fuse  not retained burn
First upload led on , power cycling, led on,many times power cycle led on

Add delay 2 ms upload,  led on
Power cycle led off

yuranikspb, 2021-06-15:
video: https://youtu.be/q0EaRjrudP4

SpenceKonde, 2021-06-15:
I've reproduced this too. I'll play with it a bit, see what other parts the issue reproduces, on, then ping my contact on the inside and ask him about it. If the 2-series doesnt have this behavior, then they know about it, fixed it, and never admitted it (not unprecedented) if everything does it, well, *shrug* maybe we'll see another issue added to the next revision of the errata.

SpenceKonde, 2021-06-15:
Bug label left because EEPROM.h should be able to work around this easily if it is as simple as it appears.

SpenceKonde, 2021-06-16:
Okay, so, this shit is insane. 

```

```

reproduces the behavior some of the time, with or without the delay. EVEN IF EEPROM IS RETAINED
So somehow the READ is getting bogus answers.... **_But simply commenting out the EEPROM.write() line (and leaving EEPROM retained) prevents the problem from occurring._** 

Either way, further down in the sketch, reading that location gives accurate answers.

SpenceKonde, 2021-06-16:
Confirmed misbehavior when all the serial stuff is commented out (so that I can look at the assembly listing) and have hope of figuring out what the shit is going on.

SpenceKonde, 2021-06-16:
Even THIS bloody simple code reproduces it!

``` 
#include <EEPROM.h>
byte flag_first = 0;
#define magicvalue 44
void setup() {
  flag_first = EEPROM.read(41);
  VPORTA.DIR |= 0x80;
  if (flag_first != magicvalue) {
    EEPROM.write(41, magicvalue);
    VPORTA.OUT |= 0x80;
  } else {
    VPORTA.OUT &= ~0x80;
  }
}
void loop() {}
```

SpenceKonde, 2021-06-16:
Okay - this is fucking wacky. 
To summarize findings:
Light is sometimes on (indicating failed read), sometimes not after power is turned on in my setup
Happens on 412, does not happen on 1626 nor on 3216. 
delay(2) vs no delay makes no difference. 
Reading from the same address a second time, before the write, yields the expected result. 
Retaining EEPROM makes no difference beyond the expected (light will come on immediately after programming if not retained, never will if retained). After that, it seems to have the usual chance of failure when power cycled - that is, the READ is failing, not the write - which rules out all the plausible explanations I had relating to this!
Brown-out detection enabled/enabled at maximum voltage vs disabled/disabled does not make a difference. 
but with EEPROM retained and that call to EEPROM.write commented out, the light never comes on. 
Incidence seems to be in the area of 10%
Startup time (SUT) makes no difference 
I am mystified

And suddenly, I can no longer reproduce it anywhere!
What the bloody hell, I haven't a clue why this is happening or how to address it.

SpenceKonde, 2021-06-16:
However, the behavior when it was happening is not something I can explain without undocumented silicon bugs. I was able to increase the delay up to 80ms and it still happened sometimes.

SpenceKonde, 2021-07-27:
Please anyone who has encountered behavior like this who has any ideas, puhleeeese post what you know!

Fins1, 2021-08-04:
I am having an intermittent problem with simple EEPROM storage on a ATtiny3216 which may be related to the above problem.  I am storing a flight number (uint16_t flightNum) in EEPROM that is incremented on each flight in a radio controlled aircraft.  Using the standard #include <EEPROM.h> library,

      EEPROM.put(EEPROMaddr, flightNum);

and the corresponding EEPROM.get statements to store the flight number, the code will run properly for 8-12 or so increments of the flight number.  In what appears to be random, the flightNum will then get corrupted, i.e. change to a much larger than expected number such as 72 or 586.  This problem has now occurred on two independent sets of hardware.  I have just started investigating so don't have much additional information.

SpenceKonde, 2021-08-05:
What BOD settings are you suing? What voltage are you operating at? Remember that BOD settings are only written when you choose "tools -> burn bootloader! (see the relevant sections of README.md, there are a number of reasons we had little choice in how we implemented this.

SpenceKonde, 2021-08-05:
When used in a mannner inconsistetnet with the manuifactiurer specificications it is not surprisingthat iwssues like that might be observed. Frankly, it is ashonishing how far on can depart from the maufactuter specificationswithout anything breaking - buit voltagage while writing the nvm apparently is one area where theyy are not joking.

Fins1, 2021-08-06:
I am operating at +3.3 V and 10 MHz.  To my knowledge, everything is operating within manufacturer specifications.  I am not using BOD since there are not any credible situations where I would be writing (put) to EEPROM where I would have a brown out.  (Certainly would not cause the 5 to 10% failure rate I am experiencing.)  There are circumstances, such as intermittent startup voltage, where I could be reading (get) from EEPROM.  It is not clear in the ATtiny3216 data sheet that that would cause EEPROM data corruption.  

Perhaps I should implement BOD just as a good design practice anyway?  I had not considered Optiboot when I designed this project as the recommendations seemed to be against using Optiboot unless you just had to.  That recommendation seems to have changed in the latest README.md?  To use Optiboot, do I just need to run "Burn Bootloader" and then upload with all the correct settings, including BOD?  I have plenty of flash and do not need a hardware reset pin.  Can I then continue to use UPDI as I do now?  The project uses a custom PCB and for now I want to continue to use the jtag2updi programmer with a Nano?  (I plan to eventually migrate to the new pyupdi-style programmer.)  Thanks for your suggestions.

SpenceKonde, 2021-08-06:
I amnot suggesting you use options or that doing that will help  just that
choosing burn bootloader (which is a mislabeled option in the ide, the
command should be marked "set fuses and burn bootloader, if any". Is
required to set "unsafe" fuses. For no bootloader near definitions, this
only sets fuses. I wish that there was a way t o change the name of that
option from a core but there is not. This is the same behavior as all Avr
devi es without bootloader gave used since the beginning of arduino - on
classic avrs none of the fuses are set whatsoever without burn bootloader,
imcluding for non bootloader boards. (So clock speed, for example or
whether yo use a crystal isn't set). On megaTinyCore when not using a
bootloader (we do not recommend the bootloader for most applications) we do
set some fuses -those that are guaranteed not to make it hard to further
program the part.so things like bod voltage and reset pin configuration
don't get set because if you have a 3.3v board you could set its bod
voltage to 4.3, for example. That would turn it into a brick until you
apply 4.3+ volts so the bod will release and let you reprogram it, and if
it's soldered to a board with othet parts which would be damaged by
4.3+volts, that will be particularly inconvenient. If you turn updi into
reset, you'd better have an HVupdi programmer..   I think there might be
one other r "unsafe" fuse. That is the same reason classic avrs require
burn bootloader for fuses. Except on classic parts, there isnatnleas t one
"unsafe" bit in each fuse.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

Fins1, 2021-08-07:
First, thanks for your information and support in resolving this problem!

I pulled a subset of my program to try to reproduce the EEPROM corruption problem I was observing.  Initially, I was able to reproduce the problem (corrupted EEPROM values), without BOD.  (I wanted to try first without BOD, then turn BOD on to see if that fixed the problem.)  The corruption was occurring >10% of the time on successive power on cycles.  In my application, the EEPROM corruption occurs < 5% of the time.  No idea why it was occurring more frequently in the test circuit?  Now ... and you know what's coming ... after several variations of my test program to try to isolate the problem, I have not been able to reproduce the failures.

So, any further information or suggestions for resolution?  Any word from Microchip?  I will continue to try modifications of the test program to see if I can reproduce the problem.  If the problem is incorrect READs, and you were able to fix the problem in your testing by reading from the same address twice, is a temporary fix reading from the same address twice?  (I was not able to try this before my program suddenly started working.)  Could there be a problem where the EEPROM needs to be exercised by many reads and writes, and that is why we can no longer reproduce the problem after some period of use?

SpenceKonde, 2021-11-09:
You weren't by any chance seeing this on a part with the UPDI pin fused as reset were you?

Fins1, 2021-11-09:
Nope.  UPDI NOT fused to Reset.  Changed EPROM address to not use 0 and made a few other changes to code to try to make reads and writes more reliable (can’t remember details without looking back at code) and haven’t had any problems with multiple systems used since.

EclipseBoom, 2022-01-22:
Just a comment here.  I ran into EEPROM write problems early in a program on Tiny1626.  8MHz@2.7V nominal.

I wanted to store the CPU-ID bytes into the EEPROM for easy dump to file by the programmer.  

The reason I found was a slowly rising power supply which when the program started to run (BOD 1.8V) seemed to pull the power below 1.8V with the extra load from the EEPROM write and/or changed the voltage enough during the write process that the EEPROM write failed and sometimes hung the part.

That drove me crazy for a long time because like your examples above it wasn't consistent.

Fixed this by increasing the power supply current, slowing the startup delay to 32mS and waiting until later in the program to do the EEPROM writes until I was sure the power was fully on to 2.7V

So dancing close to the BOD limit may be problematic if the EEPROM write circuits put a relatively large new load on the power supply.

closed 2022-02-17 by SpenceKonde