SpenceKonde/megaTinyCore #455: Deep sleep functionalities in ATTINY 1607 - explanation for wake up cycle using button interrupt - example 😅

dattasaurabh82, 2021-06-17:
Again. Not an issue 😅

**The task at hand:**
1. Detect a button press -> Fire an ISR -> clear the flag
2. Wake up the uC [ `ATTINY1607` ] from sleep. (Should happen automatically on interrupt detection)
3. Do some thing for few seconds with one of the following [ delay() over millis() over internal RTC() based timer ]
4. After task completion, go to sleep
5. Save power as much as possible during sleep (like do not keep any unused pins floating etc. Anything more? )
6. Then Step 1

Basically I want to have repetition of : 

> press button to wake up from deep sleep and do something for some time with as much power saving as possible during the running process phase and sleep mode.  

**For step 1:** Things I have tried. 
I'm using / kind of have to use `PB4` for that button and is initiated in  `setup()` with `INPUT_PULLUP` 
Now there are few ways of doing this and needs to be declared if to be used as interrupt pin. 
And moreover, if I'm correct (not sure 😕), for `ATTINY 1607`, all the pins are _asyncronous_ and can be used for interrupt detection, meaning: they can wake up the uC from sleep mode. 

So the way I have first tried declaring them as `INPUT_PULLUP` are:

### method 1:
```
/*
    FOR Low Power ATtiny1607
*/

#include <avr/sleep.h>

volatile byte interrupt1;

ISR(PORTB_PORT_vect) {
  PORTB.INTFLAGS = PORT_INT4_bm;        // Clear PB4 interrupt flag
}

void setup() {
  PORTB.PIN4CTRL = PORT_PULLUPEN_bm;    // Same as pinMode(Button, INPUT_PULLUP);

  Serial.begin(115200);
  delay(10);
  Serial.println("Startup");

  // Don't keep any othe IO floating, init them with internal pullups [ for saving power...]
  // TBD ...

  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
}

void loop() {
  if (interrupt1) {
    interrupt1 = 0;
    Serial.println(F("Woke up"));
  }

  // fake task holding simulation
  delay(5000); // TBD: try other methods like RTC and avoid delay() and millis()

  PORTA.PIN3CTRL = PORT_PULLUPEN_bm | PORT_ISC_LEVEL_gc; // Pullup + Trigger on low level
  // *** Instead of using detect LOW if I use FALLING or other methods, it also doesn't work
  
  Serial.println(F("Sleeping..."));
  Serial.flush(); // flush everything before going to sleep
  delay(10);
  sleep_cpu();

  // Continue after interrupt
  PORTA.PIN3CTRL = PORT_PULLUPEN_bm;                     // Turn off pin sense interrupt
}
```
### method 1 issues:
The awakening of the uC happens only first time button press and then it doesn't work. Basically next time it _skips_ the `delay()`.  also instead of detecting LOW as an interrupt on the button pin, if I use FALLING or other methods, it also doesn't work. What am I missing here? A structured example based on the above would help a lot ☺️ 

### method 2:
```
#include <avr/sleep.h>

volatile byte interrupt1;

ISR(PORTB_PORT_vect) {
  byte flags = PORTB.INTFLAGS;
  PORTB.INTFLAGS = flags; //clear flags
  interrupt1 = 1;
}

void setup() {
  PORTB.PIN4CTRL = 0b00001001; 
  // ISR will be fired on noticing CHANGE on that pin (other 3 modes don't bring this uC out of sleep mode)

  Serial.begin(115200);
  delay(10);
  Serial.println("Startup");

  // Don't keep any othe IO floating, init them with internal pullups [ for saving power...]
  // TBD ...

  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
}

void loop() {
  if (interrupt1) {
    interrupt1 = 0;
    Serial.println(F("Woke up"));
  }

  // fake task holding simulation
  delay(5000); // TBD: try other methods like RTC and avoid delay() and millis()
  
  Serial.println(F("Sleeping..."));
  Serial.flush(); // flush everything before going to sleep
  delay(10);
  sleep_cpu();
}
```
### method 2 issues:
This works but I'm not sure why only CHANGE interrupt is usable for waking up the uC and also not sure if there are other better ways of doing so. 

### Avoiding delay()  [ separate from this particular issue ]: 
I believe ( amateur in embedded AVR world ), that we should definitely avoid `delay()`  [ also not suitable for my use case] and `millis()` to save some memory and empty work by CPU. One option is using internal RTC of, in my case, ATTINY 1607
Now I'm no expert and I copied something from hackaday: 
```
//--- uC's internal RTC based interrupt based delay method ---//
void RTC_DELAY_init(int RTCdelay) {
  RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;    // 32.768kHz Internal Crystal Oscillator (INT32K)

  while (RTC.STATUS > 0);               // Wait for all register to be synchronized
  RTC.PER = RTCdelay;                   // Set period for delay
  RTC.INTCTRL |= RTC_OVF_bm;            // Enable overflow Interrupt which will trigger ISR
  RTC.CTRLA = RTC_PRESCALER_DIV32_gc    // 32768 / 32 = 1024 (sec) ~ 1 ms
              | RTC_RTCEN_bm                        // Enable: enabled
              | RTC_RUNSTDBY_bm;                    // Run In Standby: enabled
}

ISR(RTC_CNT_vect) {
  RTC.INTFLAGS = RTC_OVF_bm;            // Clear flag by writing '1':
  showTimePeriodOver = 1;
} 
```

Then we can use the function `RTC_DELAY_init(int RTCdelay)` as a replacement for `delay()` and `millis()` based custom delay. Basically this promises to offer less to none CPU load and doesn't occupy too much memory (in laymen's terms). 
I know that this also triggers an interrupt, which have the potential to bring the uC out of  `sleep`, which is okay for me as I'm only using it while the uC is `awake`. 
Is there a better way? May be an example can help. 
 

**Recommendations in general: ( if I may 😅  )** 
1. Example of _Wake up_ from _deep-sleep_ with _pin interrupt_ across _multiple ATTINYs_ would also be helpful. 
2. Some potential examples related to certain topics in one place would helpful in general.
3. Some other place to have discussions on topics would be, as well, helpful as this topic for example is not an issue.

hmeijdam, 2021-06-17:
3. Some other place to have discussions on topics would be, as well, helpful as this topic for example is not an issue.

That's why this repository has the discussions area.
https://github.com/SpenceKonde/megaTinyCore/discussions

dattasaurabh82, 2021-06-17:
> 1. Some other place to have discussions on topics would be, as well, helpful as this topic for example is not an issue.
> 
> That's why this repository has the discussions area.
> https://github.com/SpenceKonde/megaTinyCore/discussions

Thanks for pointing out. I will post there now.