SpenceKonde/megaTinyCore #457: ATtiny3216 with two SoftwareSerial ports

mrWheel, 2021-06-17:
Does anybody has experience with the ATtiny3216 using two SoftwareSerial ports?
I just can't get it to work!

One SoftwareSerial and the normal UART serial works great but I need an extra Serial port

SpenceKonde, 2021-06-17:
Do they both need to receive?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

mrWheel, 2021-06-17:
> Do they both need to receive?
> […](#)

Yes, they do :-(

SpenceKonde, 2021-06-18:
And I take it SoftwareSerial doesn't work? 

Note that if you cannot ensurre that both are not receving data at the same time, you will get garbage wwith nearly any software serial implementation, why not use the 1626 which has 2 hardware serial ports and can then add on a software one.

mrWheel, 2021-06-18:
> And I take it SoftwareSerial doesn't work?

no, it does not work.
Unfortunately I need 32k memory due to the size of the firmware.
And luckily there is also an ATtiny3226 which is pin-compatible with the ATtiny3216 … but this chip is nowhere available!
Anyway: thanks for your reply’s.

SpenceKonde, 2021-06-18:
Please describe exactly what you mean when you say you can't get it to work. Do you get an error message? Do you receive gibberish? Do demons fly out of your nose?

mrWheel, 2021-06-18:
> Please describe exactly what you mean when you say you can't get it to work. Do you get an error message? Do you receive gibberish? Do demons fly out of your nose?

Sorry, your absolutely right!

I get gibberish (luckily no demons ;-))

I have now changed my code to use `softs1.listen()` or `softs2.listen()` if I expect chars to come from one or the other SoftwareSerial "ports". But still at no avail.

SpenceKonde, 2021-06-18:
If you just use *one* sofwrare seriual port does it work? That output could be caused by baud msismatch toio,

mrWheel, 2021-06-18:
> If you just use _one_ sofwrare seriual port does it work? That output could be caused by baud msismatch toio,

Yes! With one UART and one SoftwareSerial it does work. I should have tested with two SoftwareSerial ports before fabricating the PCB's :-(

SpenceKonde, 2021-06-22:
Hmm, That really *ought* to work... I'll see if I can reproduce it with 2 serial adapters - but I have a few higher priority issues atm

SpenceKonde, 2021-07-26:
Had a thought (do not expect to get to work on this in the next few months, due to the length of my "urgent" tasks list with deadlines several months ago) - Are the pins you're trying to use for RX on the same port? Try using pins on different ports. 
Also keep the baud rates low. Because softwareSerial uses attachInterrupt to get the interrupt, it has a lot more latency than is necessary, and is probably worse if the two interrupots are on the same port. 

So yerah - try RX pins on different ports, and try dropping the baud rates. 

Last I heard, 32k 2-series would be out before year end, but it's all hearsay.

SpenceKonde, 2021-07-26:
If it works with the RX pins on different ports, and/or if works with a lower baud rate, then it comes down to latency and the poor architecture of that library, and it should at least be documented. 

Fixing softwareSerial is an issue beyond the scope of this core (the correct tool for fixing the problems with softwareSerial is ctrl+a, followed by backspace and a large amount of time and effort by someone with deep knowledge if AVR internals and the oppration of bitbanged serial interfaces. )

SpenceKonde, 2021-07-27:
And - if that were enough to make it work, you'd be all set here, can you verify this, since you've got a setup on hand for testing this?

mrWheel, 2021-07-27:
> And - if that were enough to make it work, you'd be all set here, can you verify this, since you've got a setup on hand for testing this?

I have a PCB where I write the firmware for, so it is not possible in this setup to change pin’s.
I use 9600bps which is already rather slow.

For this project I stopt trying to use two softwareserial ports. One (as I have now) is challenging enough.

But thanks for the megaTinyCore and all your guidance!

As soon as the ATtiny3226 will be my preferred cpu

SpenceKonde, 2021-07-28:
Okay.... I guess I'm closing this? Even though there might well be a bug here, I am not likely to be able to reproducer it unaided in the absence of  ready-to-go test case due to my already overwhelming workkload. Please reopen if a ready-rto-test sketch can plug-and-play demonstrate this issue. (my workload is such that I I need to prioritize fixing things that I do have test code that reproduces)

closed 2021-07-28 by SpenceKonde