SpenceKonde/megaTinyCore #458: SPI clock frequency too low

5e3, 2021-06-17:
I toyed around with an attiny1626 to drive a WS2812 panel from the SPI interface. At MCU clockspeed 16MHz everything is as expected:   

SPI.setClockDivider (SPI_CLK2X_bm);

results in an 8MHz SPI (125 nsec) clock.
 
When I switch the MCU clock to 20 MHz  the SPI clock seems to be at 2.5 MHz (400nsec) according to my PicoScope. I'm not aware that 20MHz MCU speed is fundamentally different from 16 MHz.  Nothing like that is mentioned in the hardware errata list so it must be expected behavior.

The question is why is the SPI clock not on 10MHz (100nsec) when the MCU is running at 20MHz. (The code behind setClockDivider() is so straightforward  that I doubt that this is a megaTinyCore software problem but nevertheless: what's going on? What did I miss? 

Any explanation?

SpenceKonde, 2021-06-18:
Are tyou sure you're doing that after it's applied SPISettings to it, which by default will crank irt back? I think it's probabl;e intentionally setting clockdiv to 2 (/8 instead of div/4) or something, see how it uses the default SPI settings class (the likes in the 50's in SPI.h
Happy to merge PR's but unlikely to get to this anty time soon, this is a low priority issue. and It's not clear what correct behavior is.

SpenceKonde, 2021-06-25:
Bug was found, an error in setClockDivider() caused the mask to ber incorrectly calculated as 0xFF instead of 0xE9. This caused the strange differences by F_CPU because if not specified begin targets a speed of 4 MHz (at most) so the clock speed was set to div4 without CLK2X, and setting the divider to 2 would result in the CLK2X being set. 

But on 20 MHz's div/4 would be 5 MHz that was higher than 4 MHz so it would go one step down to div/8 (div 16 and CLK2X)... and setting the clock divider to 2 would not change that.

closed 2021-06-25 by SpenceKonde

5e3, 2021-06-25:
Great. You are moving fast! I'm glad you found that problem. Sorry I didn't really read the code and just expected it would not work in the 16MHz use-case either if there where a problem in setClockDivider().
I have to say this here: I use your libraries for almost all of my projects. It makes it so much easier to get going. Thanks a lot for your efforts. I really appreciate it.
Nils

SpenceKonde, 2021-06-25:
Hm? No, it did work t 16, which was what was so strange.... default targets F_SCK = 4 MHz. That means the div 4x, (numeric value 0b00. is used. setClockDivider could set but not clear bits, so this could be kicked up if you asked. But for 20. thatwould give 5 MHz - too much (the speed specified in SPISettings default is the maximum it would start with, so it would use div/8 prescaling. (div/16 (0b01) and clkx2). Since setClockDivider could never clear that prescaler bit, you couldn't slow it down when running on 20.