SpenceKonde/megaTinyCore #468: Servo_megaTinyCore max angle 270 anyone?

Yardie-, 2021-06-23:
Hi Spence 
I know this is an unusual one but I have some 270 degree micro servos coming and from the companies alliexpress site I would guess they have had this problem in the past.
ie. You said this is a 270 but it will only do 180 .

I know there are simple ways around this but.... maybe you could just change value > 180 to value > 360 
or add a ifndef
SERVO_MAX_ANGLE 180

I can easily just do the map myself which I probably would as each servo will probably need an individual min and max. 
(I'm turning some old 4" pressure gauges into electronic gauges. Steampunk style so the avr will run a few at a time)
But in case someone goes WTF I set it to 265 and it only does 180 that *&%^&% Chinese seller without looking at the code..

void Servo::write(unsigned int value) {
  // treat values less than 544 as angles in degrees (valid values in microseconds are handled as microseconds)
  if (value < MIN_PULSE_WIDTH) {
    if (value > 180) {
      value = 180;
    }

    value = map(value, 0, 180, SERVO_MIN(), SERVO_MAX());
  }
  writeMicroseconds(value);
}

SpenceKonde, 2021-06-23:
How do those servos work though? The problem is that if you drive servos too far in either direction some of them can be physically damaged. What pulse timing is needed to get the extra 90 degrees? 

Your suggestion of a define doesn't work, as they are not visible across compilation units, and each .c file is a different one of those.

Yardie-, 2021-06-23:
Probably best to just show you the link.
the pwm values are there.
I guess they have two stop points further around the circle. or an extra gear/different gearing.
Sensing another gear I won't really know till they arrive. 
They wouldn't be true digital, too cheap. 
I am not going to need to drive any weight at all with them and no holding torque is really required. Just screw the gauge arm straight to the servo and position it appropriately.
I can understand you may not want to change the code over this. 
I will just work out how they work when I get them and set the appropriate timings.
It was sort of more a heads up these things exist in some form. 
When they arrive I can report the results from the real world tests.
If they work they will make cool temp / humidity ... gauges. 

https://www.aliexpress.com/item/4000056467965.html

SpenceKonde, 2021-06-23:
It sounds to me like it should just work... 
They state requirements as (using the term PWM very incorrectly!) "with270 degreetop angle(PWM 500-2500us)  ,some >userPWM520~2350us could arrive 270degree(if PWM at 500us and 2500us point the servo continous rotation, that mean you use PWM520~2350us is enough)"

In Servo.h from megaTinyCore:
```
#define MIN_PULSE_WIDTH          544     // the shortest pulse sent to a servo
#define MAX_PULSE_WIDTH         2400     // the longest pulse sent to a servo
```

Yardie-, 2021-06-24:
Ok that's cool so write 180 will just end up going 270 that's fine. 
So are the two libraries Servo and Servo_megaTinyCore are pretty much the same now.

I think the poor explanation is just a translation issue. Having tried to deal with translated Chinese data-sheets. 
Also using Google translate to translate the same it gets pretty weird.
Back translating a couple of times ends up with some very strange results. 
I think the two languages are very different in the way stuff gets described. 

I'll just work with that.
Thanks for your time.

SpenceKonde, 2021-06-24:
The two libraries included with core are identical except for the name. The purpose of the megaTinyCore one is so that if you have the "standard" servo library installed (which would normally take precedence, and doesn't support these parts properly), you can just change the #include to point to Servo_megaTinyCore.h and it will work with no other code chjanges.

Yardie-, 2021-06-24:
Sweet thanks

SpenceKonde, 2021-06-24:
Please reopen if proble,sms and let me know how they work out, closing this because at present there is no reason to belive there is a defect in the core (well here at least)

closed 2021-06-24 by SpenceKonde