SpenceKonde/megaTinyCore #470: Serial1.swap(1) Does Not Appear to Work

Eddiiie, 2021-06-25:
ATtiny 1626-SU
I was going to look into the Timer2 thing in the discussions but got hung up on this.

I seem to be having trouble with swapping pins for Serial1.
Note, softwareserial works on PIN_PC2 (the alternate USART1 TX pin)..  The pin works...
Note, I also tried Serial1.pins(PIN_PC2, PIN_PC1) with no luck
```
int i = 555;    // I read that if swap was a success that it will return a TRUE if success.  That is a 1, right?

void setup() {
  // Setup communications..
  Serial.swap(1);          // Move Serial (0) to alternate pins, PIN_PA1 (TX) and PIN_PA2 (RX)
  Serial.begin(38400);     
  Serial.println("Start..");
  Serial.flush();      // Start... was not appearing until I added flush()

  i = Serial1.swap(1);         // Move Serial (1) to alternate pins, PIN_PC2 (TX) and PIN_PC1 (RX)
  Serial.print("Serial1.swap(1) i = "); Serial.println(i);  Serial.flush();  //  Prints:  "Serial1.swap(1) i = 1"

  Serial1.begin(38400);    // ...

}

void loop() {
Serial.println("This is Serial0..");
Serial1.println("This is Serial1!");
delay(500);
}

```

The above code will produce this on Serial  (0)..
```
Start..
Serial1.swap(1) i = 1
This is Serial0..
This is Serial0..
This is Serial0..

```...

I can't get a "This is Serial1.." on any pin.

IF I comment out the line "Serial.swap(1);" the ATtiny 2616-SU stops  executing on the "Serial1.begin(38400);"  line.
 It just stops.

The reason I want to use alternate pins is because I was planning to use a 20mhz resonator and those pins are used for USART0 and I need 2 serial ports.  However, the default speed of 20Mhz appears to be fine.   Should I not use a resonator? 

I've tried doing both swaps before  Serial.begin's ,   I've tried swapping, begin serial 1, then swapping and begin serial..  I've tried swapping and enabling serial first then serial1,  makes no difference.
I tried serial1.Pins() .. nothing.

Ok, let's take a look at the timer topic. 

Thank you!

SpenceKonde, 2021-06-25:
Uh - hate to break it to you, but... that;'s not a 20 MHz oscillator you get to put there. All you can do is a 32.768 kHz watch crystal, for the RTC. No high speed crystal support except in DB and future DD series parts (supported by my DxCore) 

Nonetheless I will look into serial1 - it ought to work, but the code is very formulaic

SpenceKonde, 2021-06-25:
Confirmed the issue with serial1.swap, fix is trivial,just a copy-paste error in the variants

closed 2021-06-25 by SpenceKonde