SpenceKonde/megaTinyCore #479: [solved] Upload issue with CH340 Nano Clone

1larity, 2021-06-30:
Not really an issue, but a problem and solution. I was having no joy uploading using a chinese (CH340) Nano clone to a 1614 F and found it would only work with no resistor in circuit with the UPDI pin. Avrdude was reporting either no power or incorrect state errors during upload.
Hopefully this might be helpful to someone experiencing the similar issues in future.

SpenceKonde, 2021-07-05:
Hmmm. Thanks for reporting this, what option were you using for the wiring, and what was the value of the series resistor? I have found that it's right on the edge when not using a schottky diode between RX and TX (the resistance only configuration).

SpenceKonde, 2021-07-08:
If you could, are you able to answer these questions? Thanks

1larity, 2021-07-09:
Waiting for some Schottky's s to arrive, and working out how to use my new 'scope, then I'll get back to you.

SpenceKonde, 2021-07-11:
Oooooh, you're using a nano clone, I misinteropreted it as just a ch340 as UPDI prigrammer, the nano part completely missed me. 

For programming with jtag2updi, the total resistance should be max. 1k, preferably 100-470. 

The 4.7k resistor is for when using a CH340 or other serial adapter without an arduino as programmer.

closed 2021-07-11 by SpenceKonde

1larity, 2021-07-15:
Nice, thanks for the advice, there's quite a few tutorials around with conflicting information regarding the resistor.