SpenceKonde/megaTinyCore #483: ATtiny1616 Analog Comparators Question

nehalem501, 2021-07-08:
Hello,

I have a use case where I need to use analog comparators on 4 different signals. But these signals work in pairs, meaning that only one in each pair can be active at a time.
So the solution I thought about is to use 2 ACs (one for each pair), use 2 different positive input pins on each AC and swap every few milliseconds between the 2 positive pins to detect on which one the signal is incoming.

My question is about the AC positive inputs, is AINP1 from AC0 the same as AINP1 from AC1 (they are on different pins)? Or are the 4 positive inputs from each AC independent? The MCU's datasheet is not very clear about this.

I'm asking this question because this is the pin configuration I'm thinking about using (image below), but I want to be sure that using AINP1 simultaneously on both AC0 and AC1 with diferrent signals will work. (I will be using the 20 pin version of the chip, and internal references).
[image]

SpenceKonde, 2021-07-08:
Yeah, each AC has it's own idea of what each AIN is, and they do not conflict with eachother. 

You've also got a third AC, that could get one of the 4, so you'd only have to swap one of them..... 

maybe you could eliminate the swapping entirely by getting a 4th "comparator" by using ADC1 in free running with the window comparator functionality configured appropriately....

SpenceKonde, 2021-07-08:
Not a defect in the core so transferrring to discussions.

closed 2021-07-08 by SpenceKonde