SpenceKonde/megaTinyCore #485: Missing methods in Print class and HardwareSerial class

drmpf, 2021-07-18:
In V2.3.2
The following method is missing from Print
error: 'class Print' has no member named 'flush'
The following method is missing from HardwareSerial
error: 'class HardwareSerial' has no member named 'availableForWrite'

SpenceKonde, 2021-07-19:
Huh, yet the official core for megaAVR doesn't appear to? What are the consequences of this - though i dont directly see a problem bringing in those two  methods. I am not a C++ person, so maybe that ti the source of the nblind spot, but I'm somewhat confused over what the consequences of these not being present in the file. since Print is something that just gets subclassed into things that need to write their own versions of those methods? But yersh I'm ,m,uch wrse at C++ tjan I am at C.

drmpf, 2021-07-19:
This ArduinoCore
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Print.h
has those two methods.
My library SafeString, includes a BufferedOutput class that provides non-blocking writes and looks for availableForWrite to see if there is space.  The default return of 0 is handled
SafeString itsself implements a 'dummy' Print class which expects to override flush() 

If these methods are missing in the base Print class the compile fails.
My fix for the moment is to look for defined(MEGATINYCORE_MAJOR)  and take special action, so currently my SafeString library compiles, but you might like to add these two methods for further compatability

```
virtual int availableForWrite() { return 0; }
virtual void flush() { /* Empty implementation for backward compatibility */ }

```

SpenceKonde, 2021-07-20:
Please do let me know if there are any other "missing" methods that cause problems. 

Also, the correct way to test versions of megaTinyCore programatically is to use the MEGATINYCORE_NUM define. The major, minor, and patch versions, you may notice are defined in platform.txt with the UL postfix, so they're treated as unsigned longs until shoehorned otherwise,, which is key to the MEGATINYCORE_NUM macro being fairly straightforward - It's an unsigned 32-bit integer with such that, for example. 2.3.2 released version would have a value MEGATINYCORE_NUM = 0x02030201, the current development version has 0x02040000 (non release versions are what people get when they install straight from github).. So since the fix to print class was added during development of 2.4.0, (0x02040000) you can count on it being in any version  higher than that. So your workarounds might want to be used 
```
#if ((defined(MEGATINYCORE) && MEGATINYCORE_NUM < 0x02040001) || \
      (defined(DXCORE) && DXCORE_NUM < 0x01030701)) 
  //workaround goes here - DxCore is based on the same codebase, as megaTinyCore,and had the same issue with missing methods. 
#else                    
  // methods no longer missing because using 2.4.0 release of megaTinyCore, or 1.3.7 release of DxCore or later
  // those will be the next available releases of those cores. and the first versions guaranteed to include those missing methods. 
#endif
```

closed 2021-07-20 by SpenceKonde