SpenceKonde/megaTinyCore #495: ISR() compile error in combination with SoftwareSerial (t3216)

mrWheel, 2021-07-26:
I have this code:
```
#include <avr/sleep.h>

#define PUSH_BUTTON PIN_PA4
.
.
//-------------------------------------------------
//--- ISR seems to be a "pre-defined" name --------
//-------------------------------------------------
ISR(PORTA_PORT_vect)
{
  // This function will bring us back from sleep
  byte flags=PORTA.INTFLAGS;
  
  PORTA.INTFLAGS=flags; //clear flags
  if (flags&0x02) {
    // The program will continue from here.
    // First thing to do is disable sleep.
    sleep_disable(); 
  }
  else
  {
    return;
  }
  Serial.println("\n=> in ISR()");

  // wait for button te be released ..
  while(!digitalRead(PUSH_BUTTON)) { ; }

} // ISR()

//-------------------------------------------------
void enterSleep(void)
{
  //-- set the sleep type
  set_sleep_mode(SLEEP_MODE_PWR_DOWN); 
  //-- enable does not enter sleep mode
  sleep_enable(); 

  // wait for button te be released ..
  while(!digitalRead(PUSH_BUTTON)) { ; }

  //-- Sleep the device and wait for an interrupt to continue
  sleep_cpu(); 

  /* The program will continue from here. */
  /* First thing to do is disable sleep. */
  sleep_disable(); 

} //  enterSleep()
.
.
void setup()
{
  pinMode(PUSH_BUTTON, INPUT_PULLUP);
  //-------------------V-- pullup (1=PULLUP)
  //                   |    001 = trigger on both?
  //                   |    010 = trigger on rising?
  //                   |    011 = trigger on falling? 
  //                   |    100 = digital input buffer disabled entirely (what does that mean?)
  //                   V 
  //                    vvv 101 = interrupt on LOW level
  PORTA.PIN4CTRL=0b00001101; //-- ISC=3 trigger on falling, ISC=2 trigger rising
 .
 .
}    // setup()
```
which works great. But if I make use of SoftwareSerial I get this compiler error:
```
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_3'
/Users/WillemA/tmp/arduinoBuild/sketch/ZDD_ComModule_v04.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Using library SoftwareSerial at version 1.0 in folder: /Users/WillemA/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.3.2/libraries/SoftwareSerial 
exit status 1
Error compiling for board ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.
```
I know SoftwareSerial is not Spence's favorite but is it possible to use both ISR() ánd SoftwareSerial??

SpenceKonde, 2021-07-26:
No. It's not possible on a classic avr either. Why would you expect it to
be any better here? SoftwareSerial depends on attachInterrupt. When that
api call is used anywhere, WIntertupt.c grabs every pin interrupt for
attachInterrupt(), which is one of the most grotesque constructions in the
Arduino API - nothing that uses attachInterrupt is compatible with anything
else that manually specifies a pin interrupt, and there is no way to fix
that without sacrificing api-compatibility. The moment you or a library
touches that, you can only get pin interrupts via attachInterrupt - you
can't define them manually (which means they will always be slow, because
attached interrupts check pin number and introduce function call overhead
to the isr. This burns like 50 clock cycles minimum, most of it wasted
pushing all 10 call-used registers onto the stack so the interrupt function
can use them can use them, even though a well written pin interrupt doesn't
need more than a couple at most and the majority don't need any - though
the two additional call and ret instructions of the literal call overhead
and the check of which pin triggered it aren't innocent either...  I'm sort
of in awe that SoftwareSerial doesn't suck way more than it does, honestly,
considering how much overhead it has to deal with. Especially on classic
avrs, where there's an extra 15 clocks by my count since call is 1 clock
slower and there are the 3 standard pushes and the 10 call used registers
getting pushed by the prolog and they take 2 clocks not 1 like the modern
(avrxt) parts...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-07-26:
A simple non-timing-critical ISR likr yours could probabnly work with attachInterrupt though, assuming it works at all; you are violating numerous rules of ISR (for example, that they must run quickly, while yours is writing to serial and waiting for it to finish printing. That is a total no-no, especially for such long strings. 

To wake from sleep, the ISR dfoesn't even have to do anything, and the notification of waking from sleep can thus go int the code just after you put the system to sleep.

closed 2021-07-26 by SpenceKonde

SpenceKonde, 2021-07-26:
Also, do note that I will see and respond to comments here - this is closed ismply because uit does not represent something that I can fix. 

I did add a note about this to the pin interrupt documentation in the core.