SpenceKonde/megaTinyCore #496: Test sketch from the core for "Servo_megaTinyCore" throws a compilation error when millis are disabled

hmeijdam, 2021-07-26:
I am compiling the test sketch that comes with the core for the Servo_megaTinyCore library for the Attiny412.

When I compile with millis disabled I get a compilation error. When I compile with millis on timer TCD0 or TCA0 I get no error. From the documentation I understood that the Servo library is using TCB0 on the Attiny412.

My question: Besides using TCB0, does the servo library also require micros() to be enabled on another timer? 

Thanks,
Hans

The 'Knob' test sketch from MegaTinycore ( I only changed pin 9 to pin 2):
```
#include <Servo_megaTinyCore.h>

Servo myservo;  // create servo object to control a servo

int potpin = 0;  // analog pin used to connect the potentiometer
int val;    // variable to read the value from the analog pin

void setup() {
  myservo.attach(2);  // attaches the servo on pin 9 to the servo object
}

void loop() {
  val = analogRead(potpin);            // reads the value of the potentiometer (value between 0 and 1023)
  val = map(val, 0, 1023, 0, 180);     // scale it to use it with the servo (value between 0 and 180)
  myservo.write(val);                  // sets the servo position according to the scaled value
  delay(15);                           // waits for the servo to get there
}
```

The resulting compilation error ( copied only the last bit, where the actual error is displayed)

```
Linking everything together...
"E:\\Arduino\\arduino-1.8.15\\portable\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mmcu=attiny412 -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_487721/Knob.ino.elf" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_487721\\sketch\\Knob.ino.cpp.o" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_487721\\libraries\\Servo_megaTinyCore\\megaavr\\Servo_megaTinyCore.cpp.o" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_487721/core\\core.a" "-LC:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_487721" -lm
C:\Users\Hans\AppData\Local\Temp\cca8dUTg.ltrans0.ltrans.o: In function `ServoHandler':
E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:54: undefined reference to `clockCyclesPerMicrosecond'
E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:55: undefined reference to `clockCyclesPerMicrosecond'
E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:89: undefined reference to `clockCyclesPerMicrosecond'
E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:95: undefined reference to `clockCyclesPerMicrosecond'
E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:90: undefined reference to `clockCyclesPerMicrosecond'
C:\Users\Hans\AppData\Local\Temp\cca8dUTg.ltrans0.ltrans.o:E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore\src\megaavr/Servo_megaTinyCore.cpp:235: more undefined references to `clockCyclesPerMicrosecond' follow
collect2.exe: error: ld returned 1 exit status
Using library Servo_megaTinyCore at version 1.2.1 in folder: E:\Arduino\arduino-1.8.15\portable\packages\megaTinyCore\hardware\megaavr\2.3.2\libraries\Servo_megaTinyCore 
exit status 1
Error compiling for board ATtiny412/402/212/202.

```

SpenceKonde, 2021-07-26:
This issue is already fixed for 2.4.0 schedukled for release 2-3 weeks ago and desperately behiund schedule, it will be fixed when I finally get this released. I've felt like I'm less than a day away from releasing it for weeks.... stuff keeps coming up :-(

SpenceKonde, 2021-07-26:
It hope to iomprove the CI so that gross oversights like that would be detected before releases instead of by users.

closed 2021-07-26 by SpenceKonde

SpenceKonde, 2021-07-27:
As an aside, the root cause here was that I just #ifdef'ed out everything timing rtelated when millis was disabled - including criticaklky important macros to convert between clock cycles and microseconds which, in the absence of millis timing are even more critical!

hmeijdam, 2021-07-27:
I was already looking in wiring.c  where I found
```

inline uint16_t clockCyclesPerMicrosecond() {
  return ((F_CPU) / 1000000L);
}

```
but I lack the knowledge to make my own workaround and define the missing macro. No worries, as in the meantime I was able to have TCB0 do the servo signal pulse capture and TCA0 generate the servo signal(s). I was keen to experience if these tinies can read and generate a more jitterfree servo signal, and I am very impressed with their microsecond accurate performance. What an improvement over the old tinies.