SpenceKonde/megaTinyCore #514: i2c aka TWI on UNO vs ATTiny

dduehren, 2021-09-01:
I have an ATTiny 1604 as a slave to a 8266 master.  Under the right circumstances I get a TWI hang (SDA=0, SCL =1) which eventually creates an exception in the 8266.  When I substitute an Arduino UNO for the ATTiny, the problem doesn't occur.  (i had initially developed the code on the UNO while waiting for PCBs with the ATTiny).  

So this has made me wonder what are the differences in a) the processors, and b) the wire and twi libraries.  I haven't had a chance to dig into this yet, and I'm hoping someone can shorten my investigation.  

The slave code is very simple, the i2c code is pasted below.   At this point I do not have scope traces, though I hope to get those later today.

The I2C slave code:
`#include "i2c.h"
#include "Arduino_DebugUtils.h"
#include "boardConfig.h"
#include <Wire.h>
#if defined(ARDUINO_AVR_UNO)
#include <Adafruit_NeoPixel.h>
Adafruit_NeoPixel pixels(NUMPIXELS, pixelPin, NEO_GRB + NEO_KHZ800);
#elif (defined(__AVR_ATtinyx04__) || defined(__AVR_ATtinyx14__))
#include <tinyNeoPixel.h>
tinyNeoPixel pixels = tinyNeoPixel(NUMPIXELS, pixelPin, NEO_GRB);
#endif
extern bool humOnOff;
extern float Temperature;
extern float Humidity;

enum dtype{TEMP, HUMID,RELAY};
enum cmds{RLY, LEDGO, LEDWR, LEDCL};
uint8_t rcvbuf[5];

template <typename T> int I2C_writeAnything (const T& value)
  {
    const byte * p = (const byte*) &value;
    unsigned int i;
    for (i = 0; i < sizeof value; i++)
          Wire.write(*p++);
    return i;
  }  // end of I2C_writeAnything

template <typename T> int I2C_readAnything(T& value)
  {
    byte * p = (byte*) &value;
    unsigned int i;
    for (i = 0; i < sizeof value; i++)
          *p++ = Wire.read();
    return i;
  }  // end of I2C_readAnything

  void i2cInit(){
    Wire.begin (I2CADDRESS);
    Wire.onReceive (receiveEvent);
    Wire.onRequest (requestEvent);
    pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
  }
  
 void requestEvent(){
//    Debug.print(DBG_INFO, "i2c Request.  Temp: %f, Hum: %f", Temperature, Humidity);
 uint8_t rdBits;
 rdBits = digitalRead(DHTPIN);
 rdBits = rdBits << 1 | digitalRead(humPin);
  Wire.write(rdBits);
  Debug.print(DBG_INFO, "rdbits %x", rdBits);
/*  Wire.write(TEMP);
  I2C_writeAnything(Temperature);
  Wire.write(HUMID);
  I2C_writeAnything(Humidity);
  */
 }
 
// We presume the sender knows what it's doing.  One command requires one argument, another no arguments, and a third 5.
 void receiveEvent(int howMany){
  uint8_t i;
  if (Wire.available()>=1){
      for (i=0; i<howMany; i++){
        rcvbuf[i]=Wire.read();
    }
    uint8_t cmd = rcvbuf[0];
//    Debug.print(DBG_INFO, F("receiveEvent Cmd: %d, howMany %d"), rcvbuf[0], howMany);
    switch(cmd){
      case RLY:
        humOnOff = rcvbuf[1];
        digitalWrite(humPin, humOnOff);  //LED is active high as will be the opto circuit to feed power relay
      break;
      case LEDGO:
        pixels.show();
//        Debug.print(DBG_INFO, "pixels.show");
      break;
      case LEDWR:
//         Debug.print(DBG_INFO, "LED write");
         #if defined(ARDUINO_AVR_UNO)
         pixels.setPixelColor(rcvbuf[1], pixels.Color(rcvbuf[2], rcvbuf[3], rcvbuf[4]));
         #else
         pixels.setPixelColor(rcvbuf[1], rcvbuf[2], rcvbuf[3], rcvbuf[4]);
         #endif
//         Debug.print(DBG_INFO, "LEDLD: %d  R: %d, G:%d  B:%d", rcvbuf[1], rcvbuf[2], rcvbuf[3], rcvbuf[4]);           
      break;
      case LEDCL:
        pixels.clear();
        pixels.show();
//        Debug.print(DBG_INFO, "pixels.clear");
      break;
      default:
      break;
    }
  }
 }`

dduehren, 2021-09-02:
Still no trace of the failure.

dduehren, 2021-09-02:
> template int I2C_writeAnything (const T& value)
> template int I2C_readAnything(T& value)

Hi Spenc,

Any time frame?  And can I help?

David Duehren

There is a possible replacement for wire being tested with plans for
inclusion in the next minor release if things keep going well. It fixes
several issues and you may be hitting one of them.

Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

dduehren, 2021-09-02:
One thing I have observed is that the 1604 is much faster than the Uno.

David Duehren

SpenceKonde, 2021-09-06:
Take the latest github version 2.4.3, for a spin. (the changes are not inthe "released" 2.4.2)

SpenceKonde, 2021-09-12:
Github version of Wire now compiles cleanly with the rewritten code (thanks to @MX682X for the new version of Wire!). Major flash savings too, as promised. Now we need people to see if it works. Note the new tools menu option to select master-or-slave or master-and-slave configuration, the former uses less flash and much less RAM than the latter.