SpenceKonde/megaTinyCore #525: Sometimes(!?) Wire.begin() hangs .. core 2.4.2 ATtiny3216

mrWheel, 2021-09-10:
Hi Spence,

I have a setup with ATtiny3216 and three I2C devices. PullUp resistor's 2k2.

Most of the time it works great .. but every now and then it hangs at the Wire.begin(address) statement.

I wonder: Is the Wire.setTimeout() function enabled and does it do what I hope is does? The sketch compiles with this statement but it does not prevent the code to hang (every now and then) at the Wire.begin(address) ...
It would be so nice if it just return with an error (error 5 seems the 'timeout' error).

SpenceKonde, 2021-09-10:
Be aware that we are replacing Wire.h with a completely new version in
2.4.3, which is hoped will fix the many issues people are having with i2c

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

mrWheel, 2021-09-11:
Spence,
I know you are very busy, but is there a time-line for 2.4.3?

SpenceKonde, 2021-09-11:
No.
megaTinyCore just got 2.4.0/2, next is DxCore, then ATTinyCore 2.0.0I cannot keep deferring those any longer, and the DxCore release will probably be small (it won't have any of the I2C changes yet) so that the long overdue ATTinyCore 2.0.0 can be completed and the legacy part core put fully inyo maintenance mode.. The I2C changes will be checked into megaTinyCore during development of the next DxCore version so that during that time that I'm on ATTinyCore, we have plenty of time for people to test such a huge, high risk change. It is a change that entirely immune to CI testing except at the most basic level, because whether a sketch compiles doesn't mean much (there in fact is a version of the new library checked into the github version of megaTinyCore, If you look at the latest compile examples results (under actions) you can see that every attempt to compile any sketch including Wire.h is failing to even compile)

SpenceKonde, 2021-09-11:
I don't think 2.3.0 will be any better, the changes in 2.4.0 I don't think should have any effect on that.

mrWheel, 2021-09-11:
> I don't think 2.3.0 will be any better, the changes in 2.4.0 I don't think should have any effect on that.

Ok, my assumption was that the Wire lib was the new one as it was last updated 5 days ago.

anyway: it’s incredible what you are doing for the community! Thanks.

SpenceKonde, 2021-09-12:
Github version of Wire now compiles cleanly with the rewritten code (thanks to @MX682X for the new version of Wire!). Major flash savings too, as promised. Now we need people to see if it works. Note the new tools menu option to select master-or-slave or master-and-slave configuration, the former uses less flash and much less RAM than the latter.

SpenceKonde, 2021-09-30:
> > Huh, regarding the protection diodes, that's what I was expecting to find! But experimentally, I don't see the expected behavior on a PCF connected to power supply ground but not the supply.
> 
> For me the problem has nothing to do with the pcf’s not being connected to +5v as I have removed the high-side switch and the pcf’s are now direct connected to the power rail (+5volt) but with the same erratic behaviour.
Moving convo back here

Well, at least that's consistent with what I saw. 

Regarding the problem with the switch letting the voltage drop as the thing controlling it was reset, that's because all pins are tristated during reset. But the standard solution, pullup or pulldown resistor, would burn power which is just what you were trying to avoid..... though the current consumption stats on those PCF's are pretty crap, so a high enough value resistor would hold that control line in place during a reset, and not waste that much power in sleep. Or you could use less ridiculously weak resistors to turn it off (maybe even turn it off immediately before expected resets?) and maybe wait a bit before powering them back on....

When the bus is idle, what voltages are on the I2C lines? On all the scope traces, as MX kept pointing out, it looked like they weren't the same, which... doesn't really make sense.

Also when you next catch it in a failure - are you able to disconnect I2C lines quickly to figure out what device is malfunctioning? The fact that it's being pulled to intermediate value is *weird* Either something is trying to drive it low and something else is trying to drive it high (this should *never* happen - nothing should ever be trying to drive the I2C lines high. You don't have any digitalWrite or pinMode calls that target those pins in your sketch, right?