SpenceKonde/megaTinyCore #528: Missing delete(void*,unsigned int) and delete[](void*, unsigned int) in new.h and new.cpp

SpenceKonde, 2021-09-15:
### Discussed in https://github.com/SpenceKonde/megaTinyCore/discussions/527

<div type='discussions-op-text'>

<sup>Originally posted by **DanSkareda** September 14, 2021</sup>
I did a clean install of the Arduino IDE 1.8.16. Using the "json" link I installed the current megaTinyCore. In the sketch folder I copied the unzipped current version of OneWireNg. In its Examples folder is the DallasTemperature sample project. I opened that and did a compilation for attiny3216/Optiboot (it doesn't work in general for any circuit). The compilation ended with errors:

C:\Users\radek\AppData\Local\Temp\cccalydd.ltrans0.ltrans.o: In function UartClass::begin(unsigned long, unsigned int)': <artificial>:(.text+0x8ee): undefined reference to operator delete(void*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
exit status 1

In the spring I also tried it for attiny3216, everything was fine then. The library author's opinion on this problem is here:

https://github.com/pstolarz/OneWireNg/issues/20#issuecomment-918427756

Here is all discussion:

https://github.com/pstolarz/OneWireNg/issues/20</div>

SpenceKonde, 2021-09-15:
I have commented in that discussion; there has been at least one and possibly two prior issues relating to new which is a black box to me. Can someone please give me a definitive reference for what functions the new.* files should contain, and documentation of what they do? Or is this just some undocumented black box that gets passed around,. added to in secret, and then reported as bugs when other core authors don't divine that new has been updated?

henrygab, 2021-09-16:
As requested, here's some references:

(Click to expand / collapse each section.)

<details><summary>Reference pages for you...</summary><P/>

A couple useful pages:

* [align_val_t](https://en.cppreference.com/w/cpp/memory/new/align_val_t), [new](https://en.cppreference.com/w/cpp/memory/new/operator_new), [free](https://en.cppreference.com/w/cpp/memory/new/operator_delete) on cppreference.com
* [SO description of relevant constants](https://stackoverflow.com/questions/59172291/order-between-stdcpp-default-new-alignment-and-alignofstdmax-align-t)
* [SO discussion of requirements for allocators](https://stackoverflow.com/questions/56713868/overloading-operator-new-with-smaller-default-alignment)

</details><HR/>

<details><summary>Prototypes of new/delete functions...</summary><P/>

If you are providing all the underlying `new()` and `delete` operators, then you might need to support the following alloc/free, for C++17 compliance:

```C++
// For C++11, only need the following:
void* operator new  ( std::size_t count );
void* operator new[]( std::size_t count );
void* operator new  ( std::size_t count, const std::nothrow_t& tag ) noexcept;
void* operator new[]( std::size_t count, const std::nothrow_t& tag ) noexcept;
void operator delete  ( void* ptr ) noexcept;
void operator delete[]( void* ptr ) noexcept;

// Since C++14, also need two more delete variants...
// but, can just forward to the C++11 versions and ignore
// the parameter ... it's only for optimizations
void operator delete  ( void* ptr, std::size_t sz ) noexcept;
void operator delete[]( void* ptr, std::size_t sz ) noexcept;

// Since C++17, there's four more each for new / delete, to support allocation
// of objects with alignment greater than __STDCPP_DEFAULT_NEW_ALIGNMENT__.
void* operator new  ( std::size_t count, std::align_val_t al );
void* operator new[]( std::size_t count, std::align_val_t al );
void* operator new  ( std::size_t count, std::align_val_t al, const std::nothrow_t& ) noexcept;
void* operator new[]( std::size_t count,  std::align_val_t al, const std::nothrow_t& ) noexcept;
void operator delete  ( void* ptr, std::align_val_t al ) noexcept;
void operator delete[]( void* ptr, std::align_val_t al ) noexcept;
void operator delete  ( void* ptr, std::size_t sz, std::align_val_t al ) noexcept;
void operator delete[]( void* ptr, std::size_t sz, std::align_val_t al ) noexcept;
```

</details><HR/>

Of course, I wonder what code is allocating an object which has an alignment requirement greater than `__STDCPP_DEFAULT_NEW_ALIGNMENT__`?

SpenceKonde, 2021-09-17:
Is the nothrow stuff appropriate for embedded systems which are compiled with the no-exceptions parameter?

henrygab, 2021-09-17:
Yes, in my opinion.

[`noexcept`](https://en.cppreference.com/w/cpp/language/noexcept_spec) means the function is declared as being non-throwing.

For compiling with no-exceptions parameter, the real question is, "Why is the compiler allowing use of all that code which is **_NOT_** declared as non-throwing?   😏  (likely answer: compatibility)

Said another way, because the use of `noexcept` indicates the function is non-throwing, there is a reasonable argument that this is a **more** appropriate declaration of the function for embedded systems / compilation that expect the function to be non-throwing.

But, that's just my take on it.  I'm not a compiler expert.

SpenceKonde, 2021-09-17:
Currently we have:
```
void * operator new(size_t size, void * ptr) noexcept;
```

Where the hell did this come from?

SpenceKonde, 2021-09-17:
Also. wait, 

#if (__cpp_sized_deallocation >= 201309L)

#if (__cpp_aligned_new >= 201606L)

What do those values represent? They don't seem like meaningful values.

henrygab, 2021-09-17:
They are meaningful values; they are defined by the C++ standards.

There's a general problem that was solved (starting with C++11) by the standard.

The problem: 

* How can code be written to conditionally support new features, in a standards-compliant way?

The answer (from C++11 through C++17):

* [Feature testing](https://en.cppreference.com/w/cpp/feature_test)

So, those numeric values are defined by the standard.  If those symbols are defined **_AND_** contain a value at least as large as noted, then the compiler supports that capability in conformance with the noted revision of the standard.

<details><summary>A simple made-up example of feature testing use</summary><P/>

```c++
// NOTE: this is made-up, for illustration purposes, and not tested
#if (__cpp_static_assert >= 201411L)
    // C++17 added single-argument static_assert()
    #define STATIC_ASSERT(x) static_assert(x)
#elseif (__cpp_static_assert >= 200410L) // C++11 
    // C++11 added two-argument static_assert()
    #define STATIC_ASSERT(x) static_assert((x), #(x))
#elseif  ... // look for gcc extensions, etc.
#else
    // No compiler support found for static_assert
    // use compiler trick such as negative array index, etc.
    #define STATIC_ASSERT(x)
#endif
```

</details><HR/>

Does that help?

SpenceKonde, 2021-09-18:
That's answering a different question and I don't see how it's related
I understand feature testing, I use it all the time in my cores to detect the capability of the hardware, almost every file in the core is testing for features of the hardware based on defines from the 

But what does 201309 mean? 0x3125D? Neither of those seem to make any sense as a number related to a sized deallocation?  What do they represent? I can't imagine that there's something magic about 201,309b as the size of something to be allocated, and the idea that alignment need different implementations at a slightly higher, but also seemingly random size (201,606 bytes? lol) seems implausible, and in that interpretation it would seem like neither of these should ever be true on am embedded platform where they exceed the sum of the total flash and ram of the device. Clearly I don't understand the defines mean, much less what units the values are in.....

henrygab, 2021-09-18:
I understand your question is what those magic constant numbers mean.  There really isn't any valuable meaning within them, outside of testing relative value with the feature testing defines.

The numbers are defined by the C++ specifications.  A summary of the [feature testing](https://en.cppreference.com/w/cpp/feature_test) values states:

> Each macro expands to an integer literal corresponding to the year and month when the corresponding feature has been included in the working draft.

* _Do these magic numbers have deeper meaning?_  No.  It's just an encoded date.
* _Can you use the magic value for other, meaningful, purposes?_  No.
* _Is there a header file that defines friendlier names for these magic numbers?_  I don't know, but I've not seen one.  I just lookup the magic numbers on https://en.cppreference.com/w/cpp/feature_test when needed.
* _Do these values have units?_  No.

It's not the answer you're looking for, but it's the only answer I can offer....

henrygab, 2021-09-18:
> ```
> void * operator new  (size_t size, void * ptr) noexcept;
> void * operator new[](size_t size, void * ptr) noexcept;
> ```
> Where [...] did this come from?

See [`placement new`](https://en.cppreference.com/w/cpp/language/new#Placement_new) for how it is used.

See [`operator new`](https://en.cppreference.com/w/cpp/memory/new/operator_new), specifically the two with footnotes 9 & 10.

<details><summary>Example implementation</summary><P/>

```C++
void * operator new  (size_t size, void * ptr) noexcept { (void)size; return ptr; }
void * operator new[](size_t size, void * ptr) noexcept { (void)size; return ptr; }
```

</details><HR/>

SpenceKonde, 2021-09-18:
Aaaah! the number reprepresents a date! Now the tests make sense!

SpenceKonde, 2021-09-19:
Huh, not sure how I missed that form while reading the lists, Okay then. 

I don't suppose you know to implement the rest of this list? Since I don't really understand classes, I'm a little foggy at how to implement this.

SpenceKonde, 2021-09-19:
No, wait, the header you suggested above *doesn't* list those two functions. Which version are they associated with, like, within with #if do they go? 

Also... 
Existing implementations are present for 
```c++
void * operator new(size_t size);
void * operator new[](size_t size);
void * operator new(size_t size, void * ptr) noexcept;
void * operator new[](size_t size, void * ptr) noexcept /* Not in new.h though??? */
void operator delete(void * ptr);
void operator delete[](void * ptr);
```
You've given me 
```c++
void * operator new  (size_t size, void * ptr) noexcept;
void * operator new[](size_t size, void * ptr) noexcept;
```
And for 
```c++
void operator delete(void *ptr, unsigned int);
void operator delete[](void *ptr, unsigned int);
```'
This has been suggested 
```c++
void operator delete(void *ptr, unsigned int) {
  free(ptr);
}

void operator delete[](void *ptr, unsigned int) {
  free(ptr);
}
```
though I'm pretty sure we want size_t not unsigned int; I figure I should also cast these size_t arguments to void to get rid of compiler warnings, as **the core shalt not produce warnings**, particularly since (unless I've screwed up) I intentionally broke the IDE's feature by which warnings could be disabled, because warnings usually convey information that saves a great deal of pain and debugging practices, and warnings are far more frequently bugs (or at least engraved invitations for wouldbe bugs that might seek to manifest) than they are spurious, things that should be at a minimum recognized, so when something turns out to not work, there are red flags above good places to look for them.

What do we do about:
```c++
  void* operator new  ( std::size_t count ); //already present
  void* operator new[]( std::size_t count ); // already present
  void* operator new  ( std::size_t count, const std::nothrow_t& tag ) noexcept; // ???
  void* operator new[]( std::size_t count, const std::nothrow_t& tag ) noexcept;  // ???
  void operator delete  ( void* ptr ) noexcept; //already present
  void operator delete[]( void* ptr ) noexcept; //already present
  void * operator new  (size_t size, void * ptr) noexcept; //already present
  void * operator new[](size_t size, void * ptr) noexcept; //already present (but missing from new.h???)
    void operator delete  ( void* ptr, std::size_t sz ) noexcept; //proposed
    void operator delete[]( void* ptr, std::size_t sz ) noexcept; //proposed
    void* operator new  ( std::size_t count, std::align_val_t al ); // ???
    void* operator new[]( std::size_t count, std::align_val_t al ); // ??? 
    void* operator new  ( std::size_t count, std::align_val_t al, const std::nothrow_t& ) noexcept;
    void* operator new[]( std::size_t count,  std::align_val_t al, const std::nothrow_t& ) noexcept;
    void operator delete  ( void* ptr, std::align_val_t al ) noexcept; // just cast al to void, and free(ptr)? 
    void operator delete[]( void* ptr, std::align_val_t al ) noexcept; // just cast al to void, and free (ptr)? 
    void operator delete  ( void* ptr, std::size_t sz, std::align_val_t al ) noexcept; // ???
    void operator delete[]( void* ptr, std::size_t sz, std::align_val_t al ) noexcept; // ???
```
And, beyond 
```c++
void * operator new  (size_t size, void * ptr) noexcept;
void * operator new[](size_t size, void * ptr) noexcept;
``` 
are there any others missing from that list?

SpenceKonde, 2021-09-19:
My current work in progress .cpp is

```
#include <stdlib.h>

// For C++11, only need the following:
void *operator new  (size_t size) {
  return malloc(size);
}
void *operator new[](size_t size) {
  return malloc(size);
}
void* operator new  ( size_t size, const std::nothrow_t& tag ) noexcept {
  /* ??? what to to about tag? */
  return malloc(size);
}
void* operator new[]( size_t size, const std::nothrow_t& tag ) noexcept {
  /* ??? what to do about tag? */
  return malloc(size);
}

void  operator delete  (void * ptr) {
  free(ptr);
}
void  operator delete[](void * ptr) {
  free(ptr);
}
/* does this get a conditional like the ones below or not? */
#if (1) 
  void * operator new  (size_t size, void * ptr) noexcept {
    (void)size; 
    return ptr; 
  }
  void * operator new[](size_t size, void * ptr) noexcept {
    (void)size; 
    return ptr; 
  }
#endif 

// Since C++14, also need two more delete variants...
#if (__cpp_sized_deallocation >= 201309L)
  void  operator delete  ( void* ptr, size_t size ) noexcept {
    (void) size;
    free(ptr);
  }
  void  operator delete[]( void* ptr, size_t size ) noexcept {
    (void) size;
    free(ptr);
  }
#endif

// Since C++17, there's four more each for new / delete, to support allocation
// of objects with alignment greater than __STDCPP_DEFAULT_NEW_ALIGNMENT__.
#if (__cpp_aligned_new >= 201606L)
    void* operator new  ( size_t size, std::align_val_t al ) {
    /* ??? what about al? */
    return malloc(size);
  }
    void* operator new[]( size_t size, std::align_val_t al ) {
    /* ??? what about al? */
    return malloc(size);
  }
    void* operator new  ( size_t size, std::align_val_t al, const std::nothrow_t& tag ) noexcept {
    /* ??? what about al and tag? */
    return malloc(size);
  }
    void* operator new[]( size_t size,  std::align_val_t al, const std::nothrow_t& tag ) noexcept {
    /* ??? what about al and tag? */
    return malloc(size);
  }
    void  operator delete  ( void* ptr, std::align_val_t al ) noexcept {
    /* ??? what about al? */
    free(ptr);
  }
    void  operator delete[]( void* ptr, std::align_val_t al ) noexcept {
    /* ??? what about al? */
    free(ptr);
  }
    void  operator delete  ( void* ptr, size_t size, std::align_val_t al ) noexcept{
    /* ??? what about al? */
    (void) size;
    free(ptr);
  }
    void  operator delete[]( void* ptr, size_t size, std::align_val_t al ) noexcept {
    /* ??? what about al? */
    (void) size;
    free(ptr);
  }
#endif
```

SpenceKonde, 2021-09-20:
Just a thought - are the conditionals needed? Why not just define em all? Wouldn;t they be optimized away anyway since they wouldn't be called unless the code required them, which would mean that it was supported? Or do we still get a flash penalty

Hm, for aligned deletes we want to just cast al to void like we do size right?

SpenceKonde, 2021-09-20:
And std:L:nothrow_t "doesn;t name a type" which I guess is why those versions don;t go in. 

So my remaining question is implementation of 
```
    void* operator new  ( size_t size, std::align_val_t al ) {
    /* ??? what about al? */
    return malloc(size);
  }
    void* operator new[]( size_t size, std::align_val_t al ) {
    /* ??? what about al? */
    return malloc(size);
  }
```

henrygab, 2021-09-20:
OK.... I'm really not an expert here.  But, I see you're following the [`operator new`](https://en.cppreference.com/w/cpp/memory/new/operator_new) prototypes, and the order they are listed on that page.  Here's my attempt to respond to questions from your [current work in progress](https://github.com/SpenceKonde/megaTinyCore/issues/528#issuecomment-922556849).

<details><summary>what about parameter "tag"</summary><P/>

> ```c++
> void* operator new  ( size_t size, const std::nothrow_t& tag ) noexcept {
>   /* ??? what to to about tag? */
>   return malloc(size);
> }
> void* operator new[]( size_t size, const std::nothrow_t& tag ) noexcept {
>   /* ??? what to do about tag? */
>   return malloc(size);
> }
> ```

It is an unused parameter.  I think it's used solely to differentiate the function signature between throwing and non-throwing `operator new` functions. 

</details><HR/>
<details><summary>Do these need conditional?</summary><P/>

> ```c++
> /* does this get a conditional like the ones below or not? */
> ```

No, the non-allocating placement allocation functions (#9, #10) do not need a conditional.  They are part of the C++11 standard too.

</details><HR/>

henrygab, 2021-09-21:
The question on super-aligned allocations is more in-depth, and applies to each of the `operator new` functions that take a `std::align_val_t` parameter.

Still, I will give it a shot....

<details><summary>The problem</summary><P/>

When one of these functions is called, the caller is expecting the resulting pointer that is returned to them to be aligned according to the `al` parameter (which must be a power-of-two).

Unfortunately, when `al` exceeds `STDCPP_DEFAULT_NEW_ALIGNMENT`, then `malloc(size)` is not guaranteed to provide a pointer that meets that alignment requirement.

For this example, the following requires 16-byte alignment:

```C++
typedef struct alignas(16) _DMA_BUFFER {
    uint8_t bytes[256];
} DMA_BUFFER;
```

However, the alignment guaranteed by `malloc()` may be 4 or 8.
</details></hr>

<details><summary>Solving ... part one</summary><P/>

When that 16-byte aligned structure is allocated, one could imagine the following:

```C++
void* operator new  ( size_t size, std::align_val_t al ) { /* ... */ }
// ptr = operator new(
//     size <-- sizeof(DMA_BUFFER) == 256
//     al   <-- std::align_val_t(alignof(DMA_BUFFER)) == 16
//     )
```

It's clear that `al` is greater than `STDCPP_DEFAULT_NEW_ALIGNMENT` in this example.

One solution to provide an aligned pointer, is to simply request a larger number of bytes than the caller requested.  Specifically, you would need to request at least `al - STDCPP_DEFAULT_NEW_ALIGNMENT` bytes extra from malloc, to ensure that you can find a pointer that meets the alignment requirement, and has at least `size` allocated bytes following that aligned pointer.

Here's an example that shows this:

```C++
void* operator new  ( size_t size, std::align_val_t al ) {

if (al <= STDCPP_DEFAULT_NEW_ALIGNMENT) {
    return malloc(size);
}

// how many extra bytes required to ensure can find an aligned pointer?
size_t toAllocate = size + (al - STDCPP_DEFAULT_NEW_ALIGNMENT);
void* allocated = malloc(toAllocate);

size_t mask = al-1; // alignment mask ... the bits that must be zero
void* alignedPtr =
    (allocated & mask) == 0 ?
    allocated :
    (allocated & (~mask)) + al
    ;

return alignedPtr;
}
```

* `toAllocate` was set to 268 bytes
* `malloc()` returned the pointer `0xDD74`, stored in `allocated`
* `mask` was `0x000f`
* `alignedPtr` was therefore set to `0xDD80 ((0xDD74 & 0xFFF0) + 0x10`
* and the caller got a result of `0xDD80`

Wonderful, right?  After all, the caller happily obtained the required aligned pointer, and it's large enough to write 256 bytes.

Well ... not quite wonderful....

</details></hr>

<details><summary>The second problem</summary><P/>

The harder part comes from the corresponding call that will later occur to `void operator delete (void* ptr, std::align_val_t al) noexcept;`.

Continuing the above example, the called got the pointer `0xDD80`, so that's what it will send back to the delete operator.  Unfortunately, forwarding that value to `free()` ends up causing all sorts of really weird system behavior.

The root cause is memory corruption.   The problem is, `0xDD80` was *NOT* the pointer that was allocated via `malloc()`, and thus should *NOT* be the pointer provided to `free()`.   Oops.

The problem is, given a pointer that is some number of bytes *past* the original pointer allocated via `malloc()`, how to determine the actual pointer that was returned?  And do so with minimal overhead....

</details></hr>

<details><summary>Solving ... part two</summary><P/>

The solution is not terribly difficult, once you've seen it done.  In fact, memory managers tend to do something like this....

Allocate even more memory than requested, adding enough for a structure that can indicate how many bytes "backwards" to look.  Two options are to store a `size_t` indicating the number of "wasted" bytes at the head of the allocation, or to store a `void*` indicating the original pointer value from malloc().

```c++
void* operator new  ( size_t size, std::align_val_t al ) {

if (al <= STDCPP_DEFAULT_NEW_ALIGNMENT) {
    return malloc(size);
}

// how many extra bytes required to ensure can find an aligned pointer?
size_t toAllocate = size + (al - STDCPP_DEFAULT_NEW_ALIGNMENT);
toAllocate += sizeof(uintptr_t);  // to store the original pointer

uintptr_t allocated = malloc(toAllocate);

// ensure to save space for the back-pointer.
uintptr_t alignedPtr = allocated + sizeof(uintptr_t);

size_t mask = al-1; // alignment mask ... the bits that must be zero
uintptr_t alignedPtr =
    (allocated & mask) == 0 ?
    allocated :
    (allocated & (~mask)) + al
    ;

// save the original pointer just before the pointer value returned to caller
static_assert( alignof(uintptr_t) <= STDCPP_DEFAULT_NEW_ALIGNMENT, "" );
uintptr_t* storeOriginalPointerAt = (uintptr_t)(alignedPtr - sizeof(uintptr_t));
*storeOriginalPointerAt = allocated;

return alignedPtr;
}
```

Walking through it again:
* sizeof(`uintptr_t`) == 2
* `toAllocate` was set to 276 bytes
* `malloc()` returned the pointer `0xDD74`, stored in `allocated`
* `mask` was `0x000f`
* `alignedPtr` was therefore set to `0xDD80 ((0xDD76 & 0xFFF0) + 0x10)`
* and the caller got a result of `0xDD80`

Then, this allows:

```c++
void operator delete (void* ptr, std::align_val_t al) noexcept {

if (al <= STDCPP_DEFAULT_NEW_ALIGNMENT) {
    free(ptr);
    return;
}

// Get the original pointer value from just prior to the provided pointer.
uintptr_t aligned = ptr;
uintptr_t tmp = aligned - sizeof(uintptr_t); // go back 2 bytes
uintptr_t original = *((uintptr_t *)tmp); // extract the original pointer

// sanity check the original value... for iot, infinite loop recommended
// as it's better than memory corruption!
size_t overhead = (al - STDCPP_DEFAULT_NEW_ALIGNMENT) + sizeof(uintptr_t);
assert(original < aligned);
assert(original + overhead <= aligned);

// and finally free the original memory
free(original);
}
```
</details><hr/>

Does that make sense?

pstolarz, 2021-09-21:
@henrygab Excellent approach for non trivial problem. The main issue I see here is - you assume `__STDCPP_DEFAULT_NEW_ALIGNMENT__` alignment for `malloc(3)` which I believe is not true. See

https://en.cppreference.com/w/c/memory/malloc
https://en.cppreference.com/w/c/language/object#Alignment

So we have [`std::max_align_t`](https://en.cppreference.com/w/cpp/types/max_align_t) rather than `__STDCPP_DEFAULT_NEW_ALIGNMENT__` and the values need not to be the same (see [here](https://stackoverflow.com/questions/56171482/why-does-the-c-standard-allow-stdmax-align-t-and-stdcpp-default-new-alignm)). In such case it would be wise to test malloc's returned pointer against assumed alignment via appropriate test. For example:
```cpp
assert(!(malloc_ptr & (alignof(std::max_align_t) - 1)));
```

The other issue is - for large alignments like 256, 512, 1024 etc. we must allocate additional unused "move-space" (like 256 - 8, 512 - 8, 1024 - 8 etc) to align malloc's allocated space by moving it forward up to desired alignment. I don't think using `malloc`/`new` allocations for such purpose is a wise move, especially in the embedded (resource constrained) area. If it's possible the better approach would be using `alignas()` for statically provided bytes-table or use linker script with a space devoted for such purpose.

**As a conclusion**: I don't think we should care about C++17 align-aware allocations at this moment. This is an extra feature which may but not need to be supported (as stated by `__cpp_aligned_new` feature macro). The only cases I would recommend to handle from [this list](https://en.cppreference.com/w/cpp/memory/new/operator_new) are: (1), (2), (5), (6), (9), (10) which are trivial to implement and suitable for most use cases. If in the future someone should have had a desire to use align-aware allocs @SpenceKonde may consider to add this (but I don't think it will ever happen).

henrygab, 2021-09-22:
@pstolarz ... First, excellent find of the [answer](https://stackoverflow.com/a/56171758/6471040) for why `std::max_align_t` and `__STDCPP_DEFAULT_NEW_ALIGNMENT__` are **_technically_** allowed to be different.

Practically, I would be surprised if `malloc()` doesn't ensure the larger alignment, but I have not found proof that this is required by the specification.  Thus, the following decision tree holds:

1. If `__STDCPP_DEFAULT_NEW_ALIGNMENT__ <= alignof(std::max_align_t)`, then things are "good".
2. Similarly, if it is guaranteed that `malloc()` ensures alignment of at least `__STDCPP_DEFAULT_NEW_ALIGNMENT__`, then things are "good".

"Good", as used above, means that only the alignment-aware operator new functions need to add overhead to ensure the resulting allocations have the required alignment.

Otherwise, if I am reading this correctly, ALL operator new functions would need additional logic to ensure the allocations are properly aligned.  While possible (and it should be validated), I would find not ensuring the "Good" state to be a terrible implementation decision.

I don't have a strong recommendation for whether Spence only implements the non-alignment-aware new functions.  There are good and bad aspects to avoiding implementing them:

👍 Less work upfront, and complies with the YAGNI design philosophy

👎 A project may stumble upon a very odd error, if accidentally causing reliance on this

Certainly, implementing (1), (2), (5), (6), (9), and (10) are recommended.  Is that enough?  Probably yes?  Hard to say for sure....

henrygab, 2021-09-22:
> The other issue is - for large alignments like 256, 512, 1024 etc. we must allocate additional unused "move-space" (like 256 - 8, 512 - 8, 1024 - 8 etc) to align malloc's allocated space by moving [...]

Well, if [`std::aligned_alloc()`](https://en.cppreference.com/w/cpp/memory/c/aligned_alloc) is available, that would resolve the difficulty.  Let the standard library deal with the complexity.

@SpenceKonde , can you confirm that `std::aligned_alloc()` is **_not_** available?

Because, if it is, then the over-aligned operator new functions can simply redirect to that function.

sicsim, 2021-09-22:
So what's the solution to this issue? I just got this issue after updating some libraries and am stuck on what to do next.

pstolarz, 2021-09-22:
> Otherwise, if I am reading this correctly, ALL operator new functions would need additional logic to ensure the allocations are properly aligned. While possible (and it should be validated), I would find not ensuring the "Good" state to be a terrible implementation decision.

I did some practical experiments with C++17 for `__STDCPP_DEFAULT_NEW_ALIGNMENT__` and `std::max_align_t`. The source code may be found [here](https://gist.github.com/pstolarz/31c46cabfcee15503d2f4c26b5e03714). It looks like (at least for gcc) both of these values are equal. Moreover the `malloc()` and `operator new()` returns memory pointers with alignments more strict than these values even for single byte allocs. As an example for x64 Linux gcc I have:
```
__STDCPP_DEFAULT_NEW_ALIGNMENT__: 16

alignof(uint8_t): 1
alignof(uint16_t): 2
alignof(uint32_t): 4
alignof(uint64_t): 8
alignof(max_align_t): 16
alignof(s256_t): 256

malloc(uint8_t) mem: 0x1948c20, algn: 32
malloc(uint16_t) mem: 0x1948c40, algn: 64
malloc(uint32_t) mem: 0x1948c60, algn: 32
malloc(uint64_t) mem: 0x1948c80, algn: 128
malloc(max_align_t) mem: 0x1948ca0, algn: 32
malloc(s256_t) mem: 0x1948cd0, algn: 16

new(uint8_t) mem: 0x1948de0, algn: 32
new(uint16_t) mem: 0x1948e00, algn: 512
new(uint32_t) mem: 0x1948e20, algn: 32
new(uint64_t) mem: 0x1948e40, algn: 64
new(max_align_t) mem: 0x1948e60, algn: 32
new(s256_t) mem: 0x1948f00, algn: 256
```
Similar tests may be performed on avr-gcc of course.

My approach is: use things simple (but not primitive) with incremental improvements if required in the future. If we may assume that for our environment `__STDCPP_DEFAULT_NEW_ALIGNMENT__ <= alignof(std::max_align_t)` is with high probability - lets assume it (and assert by `static_assert` in the code) and implement w/o additional overhead/complications.

Although case `__STDCPP_DEFAULT_NEW_ALIGNMENT__ > alignof(std::max_align_t)` will not lead to as much overhead as the over-aligned case described above, it still requires extra logic and handling. I wonder if we need to care about this, if for gcc we will never end up with this case (e.g. see Arduino AVR 1.8.3, megaAVR 1.8.7, SAM 1.6.12, SAMD 1.8.11, SAMD_beta 1.6.25 cores - all of these use simple association between `new` and `malloc()`)

> A project may stumble upon a very odd error, if accidentally causing reliance on this

Probability of such reliance is low or very low. But me may protect ourselves against that by implementing over-aligned new operators as mallocs (as @SpenceKonde originally proposed) along with asserts like:
```cpp
assert(!(reinterpret_cast<size_t>(ptr) & (al - 1)));
```
In case `malloc()` can't fulfill a request alignment requirement it will fire and a user will know about it. In this case such problem will have a chance to be reported as a bug and fixed by adding this functionality (if it ever happen).

Finally the decision belongs to @SpenceKonde as the library author/maintainer.

sicsim, 2021-09-22:
> So what's the solution to this issue? I just got this issue after updating some libraries and am stuck on what to do next.

Never, mind I got it to work by following the first comment.

henrygab, 2021-09-22:
> ```c++
> assert(!(reinterpret_cast<size_t>(ptr) & (al - 1)));
> ```
>
> In case malloc() can't fulfill a request alignment requirement it will fire and a user will know about it.

@pstolarz ... `assert()` is a **_run-time_** notification, which would leave the potential bug latent.  I do not recommend reliance on run-time assertions.  Even if you accept that risk, most assert macros do nothing outside of debug builds, and there is **NO** testing that will ensure the code is going to avoid hitting that assertion.  In other words, the assertion is used for an improper purpose, because you cannot assert the truth of that statement ... it's just wishful thinking.

The goal is to find **_compile-time_** assertions.  For example, the following ensures that the "hole" of added complexity doesn't exist:
```c++
// If this static assert fires, then it causes all `operator new`
// functions to perform additional steps to ensure proper alignment
// of any structure whose alignment is:
//  alignof(std::max_align_t) < alignof(T) <= __STDCPP_DEFAULT_NEW_ALIGNMENT__
static_assert(alignof(std::max_align_t) >= __STDCPP_DEFAULT_NEW_ALIGNMENT__);
```

With that `static_assert()` added, so long as no over-aligned structure attempts to get allocated, the alignment-aware `operator new` functions should never get called.  This is why I recommended only two options:

1. Properly implement the alignment-aware `operator new` functions, **-OR-**
2. Do not provide any implementation (only possible if compiles w/o them)

For some quick test code, I've created an example [gist](https://gist.github.com/henrygab/a93e49b9ab3ff42b14e9e75b014b4d25) (also viewable on [godbolt](https://godbolt.org/z/8zY1j17Yj)).

SpenceKonde, 2021-09-23:
These alignment aware options were added in C++ 17, right? Could any code that requires these functions when compiled with C++ 17 have compiled under earlier versions? (that is, can not having the alignment aware operators cause breakage of existing code that worked compiled successfully to an older standard) If so, how is this possible? 

Some of what's been said here is a little above my head (for example. I don't know how one could invoke say, an alignment aware new operator) so it's making it hard for me to judge whether we need to provide support. In my mind it seems like there are two possibilities for code that might require this; either:
a. It was invalid code prior to C++ 17. In this case, solution 2 from henry is obviously the correct approach. 
b. It was valid code prior to C++ 17 and worked despite the absence of alignment aware new and delete, but requires them when built to C++ 17

I'm trying to wrap my head around how b could happen. Was it getting unaligned chunks of memory, but, it was asking for alignment that it didn't truly require? Was it working only in special cases which happened to include the ones that were observed, and those applications would, under certain circumstances which in practice didn't occur (maybe the new operator was only called once, and luckily happened to align correctly, or the actual requirement wasn't actually *alignment* but some consequence of it, like that the low byte of the address not roll over), fail catastrophically? 

I will note that most stuff compiled just fine without those two new delete operators; it was the OneWireNg library alone that broke (in this case, unlike the alignment awareness, it's easy to see how code could have been broken, even without knowing that we quickly flush the size passed to the sized delete operator down the toilet and hand the pointer off to free(). So you can leave operators that aren't used out, and the compiler seems unconcerned by that until it tries to compile something that requires it. 

Those alignment aware operators look kinda ugly to implement to me? So I'm inclined to favor 2 on that grounds (Should they just not exist, or should I put a stub implementation that would generate a compile error with a message of our choosing via badCall(), same way we create an error when someone's code includes something like digitalWrite(100,LOW) on a device that doesn;'t have a pin 100? A message like "This code references operator delete  ( void* ptr, std::align_val_t al). This is a C++ 17-only feature, and is not implemented on megaTinyCore nor any official Arduino core as of Q3 2021" might be more useful?

henrygab, 2021-09-24:
_(Normally, people don't call these functions directly, the compiler calls the correct one.)_

I reach the same conclusion as you as to (b) ... Those functions are only called for over-aligned code.  I don't believe correctly written C++14 code could generate a call to the alignment-aware operator new functions.  If any such code existed and worked, then you listed the cause ... luck, or happenstance, or not actually needing the alignment specified.

Yes, if stuff compiles without requiring these: Great!  No point writing complex code that doesn't get tested.

I never heard of badCall() before.  If you can generate a useful compile-time error message, that'll be an AMAZING benefit, and removes the major concern about not implementing it.

<details><summary>If you want test code to exercise the alignment-aware new operator....</summary><P/>

Check out this [gist](https://gist.github.com/henrygab/a93e49b9ab3ff42b14e9e75b014b4d25) (also viewable on [godbolt](https://godbolt.org/z/fdvbe3KPe)).

In particular, the structure at line 203, "wasted4_t", is confirmed to call the over-aligned operator new.

</details>

pstolarz, 2021-09-24:
@SpenceKonde 
> I will note that most stuff compiled just fine without those two new delete operators; it was the OneWireNg library alone that broke

Thanks for pointing this out. It forced me finally to find the root cause of the issue.

**Root cause** OneWireNg uses virtual destructor, which in turn is a deleting-destructor with an `operator delete()` called under the hood. I'm not gonna to go into gory details in this topic since it's quite complicated. If you are interested in take a look [here](https://eli.thegreenplace.net/2015/c-deleting-destructors-and-virtual-operator-delete) and [there](https://stackoverflow.com/questions/31686508/why-is-delete-operator-required-for-virtual-destructors).

So to make the compiler happy `operator delete()` need always to be accessible to compile the library. Originally the core used C++11 (which is most common case for all toolchains known to me) and the core's `new.cpp` provided required `operator delete(void*)`. Most recently the core switched to C++17 which implies C++14's `operator delete(void*, size_t)`. The latter take precedence over `operator delete(void*)` for user's provided allocation replacements (see (5-6) variants [here](https://en.cppreference.com/w/cpp/memory/new/operator_delete)). It's worth to note here the 2nd argument in the `operator delete(void*, size_t)` is just a hint a memory allocator can use to be more efficient. Its standard library implementations is identical to (1-2).

**Conclusion** To protect OneWireNg against various cores dependencies causing issues like this one, I provided changes in the library to make it independent from any core in the future (see my changes [here](https://github.com/pstolarz/OneWireNg/pull/21)). I hope it finally solve the C++ new/delete operator issues.

**Digression** All the problems we discussed here, albeit interesting, just convinced me once more - C++ is simply bad language. It is over-complicated, badly designed and hard to grasp what's going on in case of problems like this one. I tend to agree with  Edsger Dijkstra that "to think of programming in C++ would make me physically ill".

SpenceKonde, 2021-09-24:
Yeah, I figured that it was called by the compiler, not directly. 

I'm not quite sure how to get from that gist to an Arduino sketch that would let me test the error message received. 

You don't know badCall? OH! You came from ATTinyCore, not megaTinyCore/DxCore. ATTinyCore 2.0.0 will have some use of it. 

badCall() and badArg() are a pair of do-nothing functions with the error attribute that I use all over the place. I don't really understand the syntax - but every variation I've tried (including attempts to get warnings instead) has failed without showing the desired message. The avrgcc manual is not exactly clear on these points. 

Anyway, they get called like badCall("You called a function you absolutely should never have called under these circumstances") or badArg("The argument passed to this function is known at compile-time to be invalid"); This approach is fundamentally opposite Arduino's approach to handling errors, which is to sweep them under the run and pretend they didn't happen. The typical result is that the problem line fails silently. You digitalWrite() a pin that doesn't exist? It does nothing; and with warnings turned off by default, the users are given basically no information on where to look for the problem other than their own intuition. Which because they're new is usually wrong. 

Definitions of these functions:
```
  void badArg(const char*) __attribute__((error("")));
  void badCall(const char*) __attribute__((error("")));
```

And some examples of how I use them:
```c
inline __attribute__((always_inline)) void check_valid_digital_pin(pin_size_t pin) {
  if(__builtin_constant_p(pin))
    if (pin >= NUM_TOTAL_PINS && pin != NOT_A_PIN)
    // Exception made for NOT_A_PIN - code exists which relies on being able to pass this and have nothing happen.
    // While IMO very poor coding practice, these checks aren't here to prevent lazy programmers from intentionally
    // taking shortcuts we disapprove of, but to call out things that are virtually guaranteed to be a bug.
    // Passing -1/255/NOT_A_PIN to the digital I/O functions is most likely intentional.
      badArg("Digital pin is constant, but not a valid pin");
}
/* SNIP */
void digitalWrite(uint8_t pin, uint8_t val) {
  check_valid_digital_pin(pin); 
  /* Get bit mask for pin */
  uint8_t bit_mask = digitalPinToBitMask(pin);
  if (bit_mask == NOT_A_PIN) {
    return;
  }
  /* Get port */
  PORT_t *port = digitalPinToPortStruct(pin);
```
Note that check_valid_digital_pin() only checks ones that are compile time known constants (that's all it can do, but we check if we have one with __builtin_constant_p() first; and because outcome is known at compile time, it gets optimized away by the compile and there's no overhead in terms of flash use). But because it only catches constants we still have to check that we can get a valid bitmask. digitalPinToBitMask() returns NOT_A_PIN for invalid pins, (so that catches both invalid pin numbers not known until runtime Unfortunately there's a fair amount of code out there that relies on being able to do that and have it ignores. I've not gotten any reports of other invalid pin numbers being intentionally passed). 
(quick example: `digitalRead(100);` will be caught (the most I/O pins any modern AVR has is 56, including both UPDI and reset). The following will not be:
```c
volatile uint8_t i_dont_change_the_value = 0;
volatile uint8_t bogus_pin = 100;
digitalRead(bogus_pin); // bogus_pin is declared volatile; The compiler and optimizer are not permitted to make assumptions about it beyond what is implied by the datatype) 
digitalRead(100+i_dont_change_the_value ); // 100 + a volatile 0 isn't compile time known erither. 
```

badCall() is used in cases where, regardless of what (if any) arguments are being passed, that function should not be getting called, such as the functions I use for runtime manipulation of the millis timer (this was implemented when I realized that they must exist if you want a library for the modern AVRs that can use millis timekeeping while awake and transparently transition to the on-chip RTC during sleep; writing such a library is a longstanding goal that I just haven't had time to work on - but a bunch of the plumbing is (the idea is that you'd have the RTC running, stop the millis timer, clear the RTC's INTFLAGS, enable it's... hmmm... if it's already running, I guess you'd want to set RTC.CMP to the current RTC.CNT, and enable the compare match interrupt, thus maximizing the time between waking up for timekeeping. The user would do all this by calling a wrapper function, which means we'd also be able to recognize our own "
```c
void stop_millis()
{ // Disable the interrupt:
  #if defined(MILLIS_USE_TIMERNONE)
    badCall("stop_millis() is only valid with millis time keeping enabled.");
  #else
  /* snip rest of function */
  #endif
}
void restart_millis()
{
  // Call this to restart millis after it has been stopped and/or millis timer has been molested by other routines.
  // This resets key registers to their expected states.
  #if defined(MILLIS_USE_TIMERNONE)
    badCall("restart_millis() is only valid with millis time keeping enabled.");
  #else
  /* snip rest of function */
  #endif
}
void __attribute__((weak)) init_millis()
{
  #if defined(MILLIS_USE_TIMERNONE)
    badCall("init_millis() is only valid with millis time keeping enabled.");
  #else
  /* snip rest of function */
  #endif
}
void set_millis(__attribute__((unused))uint32_t newmillis)
{
[... cut ...]
(it's also used in a bunch of other places.... I think on megaTinyCore analogReadDiff() gives that for 0/1-series because they don't have that capability). The point is that with the current selections in the tools menu (board + options) there is no way that that function could ever work, whereas badArg() indicates a function which would work fine if they would be so kind as to use valid arguments, typically for pin numbers, or where there's a limited number of valid options and it is nonsensical to call the function with other values). Of course, I defined those arbitrarily. I tried for a while to make an analog that would generate a warning. I was unsuccessful

SpenceKonde, 2021-09-24:
> @SpenceKonde
> 
> > I will note that most stuff compiled just fine without those two new delete operators; it was the OneWireNg library alone that broke
> 
> Thanks for pointing this out. It forced me finally to find the root cause of the issue.
Heh, as it happens I'd meant to edit that post and remove the "just" (though yours was the only one on which it was reported), as I am certain that other libraries would manifest this.  
> **Root cause** OneWireNg uses virtual destructor, which in turn is a deleting-destructor with an `operator delete()` called under the hood. I'm not gonna to go into gory details in this topic since [...]
Fascinating. Thanks. 
> So to make the compiler happy `operator delete()` need always to be accessible to compile the library. Originally the core used C++11 (which is most common case for all toolchains known to me) and [...]
Okay, that makes sense. You have @henrygab to thank for the C++17 change! He wanted it on ATTinyCore, but we don't have working CI there cuirrently, andwhile it's going into 2.0.0, I wanted to test it out on the best-tested core Which is this one. It took months for someone to find this.... 
> **Conclusion** To protect OneWireNg against various cores dependencies causing issues like this one, I provided changes in the library to make it independent from any core in the future (see my [...]
Probably wise, particularly if it's cross-platform. If it's AVR-specific , problems with cores are somewhat easier to solve, because either it's one of mine, it's one of MCUDude's, and we're both pretty active, and don't like to leave stuff broken. Or they're using someone else's core, in which case they deserve what they get ;-) 
> **Digression** All the problems we discussed here, albeit interesting, just convinced me once more - C++ is simply bad language. It is over-complicated, badly designed and hard to grasp what's going [...]
Hmmmm, I need to actually *do* it to become physically ill, but that may just be because I don't underrstand the finer point's of the ++ part of C++. C on the other hand, to the extent that one can denigrate it, is just kinda primitive (which, frankly, is what you want on an embedded system; often the connections between C and asm are pretty direct.

commit fd0415b0 (2021-09-26): fix #528
closed 2021-09-26 by SpenceKonde