SpenceKonde/megaTinyCore #531: i2c Problems Hanging state

JDJelectronics, 2021-09-25:
Dear Spence,

i have an attiny 3216, now the problem is that i am using 6 i2c chips for an ioT project. now the chip keeps crashing at wire.begin.

and also when I reset the device it does not go beyond the state. I really have to upload an empty sketch and then upload my own sketches before it does anything again. but that is only for a short time then it crashes again at wire.begin

how can I solve this?

I am using mega tiny core 2.4.2.

Regards JDJ.

SpenceKonde, 2021-09-26:
Is it a master or slave? If you are trying to do both, that isn't supported in 2.4.2

The specifics of the bug are not important however - there is a totally rewritten version of Wire now (huge thanks to @MX682X ) ready for testing, Please retest with the new version of wire.

I just checked in a version that is believed to be working. It drastically reduces flash consumption vs the version from 2.4.2 as well as adding support for 

Except for the case of being both a master and slave, you only have to replace the library folder with the new version. By default for board manager installation of core to non-portable Arduino IDE on windows, it's located located somewhere like 
```
C:\Users\Spence\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\libraries\Wire
```
(note that appdata is hidden - using this awful location for board manager installation was probably forced by microsoft's restrictions; I don't think they would have chosen this location otherwise.)

Do also be sure that you have appropriate external pullup resistors on SCL and SDA. These are not optional. Calling the usePullups() method before Wire.begin() will turn on the internal ones, but they are far weaker than the standard requires, and may be picky about the number of devices, length of the wires, and the phase of the moon; it should not be relied upon, and if calling usePullups() ever fixes a problem, that means you are missing the external

JDJelectronics, 2021-09-27:
HI Spence,

I have modified the core files with what you have provided. I just can't get it compiled. I have 2 option at TWI/I2C:

Master and slave (save flash & ram)

and master and slave.

Also I get this output error.

`rror` while detecting libraries included by C:\Users\JDJ Work\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\libraries\Wire\src\twi_pins.c

C:\Users\JDJ Work\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\libraries\Wire\src\twi_pins.c: In function 'TWI0_usePullups':
C:\Users\JDJ Work\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\libraries\Wire\src\twi_pins.c:27:0: error: unterminated #ifndef
 #ifndef TWI_PINS_H
 
C:\Users\JDJ Work\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\libraries\Wire\src\twi_pins.c:321:1: error: expected declaration or statement at end of input
 void TWI0_usePullups() {
 ^~~~

SpenceKonde, 2021-09-27:
Oh damnit, I tried to catch a corner case, - something so trivial it
couldn't break anything, and wound up breaking closures for conditionals,
fixing now.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

JDJelectronics, 2021-09-27:
Great that you're working on it.

do you know when there is an update?

SpenceKonde, 2021-09-27:
Sorry bout that, fix went in a couple of hours ago.

JDJelectronics, 2021-09-27:
The fix is working for the compiler. but the whole i2c does nothing and hangs.

mrWheel, 2021-09-27:
This looks a lot like my problem [#525](https://github.com/SpenceKonde/megaTinyCore/issues/525) 

I have replaced the 2.4.2 `Wire` map and `boards.txt `en `programmers.txt` with the new one from the master branche but the problem is still there.
After a `Wire.begin()` sometimes the code runs beyond the `Wire.begin()` but sometimes it does not.

With the new library that's still the case :-(

So now I want to enable the WDT right before that statement and disable it afterwards .. but I cannot find how to do that and the system 'hangs' indefinitely. Even power-off-on does not always help, but sometimes it does.

Did fond how to reset the WDT 
```
void wdt_reset() {
  __asm__ __volatile__ ("wdr"::);
}
```

Can you tell me how to enable and disable the WDT (little code example)?

Thanks!!!

SpenceKonde, 2021-09-27:
https://github.com/SpenceKonde/megaTinyCore#using-watchdog-to-reset-when-hung

should cover it. 

I would also suggest that you put a LED on an unused pin and set that output and then immediately prior to the call to begin, digitalWriteFast(pin,HIGH); to turn the LED on. Then first thing after begin, digitalWriteFast(pin,LOW). 

Trying to narrow down how begin could end up hanging, because I didn't think it did anything that should be able to hang. 

Do things work with simple bus configurations? Are these devices masters or slaves?  What values of pullup resistor are in use on SDA and SCL (be sure to count any ones located on the modules you're talking to...)

mrWheel, 2021-09-27:
Hi Spence,

Thanks for your answer and sorry for my ignorance.

On my I2C bus there are 3 mux IC’s (pcf8574). Pullup resistor is 2k2 and also tested with 4k7 (there are no “modules”).
All three pcf8574 have a `mux1.begin(hexAddress3)`, `mux2.begin(hexAddress2)` and `mux3.begin(hexAddress3)`.

There is a led that lights up before every begin() statement and dims after the begin() statement.
There are also print statements with flush() before and after.
The processor “hangs” somewhere in the begin() because if it does the led stays “on” and the second println() is not executed.

I read the WDT doc you mention in your reply (thats where I found the `wdt_reset()` function) but it is incomprehensible for me how to code it in such a way that I can enable and disable the WDT on demand. I have done it before for ATmega328 processor but that involved a lot of register handling (which I found on the internet).

Or am I missing something? Is there a `wdt_enable()` and `wdt_disable()` statement?
How long after a `wdt_reset()` does the WDT fire (in the program there are statements that can take a lot of time waiting for a reply).

MX682X, 2021-09-27:
@mrWheel are you using the arduino library for the PCF8574 by RobTillaart? If so, do you set the address in the constructor? Because it seems weird that you call begin with the address again.

mrWheel, 2021-09-27:
> @mrWheel are you using the arduino library for the PCF8574 by RobTillaart? If so, do you set the address in the constructor? Because it seems weird that you call begin with the address again.

Sorry, I use a class and misinformed you. I call the constructor with an address. In the constructor I call `Wire.begin()` without an address!

SpenceKonde, 2021-09-27:
@mrWheel  - two thoughts: 

1. Wait, you're calling Wire.begin() in a constructor? That doesn't seem like a good idea, it should be called in setup()....
2. Does it reproduce if you comment out all the references to the second and third PCF8574? code that could reproduce this in a simple case would be of great value. I think I even have some of those parts - isn't that what those stupid LCD I2C backpacks use? I have a bunch of those kicking around from when I decided to switch processors and was then able to use parallel control and lose the backpack. But  the easier this is to reproduce, the faster we can figure out what's happening. 

The fact that both classes have begin methods is making the above confusing to read. Is it a mux.begin(address) or Wire.begin()  is it hanging in? While it's hung, what voltage do you measure on the I2C lines (scope of course would be even better)

I'll look at the WDT section and see if there's some key piece missing,  I swear I had that covered.

JDJelectronics, 2021-09-27:
@mrWheel Hi!,

the WatchDog would be a godsend for me now for a temporary period.

I have more than 100 devices with customers! that have problems with the i2c. there are more every day.

If I had known that those Attiny chips had this I would have chosen another AVR chip.

hope Spence can fix this

MX682X, 2021-09-27:
Alright then... @mrWheel can you post your whole code then? It might reduce our confusion

MX682X, 2021-09-27:
also, @JDJelectronics could you post the code that has troubles?

SpenceKonde, 2021-09-27:
> I have 2 option at TWI/I2C.

Right, but in your code, which mode are you using it in? 

> The fix is working for the compiler. but the whole i2c does nothing and hangs.
When does it hang? What? Wire.begin() (use the LED check 

> I have more than 100 devices with customers! that have problems with the i2c. there are more every day.
How have we gone from issue first created just a few days ago to a huge volume of problems so quickly? What has changed? 

I will also repeat to you my question about pullups. I frequently get inquiries from users who have omitted these. Do you have pullups? What values? As I asked mrWheel, when it's in the hung state, what voltage do you measure on SCL and SDA?

@MX682X 
There seems to be a sudden uptick in I2C issue reports since 2.4.x, when the baud rate calculation was reworked,. Under default conditions, are we sure it's behaving sensibly w/regards to the master baud rate?

MX682X, 2021-09-27:
also also: If you have access to an UPDI debugger: You can use microchip studio to debug Arduino sketches,. To do that, you have to select the File->Open->Open object file for debug. There you can select the file to open. It is saved in AppData/Local/Temp/arduino_buildxxxxxx (x are some numbers) after hitting the compile button in the Arduino IDE. It would significantly help locating the problem if you could say where exactly it hangs.

mrWheel, 2021-09-27:
> 1. Wait, you're calling Wire.begin() in a constructor? That doesn't seem like a good idea, it should be called in setup()....

Ok, I have changed my code to only call the Wire.begin() once in setup.
Device is under test now and see what the outcome is tomorrow. Keep you posted!

> The fact that both classes have begin methods is making the above confusing to read. Is it a mux.begin(address) or Wire.begin() is it hanging in? While it's hung, what voltage do you measure on the I2C lines (scope of course would be even better)

If it hangs again after my changes I will measure with a scope the voltages on SDA and SCL and report back.

> I'll look at the WDT section and see if there's some key piece missing, I swear I had that covered.

That would be nice!

@JDJelectronics 
Are you using code from my github? If so you should comment-out the `Wire.begin()` in the constructor and code one call to `Wire.begin()` in the `setup()` function!

mrWheel, 2021-09-27:
> Alright then... @mrWheel can you post your whole code then? It might reduce our confusion

Well, thats 1416 lines of code ...... 
My hopes are on the multiple call's to `Wire.begin()` that I have now eliminated to only one call in `setup()`.

If that indeed solves the problem I do apologize to you guys!

mrWheel, 2021-09-27:
@MX682X 
> also also: If you have access to an UPDI debugger: You can use microchip studio to debug Arduino sketches.

I have not :-(

SpenceKonde, 2021-09-27:
@mrWheel - that is a big detail that I'd missed - so you're saying this issue happens rarely - too often to live with, but rarely enough that it takes a long time (overnight) to reproduce? 

What I find rather disturbing is that there is nothing that I can find in Wire.begin() that plausibly looks like a place it could be getting stuck.... there's no loop waiting for bits to be in some state, I don't see any obvious place that an interrupt is being enabled where an interrupt might be triggered but handled improperly or sent to badISR.... 

Another LED test that might be useful - remove those writes  from around begin, and instead at start of setup() do
```
pinMode(pin,OUTPUT);
digitalWriteFast(pin,CHANGE);
```
Good way to see if the failure puts it into a reset loop or not. Better still if youve got two leds on different pis that can be simultaneously used to monitor both, of course.

MX682X, 2021-09-27:
> If that indeed solves the problem I do apologize to you guys!

This makes me worry though, since I thought that I've written the code in a way that it will not be  a problem if called multiple times. I have a theory, but I have no idea why exactly it creates trouble...

[src.zip](https://github.com/SpenceKonde/megaTinyCore/files/7238666/src.zip)

I've modified three files of the Wire library that you can find in the zip file. Please try to replace them in your apropriate folders and try to call Wire.begin() multiple times. I'd like to see if that caused the hang or not.

mrWheel, 2021-09-27:
> @mrWheel - that is a big detail that I'd missed - so you're saying this issue happens rarely - too often to live with, but rarely enough that it takes a long time (overnight) to reproduce?

Well, not really.
 
When I flash the firmware it, most of the time, starts correctly with some leds flashing "slow" to indicate the beginning of setup and before the Wire.begin() the Leds turned "on" and after it is turned "off" again at the end of setup() the led flashes "fast".  

The strange thing is that sometime after power-off-power-on or reflash it all comes to a stop at Wire.begin() and the Led stay's on. At that moment you can do what you want but it will not start up normaly again. Not if you reflash, not if you power-off-on and not if you press [reset] (interrupt). And the moment you give up all hope .. it works flawless for a long time. Until when it wakes up after sleep it hangs again...

But wait!!!

I now narrowed it down to this function:

```
//===========================================================
bool PCF8574::begin(uint8_t address) 
{
  digitalWriteFast(PIN_PC0, HIGH);
  _address = address;

  //Wire.begin(); //-- <<-- Call this once in setup()!!!!!
  if (readPin()) 
  {
    digitalWriteFast(PIN_PC0, LOW);
    return true;
  }
  else
  {
    digitalWriteFast(PIN_PC0, LOW);
    return false;  
  }
  
} //  begin()
.
.
//===========================================================
bool PCF8574::readPin() 
{
  Wire.requestFrom(_address, (uint8_t) 0x01);
  uint32_t startTime = millis();
  while ((Wire.available() < 1) && ((int)(millis() -startTime) <= 200)) {;}
  if ((int)(millis() -startTime) > 200)
  {
        return false;
  }
  _PIN = Wire.read();
  return true;

} // readPin()
```
I will investigate further tomorrow. It's now time to go to bed..

JDJelectronics, 2021-09-27:
I have used all 2 mode of TWI/I2C

the problem has been going on for a while but couldn't find the problem. the devices have worked non-stop for 3 months. now they fall away.

I couldn't put my finger on what the problem is, initially I thought it was an error with the battery system.

so I took 1 off the field for further investigation. and conclude that the system crashes at Wire.begin.

Yesterday I did an update to the device from 2.3.2 to 2.4.2 that didn't work. Then I did a rollback to 2.3.2 but that obviously didn't help but I thought I can try it.

there is a 4K7 pull up resistor on the SDA and SCL line.

I would like to share the code but I can't because of permissions.

SpenceKonde, 2021-09-27:
Interesting - Is it a master Wire.begin() or a slave one? 

What is the state of the bus lines while it's hung? 

And you say that, *with no changes anywhere* they have started failing after 3 months? And previously, they never did?

SpenceKonde, 2021-09-27:
There has been a fair amount of generally tweaking of Wire.h in 2.3.x and 2.4.x because the old implementation was bloated (the folks on the small flash parts hated it, for obvious reasons) and lacking a very often requested feature (that feature being acting as both a master and a slave in the same sketch)

SpenceKonde, 2021-09-27:
>The strange thing is that sometime after power-off-power-on or reflash it all comes to a stop at Wire.begin() --> no!! it stops at PCF8574::begin()!! and the Led stay's on. From that moment you can do what you want but it will not start up normally again. Not if you reflash, not if you power-off-on and not if you press [reset] (interrupt). And the moment you give up all hope .. it works flawless again for a long time. Until when it wakes up after sleep it hangs again...

I'll reply later in your thread as I'm not convinced these are identical problems.

mrWheel, 2021-09-27:
> > The strange thing is that sometime after power-off-power-on or reflash it all comes to a stop at Wire.begin() --> no!! it stops at PCF8574::begin()!! and the Led stay's on.
> 
> I'll reply later in your thread as I'm not convinced these are identical problems.

I have not had the problem with `Wire.begin()` anymore since I only call it only once in `setup()`

JDJelectronics, 2021-09-27:
they are battery powered devices and i think this behavior has always been there but never noticed. This is because the battery has never been empty. it was only after the battery ran out that problems started to arise in the Wire.begin. if they work 1 time without an empty battery, there is nothing wrong I think.

I think it's fine with 1x flashing until the battery runs out that something goes wrong in the core.

resetting a device with a full battery also gave me the problem.

it has never been noticed until now.

mrWheel, 2021-09-27:
> @mrWheel are you using the arduino library for the PCF8574 by RobTillaart? If so, do you set the address in the constructor? Because it seems weird that you call begin with the address again.

No, not a library from RobTilaart but a class I made myself with a lot of help from bits and pieces from the internet ..

MX682X, 2021-09-28:
Ok, so I've learnt three things now:
A. JD's problem has nothing to do with my implementation of Wire since it persists across completely different implementations
B. JD's problem has nothing to do with mrWheel's problem.
C. mrWheels problem lies in Wire.requestFrom(), not in Wire.begin()

About C: There was a bug in requestFrom in my implementation that would keep the CPU in an endless while-loop if there was no ACK after writing the slave address. This can happen if no slave with the address is on the bus, there are no pull-ups, or the supplied address is wrong. I've published a hotfix for that a couple of days ago.

MX682X, 2021-09-28:
JD's problem is related to low voltage glitches from what I can read, your Problem is somewhere on the bus. If you have the files from my src.zip, you have the hotfix. 
At this point, requestWire() should only hang up if there will be no data sent from the slave.

mrWheel, 2021-09-28:
@MX682X :
> At this point, requestWire() should only hang up if there will be no data sent from the slave.

Why does it has to hang? Is there no (working) `Wire.setTimeout()` function?

MX682X, 2021-09-28:
No, there is not. Never was as far as I know. 
That's why I haven't implemented it. Since requestFrom is a blocking function and the amount of bytes should be known beforehand I didn't see a real reason why it should be needed. Plus I didn't see a nice way to do it, afterall I tried to keep the library as small as possible.

mrWheel, 2021-09-28:
@MX682X:

Update
======
I have installed the hotfix from MX682X and I have the `Wire.begin()` again in the `PCF8574` class. `Wire.begin()` is executed for every pcf8574 chip I have connected (3).
I have tested like crazy but my board will not crash anymore (but that is a behaviour I have observed before. No guarantee the problem is gone).
I have also tested without the repeated call to `Wire.begin()` (only call it once in `setup()`)and that also does not crash the board at this point in time! 

I will continue to test today and will keep you posted if something strange happens again!

SpenceKonde, 2021-09-28:
mrWheel - What is the voltage on the pins in the failed state? I seem to remember from when I was working with I2C routinely that I2C slaves would sometimes react to adverse conditions  by inappropriately stretching the clock (forever).... And didn't you say that you needed to power cycle the whole thing to make it work again? That would also suggest that a slave was holding the bus low.   I'd try to arrange to disconnect the I2C lines from the slaves without disconnecting power, so if one or both are low, you can be sure which side is holding it down.

mrWheel, 2021-09-28:
> mrWheel - What is the voltage on the pins in the failed state?

SDA 5Volt, SCL 1.8Volt

> And didn't you say that you needed to power cycle the whole thing to make it work again? 

Well, that sometimes makes it work again but for the last 12 minutes and twice re-flashing it still will not go past the `requestFrom()` ...

JDJelectronics, 2021-09-28:
The problem is definitely caused by a dead battery.

[image]
[image]
but it now also occurs with a full battery

the battery is a 12volt 7amp lead battery

I use the described documentation for the reset

void resetViaSWR() {
  _PROTECTED_WRITE(RSTCTRL.SWRR,1);
}

following the message from @mx682 I also discovered that the problem is probably also in the Wire.requestFrom().

JDJelectronics, 2021-09-28:
The problem is definitely caused by a dead battery.

but it now also occurs with a full battery

the battery is a 12volt 7amp lead battery

I use the described documentation for the reset

void resetViaSWR() {
  _PROTECTED_WRITE(RSTCTRL.SWRR,1);
}

following the message from @mx682 I also discovered that the problem is probably also in the Wire.requestFrom(
[image]
).

mrWheel, 2021-09-28:
What I see on the Scope when al goes well
[image]
This is the "Wire.begin()"

[image]
This is just using the bus

mrWheel, 2021-09-28:
It definitely hangs at requestFrom()! Even after the `mux.begin()` (without the `Wire.begin()` in it) the code hangs on just reading of pins .. which all go through `readpin()`

mrWheel, 2021-09-28:
This is the class I use to communicate with the pcf4874's

```

#ifndef PCF8574_H
#define PCF8574_H

#define ledOn   digitalWriteFast(PIN_PC0, HIGH);
#define ledOff  digitalWriteFast(PIN_PC0, LOW);

#include <Wire.h>

#define I2CWRITE(x) Wire.write(x)
#define I2CREAD()   Wire.read()

/******************************************************************/
class PCF8574 
{
public:
  PCF8574();

  bool      begin(uint8_t address = 0x21);
  void      pinMode(uint8_t pin, uint8_t mode);
  void      digitalWrite(uint8_t pin, uint8_t value);
  uint8_t   digitalRead(uint8_t pin);
  uint32_t  pulseIn(uint8_t pin, uint8_t state, uint32_t timeout = 5000);

protected:
  volatile uint8_t  _PORT;
  volatile uint8_t  _PIN;
  volatile uint8_t  _DDR;
  uint8_t           _address;
  bool    readPin();
  void    updatePin();
};

//===========================================================
PCF8574::PCF8574() :
    _PORT(0), _PIN(0), _DDR(0), _address(0)
{
}

//===========================================================
bool PCF8574::begin(uint8_t address) 
{
  _address = address;

  //--Wire.begin(); //-- <<-- Call this once in setup()!!!!!
  if (readPin()) return true;
  
  return false;  
  
} //  begin()

//===========================================================
void PCF8574::pinMode(uint8_t pin, uint8_t mode) 
{
  switch (mode) 
  {
    case INPUT:
        _DDR &= ~(1 << pin);
        _PORT &= ~(1 << pin);
        break;

    case INPUT_PULLUP:
        _DDR &= ~(1 << pin);
        _PORT |= (1 << pin);
        break;

    case OUTPUT:
        _DDR |= (1 << pin);
        _PORT &= ~(1 << pin);
        break;

    default:
        break;
  }

  updatePin();
  
} //  pinMode()

//===========================================================
void PCF8574::digitalWrite(uint8_t pin, uint8_t value) 
{
  if (value)  _PORT |= (1 << pin);
  else        _PORT &= ~(1 << pin);

  updatePin();
  
} //  digitalWrite()

//===========================================================
uint8_t PCF8574::digitalRead(uint8_t pin) 
{
  readPin();
  return (_PIN & (1 << pin)) ? HIGH : LOW;

} //  digitalRead()

//===========================================================
bool PCF8574::readPin() 
{
  ledOn;
  Wire.requestFrom(_address, (uint8_t) 0x01);
  ledOff;
  uint32_t startTime = millis();
  while ((Wire.available() < 1) && ((int)(millis() -startTime) <= 200)) {;}
  if ((int)(millis() -startTime) > 200)
  {
    return false;
  }
  _PIN = I2CREAD();
  return true;

} // readPin()

//===========================================================
void PCF8574::updatePin() 
{
  uint8_t value = (_PIN & ~_DDR) | _PORT;

  Wire.beginTransmission(_address);
  I2CWRITE(value);
  Wire.endTransmission();
  //-- needs at least 1.3us free time between start and stop
  delayMicroseconds(2);
  
} // updatePin()

#endif
/* EOF */
```

MX682X, 2021-09-28:
@mrWheel thank you for posting the osciloscope pictures. From what I can see either your SDA or SCL voltage is out of recomendated specifications for the PCF8574 and the 3216. As far as I can see, the SDA Voltage is 3.3V and SCL is 5V. If the VCC is 5V, it needs at least 3.6V to be in the recomended area. That might explain why it works only sometimes.

mrWheel, 2021-09-28:
> @mrWheel thank you for posting the osciloscope pictures. From what I can see either your SDA or SCL voltage is out of recomendated specifications for the PCF8574 and the 3216. As far as I can see, [...]

Are you sure?
Following three screen shots are all from the same capture.
Vertical scale is 2Volt/Div so [Blue]SCL ~ 5 Volt and [Yellow]SDA ~ 4.2 Volt.

Time base 200 microSec/Div:
[image]

Time base 100 milliSec/Div:
[image]

This one is zoomed in so the screen is divided in two:
[image]

(or am I missing something?)

MX682X, 2021-09-28:
> (or am I missing something?)

yes, the previous picture had a smaller ampitude 
[image]

But I do wonder, where do theese peaks come from?

[image]

mrWheel, 2021-09-28:
> yes, the previous picture had a smaller ampitude !

In the picture with the “lower amplitude” the screen is horizontal divided in two. So the amplitude is in real twice the size in the picture..

> Is it possible to zoom into theese? I wonder why the oscilloscope says NACK...

The time base is 100ms while the time base with the more readable data is 2ms. I think the “spike” is just compressed data. The decoder is not capable to decode that compressed data.

MX682X, 2021-09-28:
> In the picture with the “lower amplitude” the screen is horizontal divided in two. So the amplitude is in real twice the size in the picture..

If it were the case, you'd have more then 6V on SDA. Also, the SCL Line shows the 5V correctly

SpenceKonde, 2021-09-28:
mrWheel  - 
Regardless of which line it is and what the specific voltage is, why are they not both going up to the same value? I am hard pressed to think of any situation where that is possible with correct wiring and good connections...  One of those two, when idle, looks to be getting pulled up to the supply rail. The other, since the voltage is slightly different.... is getting pulled up to.....? a different supply rail?! What?
The 1.8v SCL hang is weird too. With a 2.2k pullup to... 5v? on the line, 3.2v drop.... something was pulling 5.1mA?  which is weird (hmm, and it persisted across all sorts of attempts to revive....). 
Loose wires? cold solder joints? crap cables? do any of the scope traces jump around while handling, poking, or tapping it?  different voltages at different points on the supply rail due to poor connections or hair-thin wire used to supply power? (If there's dupont line (particularly the kind that was made from ribbon cable - that stuff is bad news) or stuff that was crimped at home using an inappropriate crimp tool) involved, or a breadboard, those are all things to look at. Dodgy sockets or connectors?

SpenceKonde, 2021-09-29:
@JDJelectronics - to add to above, I recommend to you the same trick of disconnecting I2C without power is any lines are being held low,, to find out what device has malfunctioned. 

In your case, if I had to guess,... i'd say the order that things are powering on is relevant, or something breaks when you try to init the second time.

mrWheel, 2021-09-29:
> @mrWheel Regardless of which line it is and what the specific voltage is, why are they not both going up to the same value? I am hard pressed to think of any situation where that is possible with
>..
> is bad news) or stuff that was crimped at home using an inappropriate crimp tool) involved, or a breadboard (, those are all things to look at. Dodgy sockets or connectors?

The circuit is soldered on a PCB so no dupont wires, dodgy sockets or inappropriate crimp tools.

> In general: Wire.beginTransaction() doesn't have any while loops in it. _That shouldn't be able to hang._

I’m beginning to suspect some hardware (design) issue like maybe, as you wrote, a hair-thin shortage on the PCB. Strange thing though is that I have the same problem on multiple boards.

> I’d say the order that things are powering on is relevant, or something breaks when you try to init the second time.

Is it???
Two of the pcf8574’s are powered by a high-side switch to lower the current in sleep mode. After sleep they are powered on again. Could that really be the problem? Will investigate!

SpenceKonde, 2021-09-29:
> Is it??? Two of the pcf8574’s are powered by a high-side switch to lower the current in sleep mode. After sleep they are powered on again. Could that really be the problem? Will investigate!

I was talking to JD, but I would have immediately questioned you on this had you mentioned it! 

My initial response was "no way would that work at all" But, taking one of those stupid I2C backpacks to the character LCDs with the same chip (bought a bunch for a project. right around when the 328pb became common. I'd initially planned to use 328p + backpack, and I wasn;'t happy with the mechanics but what was I going to do? Didn't quite have pins for parallel mode, but the price jump to a larger chip was big.... Then 328pbs came into my awareness, swapped to that since it had the extra pins I needed. 
 and put the backpacks in my character LCD drawer
[image]
and the damned things have been there ever since (note the LCDs that came with them preinstalled too - not fun to desolder without damaging the LCD board, because those pads are very easy to lift), and they have been there ever since. So I located and removed the pullups on the board, and removed the pot, I then dialed my bench supply to 5v. puit ground to it's ground, and clipped one side of a 1k resistor to 5v. Negative of DVM to groundl positive to other side of resistor. touched it to the SCL and SDA pins expecting them to get instantly pulled down.... but they weren't loaded at all by it! It looks like that part... permits you to connect voltages way above what itlets you put on it's normal I/O pins on SCL./SDA? Really?

SpenceKonde, 2021-09-29:
I don't see that mentioned in the darasheet, but with no drop *shrug* okay. that seems unusual,.... 

https://www.ti.com/lit/ds/symlink/pcf8574.pdf?ts=1632812683687

Are your powercycles of the port expanders consistent with section 10?  They have specs on how fast or slow the power can come on. Curiously no specs on how large the decoupling caps need to be. 

You're not applying any voltages to the other pins while the devices are powered off are you? That most certainly is not permitted.

mrWheel, 2021-09-29:
> I don't see that mentioned in the datasheet, but with no drop shrug okay. that seems unusual,....
> https://www.ti.com/lit/ds/symlink/pcf8574.pdf?ts=1632812683687
> Are your powercycles of the port expanders consistent with section 10? They have specs on how fast or slow the power can come on. Curiously no specs on how large the decoupling caps need to be.

The timing is way beyond the values in section 10. I see no problems there.

> You're not applying any voltages to the other pins (like the 8 port pins you get) while the devices are powered off are you? 
> That most certainly is not permitted. And I poked them as a sanity check and the datasheet does make it explicitly clear that they can't have voltages applied to them when not powered.

Well, except for the SDA and SCL that are connected to the bus at all time.
But as a test I have switched the high-side switch "on" all the time and still have the problem. 
The high-side switch is driven by the AT3216 and during software reset the voltage on the pcf8574's drop to 3 volt ..(?)

So, just eliminated the high-side switch to the pcf's and what do you think? It seems to work again.
Don't hold your breath though. Seen it work before until it was broke again.

But I will keep you posted!

MX682X, 2021-09-29:
> The high-side switch is driven by the AT3216 and during software reset the voltage on the pcf8574's drop to 3 volt ..(?)

the PCF aswell as many otheer devices like the ATTiny have ESD Diodes on Pins to protect the internals from static discharges. 
Bsically GND ---|>|--- PIN --- |>| --- VCC
Theese have a forward voltage, in case of the PCF, min 1.2V.
When the high-side switches are off while the SDA and SCL lines are powered, the current can flow through the Pull-Ups on SDA/SCL, through the ESD Diodes to the VCC pin and power it that way. 

 I've seen this behaviour on an OLED that still worked even without the VCC connected.
 
 A dirty but quick way to test it would be by setting SDA/SCL pins low (by making them Outputs) for 100ms or so after a software reset and before Wire.begin() is called.

mrWheel, 2021-09-29:
@MX682X 
> A dirty but quick way to test it would be by setting SDA/SCL pins low (by making them Outputs) for 100ms or so after a software reset and before Wire.begin() is called. I haven't tried this yet though, so i can not gurantee that it will work.

in `setup()` I have this:
```
void setup()
{
  pinMode(PIN_SCL, OUTPUT);
  digitalWriteFast(PIN_SCL, LOW);
  pinMode(PIN_SDA, OUTPUT);
  digitalWriteFast(PIN_SDA, LOW);
 .
 .
```
It takes longer then 100ms before `Wire.begin()` is executed but it has no effect.

There is definitely something serious wrong with my design and setup.
Back to the drawing board.

I do apologize to you guys for eating up your time.

SpenceKonde, 2021-09-29:
Huh, regarding the protection diodes, that's what I was expecting to find! But experimentally, I don't see the expected behavior on a PCF connected to power supply ground but not the supply. 
But I found that, across a 1k resistor (simulating a pullup) from +5v to one of the data lines, ground connected but nothing else.... there was no voltage drop across the resistor, indicating no current flowing and hence, that the SCL and SDA lines don;'t  have diodes... the port pins do, and connecting the same resistor end to those, I saw a 1.x volt drop across the resistor ending up with 3.x volts on the side connected to the pin (which is what I expect when theres a protection diode.

mrWheel, 2021-09-29:
> Huh, regarding the protection diodes, that's what I was expecting to find! But experimentally, I don't see the expected behavior on a PCF connected to power supply ground but not the supply.

For me the problem has nothing to do with the pcf’s not being connected to +5v as I have removed the high-side switch and the pcf’s are now direct connected to the power rail (+5volt) but with the same erratic behaviour.

MX682X, 2021-09-30:
> I do apologize to you guys for eating up your time.

Well, this whole ordeal made me stare at the code long enough to figure out a graceful way to add a timeout, so there was a gain from that at least... Anyway, the library is still in alpha, and what better way to test it then on a buggy setup, huh?

[src.zip](https://github.com/SpenceKonde/megaTinyCore/files/7257704/src.zip)

`requestFrom()` and `endTransmission()` return the amount of bytes that were successfully written/read. If the returned value does not correspond to the expected value, there was an error which migh have lead to a timeout. The timout is a #define and not changable in the sketch (yet). Right now it's 10ms between two bytes.

SpenceKonde, 2021-09-30:
You cannot assume that millis is enabled - people particularly on smaller parts often disable it to save flash, using delay() for timing (the delay implementation on parts with millis disabled is  is really small). Or they disable it because it eats up a timer, and they can't spare the timer. Some of these parts have only two damned timers, others only 3, and I have codebase that I'm developing a bunch of things on, it needs 2 TCBs and a TCA and two CCL LUTs  for basic operation.... The more advanced versions will also need at least another TCA or TCD and 2 more LUTs.  That blocks up every timer on low pincount DD! 

I had a much less sophisticated idea though, 
What I was thinking was (starting from what is in the repo now) at each while loop that we could hang on

```
uint16_t timeout = 0; 
while (--timeout && SOME_PERIPHERAL_REGISTER != SOME_CONSTANT);
if (!timeout) goto theEnd;
```
or maybe:
```
uint16_t timeout = 0; 
do { 
  if (!(--timeout)) goto theEnd;
} while (SOME_PERIPHERAL_REGISTER != SOME_CONSTANT)
```

Would have to check which one compiled into something that was more sane. 

then subsequent loops could just set timeout back to zero and use the same trick. 

predecrementing a will immediately roll it over (unsigned so it;s legal), so you'd get 65k passes through the loop.. and

```
while (SOME_PERIPHERAL_REGISTER != SOME_CONSTANT); 
```

is gonna be like 6 clocks per cycle (LDS - 3 clocks to load using a direct address, cpi 1 clock to compare, brne that will be true to go to start of loop if they're not equal. 

with my addition. -- would be  SBIW - 2 clocks (subtract immediate from word), then 1 more for a breq that isn't true, so probably 9 clocks per loop, 589k clock cycles, making it a bit under 25ms at 24 MHz, which would be plenty long.  Might instead end up at 10 clocks if the target was more than 64 instruction words away (hence out of range of conditional branch) the compiler would instead do brne .+2 rjmp theEnd (ie, skip the rjmp unless we reached 0 with that last decrenet, so the conditional branch would usually be true and take 2 clocks). And even on an E-spec (125 C temp range) DB-series with external clock, the most these parts are going to do is 48 MHz, leaving us with just over 12ms timeouts, and you think 10ms is okay. At low clock speeds we'd want to dial that down of course. 

If you need to take some action to deal with the fact that it timed out (do you?), the timeout could be given it's own label to goto, immediately before theEnd.  if it was, then you go there through a timeout instead of by success..... I reckon the flash overhead is like 3-4 instruction words for each loop, plus at worst each time we reset timeout, which is tiny.... 

Where will it get stuck if a slave device gets in a bad state and won't let go of one or both lines?

I definitely remember that happening to me many times almost a decade ago, when I was working with I2C, and when it did it often looked similar to what mrWheel is seeing. Was it loose wires? power supply noise glitching the slaves? were the wires too long? Were the pullups strong enough? Maybe I should have made them stronger...  Most of the time I never figured out what caused them, either - they were always intermittent problems, they just went away and happened less over time. I'm sure some of it was dupont jumpers. I never realized just how awful that stuff was until the past 14 months or so. First, I discovered that a portion of my stock of it was flakey even brand new, then i found an affordable place for strands of it that were actually crimped onto wires instead of ribbon cable, and then noticed that the terminals on looked totally different. Then I finally got a crimp tool that worked and was able to crimp my own, and on a whim did a pull test on commercial line. Commercially available dupont line that is made from ribbon cable fails the pull test almost every time. most likely a combination of everything.  but the whole debugging process just absolutely sucked. 

(you have to feel bad for dupont over this. They made a very nice connector. It was very reliable. Then Harwin copied the concept, but instead of a spring of a material that was actually springy to hold it in place... they made the whole thing out of one piece of stamped metal and shortened the wire crimp by a third or so. Then the chinese manufacturers ripped of harwins design.... some now make reasonably faithful copies of harwin connectors, which are not terrible, and some decided it was still to costly, and shrank the wire crimp by another 40%, used thinner metal, and a smaller stamp outline. Then those get sold to jumper wire "factories" consisting of nothing other than boxes for inventory and some automatic crimping machines in some bottom tier city in china,  who buys "30 AWG" ribbon cable that's actually 34 AWG,  adjusts his machine incorrectly, and ships the crap jumpers it makes out of the country ... so now, decades after dupont got out of the connector business, THOSE are called DuPont line!)

mrWheel, 2021-09-30:
> Well, this whole ordeal made me stare at the code long enough to figure out a graceful way to add a timeout, so there was a gain from that at least... Anyway, the library is still in alpha, and what [...]

@MX682X 
I tried it but for me it makes no difference. I do suspect the T3216 is almost gone to it's makers due to powering it by the GPIO-pins (not intentionally but it might have happened, I don't know. Replacing the chip would be an option but I think my project will benefit  from the AVR-DB processors due to the extra UARTs (which I need to get rid of SoftwareSerial) and the overwhelming number of digital pins which makes the multiplexers (and I2C) redundant.

[4 more replies]